dusty-nv/jetson-containers #218: Quiestion: how to efficiently and fast transfer a docker image from the host to the Jetson?

ikhann, 2023-01-26:
Hi!

I am faced with the issue of a need for free disk space when transferring the Docker image to Jetson. 
I got the error "no space left on device" when running the command "docker load <image>":

[image]

But at the same time, I see that Jetson has more space than required for the docker image.

I will be very pleased if you help solve it and advise how to efficiently and fast transfer a docker image with a high degree of compression and load it afterward.

I really appreciate any help you can provide.

dusty-nv, 2023-01-26:
@ikhann I think the issue is that the docker image first needs to be transferred similar to an archive (like .tar.gz or similar), then decompressed, so the process can actually take up more disk space that just the image itself.   I'm not sure how to change this behavior with Docker, although perhaps some way exists.   Is it possible for you to mount additional storage?

ikhann, 2023-01-26:
> @ikhann I think the issue is that the docker image first needs to be transferred similar to an archive (like .tar.gz or similar), then decompressed, so the process can actually take up more disk [...]

Hi!
[This](https://forums.developer.nvidia.com/t/moving-docker-directory-to-nvme-ssd-cloudnative-demo-on-jetson/185248) helped me solve the problem, but only partially.

But now it really looks like I don't have enough space to load the docker image.

Result of the df -h command before transferring 25GB docker image:
[image]

Result of the df -h command before the no space error:
[image]

I use a 64 GB Sd card,  so I guess I should increase the amount of space with a 128 GB sd card or an external nvme card.

Thank you!

closed 2023-01-28 by ikhann