dusty-nv/jetson-containers #219: Listeners not reconnecting properly inside docker container after node restart

dschack1, 2023-02-01:
I'm having an issue where, when there are a large number of subscribers to a topic, killing and restarting the node that publishes that topic results in several of the connections not re-establishing correctly. 

I am able to reproduce this problem with a simple node that publishes continuously to a topic, and with 40 "rostopic echo"s that subscribe to that topic inside of the ROS Noetic docker that is provided as part of this repository (dustynv/ros:noetic-ros-base-l4t-r35.1.0). The ROS master is also running inside that same docker image. 

After a spin up a ROS core, my publisher node, and 40 subscribers, if I kill and restart the publisher node, I see CPU usage for the publisher node spike for several seconds to 100%. Screenshot below
[image]

For 30-90 seconds after restarting the node, roswtf will indicate that several nodes that are supposed to be connected are not connected. The subscribers that aren't connected at the beginning will not recieve messages from the publisher, and these initial messages will be lost. 
[image]

In addition to this, sometimes a few subscribers will remain unconnected to the publisher seemingly indefinitely (>30 minutes), though that is more challenging to reproduce.

When running the same test code outside of a docker container on the Jetson, the connections recover from the restart almost immediately (<5s) and there is no spike in CPU usage when the node first starts up. Running this test in an x86 docker on a regular laptop gives the same results as outside of the docker on the Jetson. 

This leads me to believe that something about either the way this particular docker image is configured, or something about running docker images on the Jetsons in general. Any suggestions about how to dig into that further? 

I am happy to upload my docker image with these tests scripts if they will be helpful for someone trying to debug this.

Levi-Armstrong, 2023-02-07:
@johnwason Have you seen anything like this before?

johnwason, 2023-02-07:
@Levi-Armstrong I haven't seen this exact issue before. There may be a kernel limit in /proc on the number of sockets a process can have open concurrently. There may also be an issue with too many sockets in TIME_WAIT.

What does:

```
netstat -an
```

say during the problem?

dschack1, 2023-02-08:
@johnwason - thanks for the reply. 
[netstatOutput.txt](https://github.com/dusty-nv/jetson-containers/files/10688660/netstatOutput.txt)

I've attached the output of netstat -an.

Here are some additional outputs that may be useful:

`root@cosmo-spacely:/# ulimit -a
core file size          (blocks, -c) unlimited
data seg size           (kbytes, -d) unlimited
scheduling priority             (-e) 0
file size               (blocks, -f) unlimited
pending signals                 (-i) 13939
max locked memory       (kbytes, -l) 64
max memory size         (kbytes, -m) unlimited
open files                      (-n) 1048576
pipe size            (512 bytes, -p) 8
POSIX message queues     (bytes, -q) 819200
real-time priority              (-r) 0
stack size              (kbytes, -s) 8192
cpu time               (seconds, -t) unlimited
max user processes              (-u) unlimited
virtual memory          (kbytes, -v) unlimited
file locks                      (-x) unlimited
`
`root@cosmo-spacely:/# cat /proc/sys/fs/file-max
9223372036854775807
`

johnwason, 2023-02-17:
I see a lot of TIME_WAIT sockets. If you wait about 15 minutes after stopping, does it work normally when you restart? It takes about 10-15 minutes to clear all TIME_WAIT sockets.

closed 2025-02-21 by johnnynunez