dusty-nv/jetson-containers #222: Cannot build image due to `ImportError: libcublas.so.10: cannot open shared object file: No such file or directory`

ggoretkin-bdai, 2023-02-09:
(Same error message as https://github.com/dusty-nv/jetson-containers/issues/159 , but occurring during image build time)

I am running into the error message
> `ImportError: libcublas.so.10: cannot open shared object file: No such file or directory`

while building an image, since the [Dockerfile](https://github.com/dusty-nv/jetson-containers/blob/53a882c1d373301d2998f3026a1028ee36611853/Dockerfile.ros.humble#L92) runs [`import cv2](https://github.com/dusty-nv/jetson-containers/blob/53a882c1d373301d2998f3026a1028ee36611853/scripts/opencv_install.sh#L69)

I have not yet gotten to the point of seeing if the runtime docker environment is providing `libcublas.so.10`. I am trying to build the image on an x86 host that doesn't have NVidia hardware. Is this possible?

ggoretkin-bdai, 2023-02-10:
It seems like the `l4t` images are missing these files:

```
$ docker run --platform linux/aarch64 --rm -it nvcr.io/nvidia/l4t-base:r32.5.0 find / -name "libcublas*" 2>&1 | grep -v "Permission denied"

$ docker run --platform linux/aarch64 --rm -it nvcr.io/nvidia/l4t-base:r32.7.1 find / -name "libcublas*" 2>&1 | grep -v "Permission denied"
/usr/local/cuda-10.2/targets/aarch64-linux/lib/stubs/libcublasLt.so
/usr/local/cuda-10.2/targets/aarch64-linux/lib/stubs/libcublas.so
```

whereas if I do it directly on a r32.7.2 native installation:

```
$ find / -name "libcublas*" 2>&1 | grep -v "Permission denied"
/usr/local/cuda-10.2/targets/aarch64-linux/lib/libcublas.so
/usr/local/cuda-10.2/targets/aarch64-linux/lib/libcublas.so.10
/usr/local/cuda-10.2/targets/aarch64-linux/lib/libcublas.so.10.2.3.300
/usr/local/cuda-10.2/targets/aarch64-linux/lib/libcublasLt.so
/usr/local/cuda-10.2/targets/aarch64-linux/lib/libcublasLt.so.10
/usr/local/cuda-10.2/targets/aarch64-linux/lib/libcublasLt.so.10.2.3.300
/usr/local/cuda-10.2/targets/aarch64-linux/lib/stubs/libcublas.so
/usr/local/cuda-10.2/targets/aarch64-linux/lib/stubs/libcublasLt.so
/usr/share/doc/libcublas-dev
/usr/share/doc/libcublas10
/var/lib/dpkg/info/libcublas-dev.list
/var/lib/dpkg/info/libcublas-dev.md5sums
/var/lib/dpkg/info/libcublas10.list
/var/lib/dpkg/info/libcublas10.md5sums
```
`r32.5.0` is the base image used here: https://github.com/dusty-nv/jetson-containers/blob/53a882c1d373301d2998f3026a1028ee36611853/Dockerfile.ros.humble#L5

ggoretkin-bdai, 2023-02-10:
ah, sorry, the last message is not quite relevant. This is the valid test (need the nvidia runtime)

```
$ sudo docker run --runtime=nvidia --platform linux/aarch64 --rm -it nvcr.io/nvidia/l4t-base:r32.7.1 find / -name "libcublas*" 2>&1 | grep -v "Permission denied"
/usr/local/cuda-10.2/targets/aarch64-linux/lib/libcublas.so
/usr/local/cuda-10.2/targets/aarch64-linux/lib/libcublas.so.10
/usr/local/cuda-10.2/targets/aarch64-linux/lib/libcublas.so.10.2.3.300
/usr/local/cuda-10.2/targets/aarch64-linux/lib/libcublasLt.so
/usr/local/cuda-10.2/targets/aarch64-linux/lib/libcublasLt.so.10
/usr/local/cuda-10.2/targets/aarch64-linux/lib/libcublasLt.so.10.2.3.300
/usr/local/cuda-10.2/targets/aarch64-linux/lib/stubs/libcublas.so
/usr/local/cuda-10.2/targets/aarch64-linux/lib/stubs/libcublasLt.so
```

But then it is still unclear to me how to build the docker image.

ggoretkin-bdai, 2023-02-10:
From https://github.com/dusty-nv/jetson-containers/issues/205#issuecomment-1310393417 it seems that the intention is not that this image can be built on `x86`. Building it on the Jetson means that the default docker runtime is nvidia, and this error does not show up.

dusty-nv, 2023-02-10:
Thats correct, I've only built these containers natively on Jetson (not x86).  Sorry for the confusion.

ByerRA, 2023-09-18:
I've ran into this issue today when trying to build the "l4t-ml" image on my Jetson Nano.  When torch is installed and it tries to test the installation I get the same error message.

dusty-nv, 2023-09-19:
@ByerRA is your default docker runtime set to NVIDIA? https://github.com/dusty-nv/jetson-containers/blob/master/docs/setup.md#docker-default-runtime

ByerRA, 2023-09-19:
> @ByerRA is your default docker runtime set to NVIDIA? https://github.com/dusty-nv/jetson-containers/blob/master/docs/setup.md#docker-default-runtime

Yes, NVIDIA is set as the default runtime.

`$ sudo docker info | grep -i "Default Runtime"`
`Default Runtime: nvidia`

I can confirm that the "nvcr.io/nvidia/l4t-base:r32.7.1" image which is the image being selected as the base build image in the build script does not have the "libcublas.so.10" library.

dusty-nv, 2023-09-19:
@ByerRA on JetPack 4, the CUDA/cuDNN/TensorRT libraries get mounted into container by `--runtime nvidia` from the host device - do you have libcublas.so.10 under `/usr/local/cuda/lib64/` ?

If you start l4t-base like this, it should be there - otherwise there's an issue with your NVIDIA Container Runtime:

```bash
sudo docker run -it --rm --net=host --runtime nvidia -e DISPLAY=$DISPLAY -v /tmp/.X11-unix/:/tmp/.X11-unix nvcr.io/nvidia/l4t-base:r32.7.1
```

dusty-nv, 2023-09-19:
@ByerRA it would appear you have the files, so not sure why `--runtime nvidia` is not finding them.  If you have another SD card, I would recommend trying a fresh JetPack image and checking that l4t-base works from the start.

ByerRA, 2023-09-22:
I flashed a new card with JetPack and now while I no longer get the errors about the missing cuda libraries I'm now getting the following build error...

[24/35] /usr/local/cuda-10.2/bin/nvcc  -DWITH_CUDA -I/torchvision/torchvision/csrc -I/usr/local/lib/python3.6/dist-packages/torch/include -I/usr/local/lib/python3.6/dist-packages/torch/include/torch/csrc/api/include -I/usr/local/lib/python3.6/dist-packages/torch/include/TH -I/usr/local/lib/python3.6/dist-packages/torch/include/THC -I/usr/local/cuda-10.2/include -I/usr/include/python3.6m -c -c /torchvision/torchvision/csrc/ops/cuda/roi_align_kernel.cu -o /torchvision/build/temp.linux-aarch64-3.6/torchvision/torchvision/csrc/ops/cuda/roi_align_kernel.o -D__CUDA_NO_HALF_OPERATORS__ -D__CUDA_NO_HALF_CONVERSIONS__ -D__CUDA_NO_BFLOAT16_CONVERSIONS__ -D__CUDA_NO_HALF2_OPERATORS__ --expt-relaxed-constexpr --compiler-options ''"'"'-fPIC'"'"'' -DTORCH_API_INCLUDE_EXTENSION_H '-DPYBIND11_COMPILER_TYPE="_gcc"' '-DPYBIND11_STDLIB="_libstdcpp"' '-DPYBIND11_BUILD_ABI="_cxxabi1011"' -DTORCH_EXTENSION_NAME=_C -D_GLIBCXX_USE_CXX11_ABI=1 -gencode=arch=compute_53,code=sm_53 -gencode=arch=compute_62,code=sm_62 -gencode=arch=compute_72,code=sm_72 -std=c++14
ninja: build stopped: subcommand failed.
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/cpp_extension.py", line 1723, in _run_ninja_build
    env=env)
  File "/usr/lib/python3.6/subprocess.py", line 438, in run
    output=stdout, stderr=stderr)
subprocess.CalledProcessError: Command '['ninja', '-v']' returned non-zero exit status 1.

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "setup.py", line 508, in <module>
    'clean': clean,
  File "/usr/local/lib/python3.6/dist-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/usr/lib/python3.6/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/usr/lib/python3.6/distutils/dist.py", line 955, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.6/dist-packages/wheel/bdist_wheel.py", line 299, in run
    self.run_command('build')
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/lib/python3.6/distutils/command/build.py", line 135, in run
    self.run_command(cmd_name)
  File "/usr/lib/python3.6/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.6/distutils/dist.py", line 974, in run_command
    cmd_obj.run()
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/build_ext.py", line 79, in run
    _build_ext.run(self)
  File "/usr/local/lib/python3.6/dist-packages/Cython/Distutils/old_build_ext.py", line 186, in run
    _build_ext.build_ext.run(self)
  File "/usr/lib/python3.6/distutils/command/build_ext.py", line 339, in run
    self.build_extensions()
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/cpp_extension.py", line 735, in build_extensions
    build_ext.build_extensions(self)
  File "/usr/local/lib/python3.6/dist-packages/Cython/Distutils/old_build_ext.py", line 195, in build_extensions
    _build_ext.build_ext.build_extensions(self)
  File "/usr/lib/python3.6/distutils/command/build_ext.py", line 448, in build_extensions
    self._build_extensions_serial()
  File "/usr/lib/python3.6/distutils/command/build_ext.py", line 473, in _build_extensions_serial
    self.build_extension(ext)
  File "/usr/local/lib/python3.6/dist-packages/setuptools/command/build_ext.py", line 202, in build_extension
    _build_ext.build_extension(self, ext)
  File "/usr/lib/python3.6/distutils/command/build_ext.py", line 533, in build_extension
    depends=ext.depends)
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/cpp_extension.py", line 565, in unix_wrap_ninja_compile
    with_cuda=with_cuda)
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/cpp_extension.py", line 1404, in _write_ninja_file_and_compile_objects
    error_prefix='Error compiling objects for extension')
  File "/usr/local/lib/python3.6/dist-packages/torch/utils/cpp_extension.py", line 1733, in _run_ninja_build
    raise RuntimeError(message) from e
RuntimeError: Error compiling objects for extension
The command '/bin/sh -c git clone --branch ${TORCHVISION_VERSION} --recursive --depth=1 https://github.com/pytorch/vision torchvision &&     cd torchvision &&     git checkout ${TORCHVISION_VERSION} &&     python3 setup.py bdist_wheel &&     cp dist/torchvision*.whl /opt &&     pip3 install --no-cache-dir --verbose /opt/torchvision*.whl &&     cd ../ &&     rm -rf torchvision' returned a non-zero code: 1
Traceback (most recent call last):
  File "/usr/lib/python3.6/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.6/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/rbyer/jetson-containers/jetson_containers/build.py", line 95, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.simulate, args.skip_tests, args.test_only, args.push)
  File "/home/rbyer/jetson-containers/jetson_containers/container.py", line 128, in build_container
[... cut ...]
subprocess.CalledProcessError: Command 'docker build --network=host --tag jetson_l4t-ml:r32.7.4-torchvision --file /home/rbyer/jetson-containers/packages/pytorch/torchvision/Dockerfile --build-arg BASE_IMAGE=jetson_l4t-ml:r32.7.4-pytorch --build-arg TORCHVISION_VERSION="v0.11.1" --build-arg TORCH_CUDA_ARCH_LIST="5.3;6.2;7.2" /home/rbyer/jetson-containers/packages/pytorch/torchvision 2>&1 | tee /home/rbyer/jetson-containers/logs/20230922_132123/build/jetson_l4t-ml_r32.7.4-torchvision.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.

dusty-nv, 2023-09-22:
@ByerRA there's not an actual compilation error in there, I think it ran out of memory.  Can you try mounting more swap?

https://github.com/dusty-nv/jetson-containers/blob/master/docs/setup.md#mounting-swap