dusty-nv/jetson-containers #225: OpenCV import fails using l4t-ml:r35.1.0-py3 container

kis-balazs, 2023-02-15:
Greetings!

I am building the container at hand and when I try simply ```import cv2```, the following trace pops up:

```
...
  File "/usr/local/lib/python3.8/dist-packages/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/__init__.py", line 181, in <module>
    bootstrap()
  File "/usr/local/lib/python3.8/dist-packages/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/__init__.py", line 175, in bootstrap
    if __load_extra_py_code_for_module("cv2", submodule, DEBUG):
  File "/usr/local/lib/python3.8/dist-packages/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/__init__.py", line 28, in __load_extra_py_code_for_module
    py_module = importlib.import_module(module_name)
  File "/usr/lib/python3.8/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "/usr/local/lib/python3.8/dist-packages/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/cv2/gapi/__init__.py", line 301, in <module>
    cv.gapi.wip.GStreamerPipeline = cv.gapi_wip_gst_GStreamerPipeline
AttributeError: partially initialized module 'cv2' has no attribute 'gapi_wip_gst_GStreamerPipeline' (most likely due to a circular import)
```

It seems like being a recursive import, circular structure but I don't exactly understand where could that be. Help would be highly appreciated :)

kis-balazs, 2023-02-15:
I was making the mistake of installing opencv-python in my requirements file. My bad! ^^

closed 2023-02-15 by kis-balazs

dusty-nv, 2023-02-15:
No worries @kis-balazs, glad you got it figured out!