dusty-nv/jetson-containers #226: Wiki : How to build and run ROS2 Humble on Jetson AGX Xavier

htex0189, 2023-02-16:
Hi,

I am trying to install ROS2 Humble on Jetson AGX Xavier. Since Jetson AGX Xavier currently is with ubuntu 20.04 so I think I have to use the ROS container. 

I did the following 
$ git clone https://github.com/dusty-nv/jetson-containers.git
$ cd jetson-containers
$ scripts/docker_build_ros.sh --distro humble --package desktop

It has built successfully with some stderr for some packages.

Please could you provide the next steps to test ros and run/work with ros?

dusty-nv, 2023-02-16:
Hi @htex0189, you can run `scripts/docker_run.sh -c <CONTAINER>` to start the container - for example, `scripts/docker_run.sh -c ros:humble-desktop-l4t-r35.1.0` if you are on L4T R35.1

Merwanski, 2024-08-15:
@htex0189  Using the following docker images worked for me https://hub.docker.com/r/dustynv/ros/tags
Thank you @dusty-nv

closed 2025-02-21 by johnnynunez