dusty-nv/jetson-containers #227: "unauthorized: authentication required". Error ROS FOXY Jetson AGX Xavier

lucyannofrota, 2023-02-16:
### I'm getting an error using the following command:
`sudo ./scripts/docker_build_ros.sh --distro foxy --package desktop`
### Output:

> ARCH:  aarch64
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R35.2.1
L4T Base Image:   nvcr.io/ea-linux4tegra/l4t-jetpack:r35.2.1
selecting OpenCV for L4T R35.2.1...
OPENCV_URL=https://nvidia.box.com/shared/static/2hssa5g3v28ozvo3tc3qwxmn78yerca9.gz
OPENCV_DEB=OpenCV-4.5.0-aarch64.tar.gz
ROS_DISTRO:   foxy
ROS_PACKAGE:  desktop
WITH_PYTORCH: off
>
> Building container ros:foxy-desktop-l4t-r35.2.1
BASE_IMAGE=nvcr.io/ea-linux4tegra/l4t-jetpack:r35.2.1
>
> Building ros:foxy-desktop-l4t-r35.2.1 container...
Sending build context to Docker daemon  178.7kB
> 
> [WARNING]: Empty continuation line found in:
RUN mkdir -p ${ROS_ROOT}/src `&&`     cd ${ROS_ROOT} `&&`     rosinstall_generator --deps --rosdistro ${ROS_DISTRO}  ${ROS_PKG}          launch_xml             launch_yaml              launch_testing          launch_testing_ament_cmake              demo_nodes_cpp          demo_nodes_py           example_interfaces     camera_calibration_parsers               camera_info_manager             cv_bridge               v4l2_camera             vision_opencv           vision_msgs    image_geometry           image_pipeline          image_transport                 compressed_image_transport              compressed_depth_image_transport       > ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall `&&`     cat ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall `&&`     vcs import src < ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall `&&`     rm -r ${ROS_ROOT}/src/ament_cmake `&&`     git -C ${ROS_ROOT}/src/ clone https://github.com/ament/ament_cmake -b ${ROS_DISTRO} `&&`     apt-get update `&&`     cd ${ROS_ROOT} `&&`     rosdep init &&     rosdep update &&     rosdep install -y           --ignore-src        --from-paths src    --rosdistro ${ROS_DISTRO}      --skip-keys "libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv" &&     rm -rf /var/lib/apt/lists/* `&&`     apt-get clean `&&`     colcon build         --merge-install         --cmake-args -DCMAKE_BUILD_TYPE=Release &&     rm -rf ${ROS_ROOT}/src `&&`     rm -rf ${ROS_ROOT}/logs `&&`     rm -rf ${ROS_ROOT}/build `&&`     rm ${ROS_ROOT}/*.rosinstall
>
> [WARNING]: Empty continuation lines will become errors in a future release.
>
> Step 1/36 : ARG BASE_IMAGE=nvcr.io/nvidia/l4t-base:r32.5.0
Step 2/36 : FROM ${BASE_IMAGE}
unauthorized: authentication required

### Extra notes
I was able to build the container without the script "docker_build_ros.sh" and got ROS2 working fine. But I couldn't execute GUI applications like rviz2. I was getting the following error:

> qt.qpa.xcb: could not connect to display 
> qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
> This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
>
> Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.
>
> Aborted (core dumped)

So I decided to try the "docker_build_ros.sh" hoping to solve the problem.

dusty-nv, 2023-02-16:
> unauthorized: authentication required

Hi @lucyannofrota, if you pull the latest, this should be fixed now.  `nvcr.io/nvidia/l4t-jetpack:r35.2.1` was just released publicly today, which is the correct base container for this.  Also, there is a pre-built Foxy-desktop image for JetPack 5.1 here:  `dustynv/ros:foxy-desktop-l4t-r35.2.1`

lucyannofrota, 2023-02-17:
Now the script worked smoothly and was able to test most of the functionalities (including the use of GUI).
Thank you very much for your response @dusty-nv, you're an awesome repository manager! You did an excellent job in this repository!

Hope you have a fantastic day sir.

closed 2023-02-17 by lucyannofrota