dusty-nv/jetson-containers #231: Trying to create ROS2 Humble desktop container but agot access denied?

luvwinnie, 2023-03-24:
I have  done the `docker login nvcr.io`, then When I try to build the container it shows access denied. How can I solve this?

```ai@machine01:~/jetson-containers$ ./scripts/docker_build_ros.sh --distro humble --package desktop
ARCH:  aarch64
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R34.1.1
L4T Base Image:   jetpack:r34.1.1
selecting OpenCV for L4T R34.1.1...
OPENCV_URL=https://nvidia.box.com/shared/static/2hssa5g3v28ozvo3tc3qwxmn78yerca9.gz
OPENCV_DEB=OpenCV-4.5.0-aarch64.tar.gz
ROS_DISTRO:   humble
ROS_PACKAGE:  desktop
WITH_PYTORCH: off

Building container ros:humble-desktop-l4t-r34.1.1
BASE_IMAGE=jetpack:r34.1.1

Building ros:humble-desktop-l4t-r34.1.1 container...
Sending build context to Docker daemon  179.2kB
[WARNING]: Empty continuation line found in:
    RUN mkdir -p ${ROS_ROOT}/src &&     cd ${ROS_ROOT} &&     rosinstall_generator --deps --rosdistro ${ROS_DISTRO} ${ROS_PKG} 		launch_xml 		launch_yaml 		launch_testing 		launch_testing_ament_cmake 		demo_nodes_cpp 		demo_nodes_py 		example_interfaces 		camera_calibration_parsers 		camera_info_manager 		cv_bridge 		v4l2_camera 		vision_opencv 		vision_msgs 		image_geometry 		image_pipeline 		image_transport 		compressed_image_transport 		compressed_depth_image_transport 		> ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     cat ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     vcs import src < ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall &&     rm -r ${ROS_ROOT}/src/ament_cmake &&     git -C ${ROS_ROOT}/src/ clone https://github.com/ament/ament_cmake -b ${ROS_DISTRO} &&     apt-get update &&     cd ${ROS_ROOT} &&     rosdep init &&     rosdep update &&     rosdep install -y     	  --ignore-src        --from-paths src 	  --rosdistro ${ROS_DISTRO} 	  --skip-keys "libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv" &&     rm -rf /var/lib/apt/lists/* &&     apt-get clean &&     colcon build         --merge-install         --cmake-args -DCMAKE_BUILD_TYPE=Release &&     rm -rf ${ROS_ROOT}/src &&     rm -rf ${ROS_ROOT}/logs &&     rm -rf ${ROS_ROOT}/build &&     rm ${ROS_ROOT}/*.rosinstall
[WARNING]: Empty continuation lines will become errors in a future release.
Step 1/35 : ARG BASE_IMAGE=nvcr.io/nvidia/l4t-base:r32.5.0
Step 2/35 : FROM ${BASE_IMAGE}
pull access denied for jetpack, repository does not exist or may require 'docker login': denied: requested access to the resource is denied```

dusty-nv, 2023-03-24:
> L4T BSP Version:  L4T R34.1.1

Hi @luvwinnie, L4T R34.1 was a developer preview release.  I would recommend upgrading to the latest JetPack, or using the existing image `dustynv/ros:humble-ros-base-l4t-r34.1.1`

luvwinnie, 2023-03-25:
Hi @dusty-nv , Thank you for reply. I was thinking to upgrade the jetpack and install the ros2 humble. 

Do you know any guide or websites that help us upgrade jetpack and install ros2 humble step by step?

dusty-nv, 2023-03-26:
> I was thinking to try the isaac_ros_dnn_inference but seems like foxy can't build due to some packages might need ROS2 Humble.

Isaac ROS has containers in https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common for Humble and the ROS nodes

closed 2025-02-21 by johnnynunez