dusty-nv/jetson-containers #232: Support l4t 35.3.1

wep21, 2023-03-30:
I have tried to build image with `nvcr.io/nvidia/l4t-base:35.3.1`, but I had trouble with installing opencv as below.
```
dpkg: dependency problems prevent configuration of opencv-libs:
 opencv-libs depends on libavcodec57 (>= 7:3.4.8) | libavcodec-extra57 (>= 7:3.4.8); however:
  Package libavcodec57 is not installed.
  Package libavcodec-extra57 is not installed.
 opencv-libs depends on libavformat57 (>= 7:3.4.8); however:
  Package libavformat57 is not installed.
 opencv-libs depends on libavutil55 (>= 7:3.4.8); however:
  Package libavutil55 is not installed.
 opencv-libs depends on libswscale4 (>= 7:3.4.8); however:
  Package libswscale4 is not installed.
 opencv-libs depends on libtbb2; however:
  Package libtbb2 is not installed.
 opencv-libs depends on libtesseract4; however:
  Package libtesseract4 is not installed.
```
I guess we need to upgrade opencv version, but I don't know the url which provides newer opencv built with cuda.

dusty-nv, 2023-03-30:
You would need to install the ffmpeg libraries, but it would be likely that until l4t-jetpack:r35.3.1 is released you will run into further issues I would think.   For the time being, you can just run l4t-jetpack:r35.2.1 (and it's derivative containers) on L4T R35.3.1.  Since JetPack5, you don't need to upgrade the containers all the time anymore (because CUDA/ect are installed inside the containers now as opposed to being mounted from the host - and CUDA/cuDNN/TensorRT has not changed versions from L4T R35.2.1 to R35.3.1)

wep21, 2023-03-31:
I see. Thank you for your kind explanation.

closed 2023-03-31 by wep21

wep21, 2023-03-31:
@dusty-nv I have tried to build with `nvcr.io/nvidia/l4t-base:r35.2.1`, but the same errors above still occurred.
Could you check it out?

dusty-nv, 2023-03-31:
Can you try building with l4t-jetpack:r35.2.1 instead?

dusty-nv, 2023-03-31:
What container / dockerfile are you trying to build?   Because in my containers, I use the opencv_install.sh script, and the dependencies with dpkg get installed here:

https://github.com/dusty-nv/jetson-containers/blob/d0b715e12105aeeeb55aba9f08c47c2d64a78f6b/scripts/opencv_install.sh#L25

wep21, 2023-04-05:
I have reproduced the error as below.
```
git clone https://github.com/dusty-nv/jetson-containers.git
cd jetson-containers/
docker run -it --rm --net=host -v ./scripts:/scripts nvcr.io/nvidia/l4t-base:r35.2.1
scripts/opencv_install.sh https://nvidia.box.com/shared/static/5v89u6g5rb62fpz4lh0rz531ajo2t5ef.gz OpenCV-4.5.0-aarch64.tar.gz
```

Jinshiya, 2023-08-11:
I also get the same error @wep21 , Were you able to solve it?

Albertchamberlain, 2023-12-26:
Same problem, Were you able to solve it?