dusty-nv/jetson-containers #234: How to keep persistence?

TomGrimwood, 2023-04-14:
Hi Dusty and all

I think I am missing something with these containers, How do I use these for development?
I create a container with the command 
```bash
scripts/docker_run.sh -c dustynv/ros:galactic-desktop-l4t-r35.2.1
```
Then I copy some files in, and build a package I made.
But when I restart the computer, or close the terminal, all the modifications to the container are gone?

I was familiar with the Isaac_ros_common containers with their run_dev.sh script which doesn't erase my changes to the container

How do you use these containers for development, to start and stop the container easily without resetting my modifications?

talbertherndon, 2025-01-30:
I feel like by default it should persist... I build out a full Jupyter notebook only to come back the next day and my docker container gone!

talbertherndon, 2025-01-30:
No your al good—thanks for creating this package! I resolved the issue by removing the --rm flag, as you suggested. Also, assigning a name to the container is essential for maintaining persistence across restarts/reboots. 

I also downloaded and would recommend for beginners to get Portainer to have a better visual/control on whats going on.