dusty-nv/jetson-containers #235: ImportError: libcublas.so.10: cannot open shared object file: No such file or directory

kenloumixx, 2023-04-15:
Hello, 
I use the following image as the base image for my container
FROM nvcr.io/nvidia/l4t-pytorch:r32.7.1-pth1.10-py3

But when OpenCV installation test scripts starts, this message appears. 
`File "/usr/local/lib/python3.6/dist-packages/cv2/init.py", line 86, in bootstrap
import cv2
ImportError: libcublas.so.10: cannot open shared object file: No such file or directory`

I referred to other issues in this repo, and run nvcr.io/nvidia/l4t-pytorch:r32.7.1-pth1.10-py3 image alone. Inside the container, `import cv2` works well without the above error. 

In my local host, the 'libcublas.so' located in 
`/usr/local/cuda-10.2/targets/aarch64-linux/lib/libcublas.so
/usr/local/cuda-10.2/targets/aarch64-linux/lib/libcublas.so.10
/usr/local/cuda-10.2/targets/aarch64-linux/lib/stubs/libcublas.so
/usr/local/cuda-10.2/targets/aarch64-linux/lib/libcublas.so.10.2.3.300
/usr/local/cuda-10.2/doc/man/man7/libcublas.so.7
`
I discovered that the libcublas files are also in the same location as in the local host. However, when I bring the base image the 'libcublas.so' located in
`/usr/local/cuda-10.2/targets/aarch64-linux/lib/stubs/libcublas.so
/usr/local/cuda-10.2/targets/aarch64-linux/lib/stubs/libcublasLt.so
`
What can I do to handle issue? thank you!

dusty-nv, 2023-04-15:
Hi @kenloumixx, did you set your default docker runtime yet to nvidia?   https://github.com/dusty-nv/jetson-containers#docker-default-runtime

It should then use the nvidia runtime during `docker build` operations and enable you to use those libraries.

kenloumixx, 2023-04-15:
@dusty-nv Yes I set the default docker runtime following the link, restart docker using `systemctl restart docker` and get the same results!

dusty-nv, 2023-04-17:
@kenloumixx if you check `sudo docker info`, does it show `Default Runtime: nvidia` ?

Also which version of L4T are you running?    Is it L4T R32.7.x?   (you can check this with `cat /etc/nv_tegra_release`)

kenloumixx, 2023-04-17:
@dusty-nv Yes, when I checked `sudo docker info` it says `Default Runtime: nvidia`. My L4T version is R32.7.3,
`# R32 (release), REVISION: 7.3, GCID: 31982016, BOARD: t186ref, EABI: aarch64, DATE: Tue Nov 22 17:32:54 UTC 2022` 

Here is the output of `sudo docker info`. 

Client:
 Context:    default
 Debug Mode: false
 Plugins:
  buildx: Docker Buildx (Docker Inc.)
    Version:  v0.10.4
    Path:     /usr/libexec/docker/cli-plugins/docker-buildx
  compose: Docker Compose (Docker Inc.)
    Version:  v2.17.2
    Path:     /usr/libexec/docker/cli-plugins/docker-compose

Server:
 Containers: 2
  Running: 1
  Paused: 0
  Stopped: 1
 Images: 11
 Server Version: 23.0.3
 Storage Driver: overlay2
  Backing Filesystem: extfs
  Supports d_type: true
  Using metacopy: false
  Native Overlay Diff: true
  userxattr: false
 Logging Driver: json-file
 Cgroup Driver: cgroupfs
 Cgroup Version: 1
 Plugins:
  Volume: local
  Network: bridge host ipvlan macvlan null overlay
  Log: awslogs fluentd gcplogs gelf journald json-file local logentries splunk syslog
 Swarm: inactive
 Runtimes: io.containerd.runc.v2 nvidia runc
 Default Runtime: nvidia
 Init Binary: docker-init
 containerd version: 2806fc1057397dbaeefbea0e4e17bddfbd388f38
 runc version: v1.1.5-0-gf19387a
 init version: de40ad0
 Security Options:
  seccomp
   Profile: builtin
 Kernel Version: 4.9.299-tegra
 Operating System: Ubuntu 18.04.6 LTS
 OSType: linux
 Architecture: aarch64
 CPUs: 4
 Total Memory: 15.45GiB
 Name: ubuntu
 ID: 4GS3:QSLK:4Y4D:ZKYX:47WL:4B72:UJ3B:HVSQ:XERZ:TJIH:7ORA:XORK
 Docker Root Dir: /mnt/mmcblk1p1/var/lib/docker
 Debug Mode: false
 Registry: https://index.docker.io/v1/
 Experimental: false
 Insecure Registries:
  127.0.0.0/8
 Live Restore Enabled: false

dusty-nv, 2023-04-18:
OK, since you have confirmed that the container works with `import cv2` when you are running it, just not when you are building it, I'm not sure what the issue is at this point.  Can you post the console log of when you build the container?

kenloumixx, 2023-04-18:
Here is the console log, thanks!

```
[+] Building 29.4s (16/26)                                                       => WARN: Empty continuation line found in: RUN mkdir -p ${ROS_ROOT}/src   0.1s  => [ 1/23] FROM nvcr.io/nvidia/l4t-ml:r32.7.1-py3                         0.0s
 => CACHED [ 2/23] WORKDIR /tmp                                            0.0s[ => CACHED [ 3/23] RUN locale-gen en_US en_US.UTF-8 && update-locale LC_A  0.0s  => CACHED [ 4/23] RUN update-alternatives --install /usr/bin/python pyth  0.0s
 => CACHED [ 5/23] RUN apt-get update &&     apt-get 

> install -y --no-inst  0.0s [+] Building 53.5s (17/26)                                                       => [internal] load build definition from Dockerfile.ros.humble            0.1s

 => => transferring dockerfile: 7.31kB                                     0.0s8 => WARN: Empty continuation line found in: RUN mkdir -p ${ROS_ROOT}/src   0.1s  => [internal] load .dockerignore                                          0.0s
 => => transferring context: 2B                                            0.0sa => [internal] load metadata for nvcr.io/nvidia/l4t-ml:r32.7.1-py3         0.0s  => [internal] load build context                                          0.0s
 => => transferring context: 276B                                          0.0sa => [ 1/23] FROM nvcr.io/nvidia/l4t-ml:r32.7.1-py3                         0.0s  => CACHED [ 2/23] WORKDIR /tmp                                            0.0s
 => CACHED [ 3/23] RUN locale-gen en_US en_US.UTF-8 && update-locale LC_A  0.0s[ => CACHED [ 4/23] RUN update-alternatives --install /usr/bin/python pyth  0.0s  => CACHED [ 5/23] RUN apt-get update &&     apt-get install -y --no-inst  0.0s
 => CACHED [ 6/23] RUN curl -sSL https://raw.githubusercontent.com/ros/ro  0.0s 
 => CACHED [ 7/23] RUN echo "deb [arch=$(dpkg --print-architecture) signe  0.0s 
 => CACHED [ 8/23] RUN add-apt-repository -y ppa:ubuntu-toolchain-r/test   0.0s 
 => CACHED [ 9/23] RUN update-alternatives --install /usr/bin/gcc gcc /us  0.0s 
 => CACHED [10/23] RUN apt-get update &&     apt-get install -y --no-inst  0.0s 
 => CACHED [11/23] RUN python3 -m pip install -U   argcomplete   flake8-b  0.0s 
 => CACHED [12/23] COPY jetson-containers/scripts/opencv_install.sh /tmp/  0.0s 
 => ERROR [13/23] RUN cd /tmp && ./opencv_install.sh https://nvidia.box.  53.3s
------
 > [13/23] RUN cd /tmp && ./opencv_install.sh https://nvidia.box.com/shared/static/5v89u6g5rb62fpz4lh0rz531ajo2t5ef.gz OpenCV-4.5.0-aarch64.tar.gz:
#0 0.249 OPENCV_URL = https://nvidia.box.com/shared/static/5v89u6g5rb62fpz4lh0rz531ajo2t5ef.gz
#0 0.249 + OPENCV_URL=https://nvidia.box.com/shared/static/5v89u6g5rb62fpz4lh0rz531ajo2t5ef.gz
#0 0.249 + OPENCV_DEB=OpenCV-4.5.0-aarch64.tar.gz
#0 0.249 + echo 'OPENCV_URL = https://nvidia.box.com/shared/static/5v89u6g5rb62fpz4lh0rz531ajo2t5ef.gz'
#0 0.250 OPENCV_DEB = OpenCV-4.5.0-aarch64.tar.gz
#0 0.250 + echo 'OPENCV_DEB = OpenCV-4.5.0-aarch64.tar.gz'
#0 0.251 ++ uname -i
#0 0.254 + ARCH=aarch64
#0 0.254 + echo 'ARCH:  aarch64'
#0 0.254 + apt-get purge -y '.*opencv.*'
#0 0.254 ARCH:  aarch64
#0 0.292 Reading package lists...
#0 0.356 Building dependency tree...
#0 0.362 Reading state information...
#0 0.417 The following packages were automatically installed and are no longer required:
#0 0.417   libgif7 liblept5 libtbb2 libtesseract4
#0 0.417 Use 'apt autoremove' to remove them.
#0 0.426 The following packages will be REMOVED:
#0 0.426   opencv-dev* opencv-libs* opencv-licenses* opencv-main* opencv-python*
#0 0.426   opencv-scripts*
#0 1.176 0 upgraded, 0 newly installed, 6 to remove and 0 not upgraded.
#0 1.176 After this operation, 825 MB disk space will be freed.
(Reading database ... 53619 files and directories currently installed.)
#0 1.414 Removing opencv-dev (4.5.0) ...
#0 1.690 Removing opencv-libs (4.5.0) ...
#0 1.874 Removing opencv-licenses (4.5.0) ...
#0 1.992 Removing opencv-main (4.5.0) ...
#0 2.114 Removing opencv-python (4.5.0) ...
#0 2.188 dpkg: warning: while removing opencv-python, directory '/usr/lib/python3.6/dist-packages' not empty so not removed
#0 2.233 Removing opencv-scripts (4.5.0) ...
#0 2.417 Processing triggers for libc-bin (2.27-3ubuntu1.4) ...
#0 2.449 /sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libcudnn_cnn_infer.so.8 is empty, not checked.
#0 2.449 /sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libcudnn_ops_train.so is empty, not checked.
#0 2.450 /sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libcudnn_adv_infer.so.8 is empty, not checked.
#0 2.450 /sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libvulkan.so.1.2.141 is empty, not checked.
#0 2.450 /sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libv4l2.so.0.0.999999 is empty, not checked.
#0 2.450 /sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libnvinfer.so.8 is empty, not checked.
#0 2.450 /sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libnvinfer.so is empty, not checked.
#0 2.450 /sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libv4lconvert.so.0.0.999999 is empty, not checked.
#0 2.450 /sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libdrm_nvdc.so is empty, not checked.
#0 2.450 /sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libcudnn_cnn_train.so.8 is empty, not checked.
#0 2.450 /sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libcudnn_adv_infer.so is empty, not checked.
#0 2.450 /sbin/ldconfig.real: File /usr/lib/aarch64-linux-gnu/libcuda.so is empty, not checked.
[... cut ...]
#0 52.73 + python3 -c 'import cv2; print('\''OpenCV version:'\'', str(cv2.__version__)); print(cv2.getBuildInformation())'
#0 53.12 Traceback (most recent call last):
#0 53.12   File "<string>", line 1, in <module>
#0 53.12   File "/usr/local/lib/python3.6/dist-packages/cv2/__init__.py", line 96, in <module>
#0 53.12     bootstrap()
#0 53.12   File "/usr/local/lib/python3.6/dist-packages/cv2/__init__.py", line 86, in bootstrap
#0 53.12     import cv2
#0 53.12 ImportError: libcublas.so.10: cannot open shared object file: No such file or directory
------

```

kenloumixx, 2023-04-18:
@dusty-nv How can I disable BUILDKIT?
I tried it in two ways 
```
$ export DOCKER_BUILDKIT=0
$ docker build -t my-image .
```
`docker build --progress=plain -t my-image .`

Neither worked. Is there other way that you recommend?

kenloumixx, 2023-05-08:
I splashed the jetpack again and it works! thank you

closed 2023-05-08 by kenloumixx