dusty-nv/jetson-containers #236: Orin Nano ros humble desktop builds but pytorch is failing on jetson-inference

brianlmerritt, 2023-04-21:
The pytorch part of the build outputs:

```
Building ros:humble-pytorch-l4t-r35.3.1 container...
[sudo] password for blm: 
Sorry, try again.
[sudo] password for blm: 
Sending build context to Docker daemon  181.2kB
[WARNING]: Empty continuation line found in:
   ......
[WARNING]: Empty continuation lines will become errors in a future release.
Step 1/33 : ARG BASE_IMAGE=nvcr.io/nvidia/l4t-base:r32.5.0
Step 2/33 : FROM ${BASE_IMAGE}
manifest for dustynv/jetson-inference:r35.3.1 not found: manifest unknown: manifest unknown

```

dusty-nv, 2023-04-21:
@brianlmerritt I don't have `jetson-inference:r35.3.1` yet, but will start building it (and the l4t-pytorch that it depends on)

For now, you should be able to hardcode `dustynv/jetson-inference:r35.2.1` in this line:   https://github.com/dusty-nv/jetson-containers/blob/59f840abbb99f22914a7b2471da829b3dd56122e/scripts/docker_build_ros.sh#L134

brianlmerritt, 2023-04-22:
Cool and thanks again!

closed 2023-04-22 by brianlmerritt