dusty-nv/jetson-containers #237: Issues building OpenCV 4.5.5 (OpenGL and Python)

stan-guer, 2023-04-25:
Building OpenCV 4.5.5 (and 4.5.4 for that matter) with `./scripts/docker_build_opencv.sh` seems to come with the following problems:

```
/usr/bin/ld: ../../lib/libopencv_highgui.so.4.5.5: undefined reference to `cvSetOpenGlDrawCallback'
/usr/bin/ld: ../../lib/libopencv_highgui.so.4.5.5: undefined reference to `cvSetOpenGlContext'
/usr/bin/ld: ../../lib/libopencv_highgui.so.4.5.5: undefined reference to `cvUpdateWindow'
collect2: error: ld returned 1 exit status
make[2]: *** [apps/annotation/CMakeFiles/opencv_annotation.dir/build.make:90: bin/opencv_annotation] Error 1
make[1]: *** [CMakeFiles/Makefile2:6923: apps/annotation/CMakeFiles/opencv_annotation.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

If I build with `-D BUILD_opencv_apps=OFF ` to just omit the offending `opencv_annotation` , the same error will happen during the process of building some ROS package that uses OpenCV.

It seems like the workaround that actually works is to disable OpenGL and enable GTK for highgui: `-D WITH_OPENGL=OFF -D WITH_GTK=ON`. 

Another issue later occurs on the python check at https://github.com/dusty-nv/jetson-containers/blob/master/scripts/opencv_install.sh#L69 when building the target Docker container. OpenCV 4.5.5 has python install destination at `site-packages` and not `dist-packages`.

OpenCV 4.5.5:
```
--   Python 3:
--     Interpreter:                 /usr/bin/python3 (ver 3.8.10)
--     Libraries:                   /usr/lib/aarch64-linux-gnu/libpython3.8.so (ver 3.8.10)
--     numpy:                       /usr/lib/python3/dist-packages/numpy/core/include (ver 1.17.4)
--     install path:                lib/python3.8/site-packages/cv2/python-3.8
```
OpenCV 4.5.0:
```
--   Python 3:
--     Interpreter:                 /usr/bin/python3 (ver 3.8.10)
--     Libraries:                   /usr/lib/aarch64-linux-gnu/libpython3.8.so (ver 3.8.10)
--     numpy:                       /usr/lib/python3/dist-packages/numpy/core/include (ver 1.17.4)
--     install path:                lib/python3.8/dist-packages/cv2/python-3.8
```

I build using `./scripts/docker_build_ros.sh --distro humble` after running `./scripts/docker_build_opencv.sh`. The Dockerfiles and install scripts have been modified to use the local opencv tar and not to fetch the one from the nvidia cloud. 

Maybe this information will be of help to people to fix their problems and maybe some fix can be pushed to the repo.

pacientes, 2023-06-22:
have you solved this?

stan-guer, 2023-06-22:
> have you solved this?

Yes, I've linked towards the solutions in the post above.

closed 2025-02-21 by johnnynunez