dusty-nv/jetson-containers #239: Error messages with 'undefined symbol'

kenloumixx, 2023-04-29:
Hello, I am a newbie for ROS1 and ROS2. I use Jetson Xavier AGX with Jetpack SDK 4.6.3, which supports Ubuntu 18.04. Although the upper version of Ubuntu is compatible with AGX, to simulate the final code in TX2 I have to fix the Ubuntu version as 18.04.

Using your docker container, I build ROS2 humble from source and build packages from source too.

I succeeded to build all the packages that I needed, but when I run the launch file the below error appears

Error1) The package that has just been built outputs an undefined symbol [robot_state_publisher-1] /root/ros2_humble/install/lib/robot_state_publisher/robot_state_publisher: symbol lookup error: /root/ros2_humble/install/lib/robot_state_publisher/robot_state_publisher: undefined symbol: _ZNK12class_loader11ClassLoader14getLibraryPathB5cxx11Ev

What I did) Installed the robot state publisher package by git clone and build from source but the same error keeps occurring

Error2) Bunch of undefined symbol error occurs [component_container_isolated-3] [INFO] [1682701839.864571588] [nav2_container]: Load Library: /root/ros2_pkgs/install/lib/libcontroller_server_core.so [component_container_isolated-3] [ERROR] [1682701839.979859721] [nav2_container]: Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libcontroller_server_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [component_container_isolated-3] [INFO] [1682701839.983249149] [nav2_container]: Load Library: /root/ros2_pkgs/install/lib/libmap_server_core.so [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'controller_server' of type 'nav2_controller::ControllerServer' in container '/nav2_container': Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libcontroller_server_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [component_container_isolated-3] [ERROR] [1682701840.003610261] [nav2_container]: Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libmap_server_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'map_server' of type 'nav2_map_server::MapServer' in container '/nav2_container': Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libmap_server_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [component_container_isolated-3] [INFO] [1682701840.056172602] [nav2_container]: Load Library: /root/ros2_pkgs/install/lib/libsmoother_server_core.so [component_container_isolated-3] [ERROR] [1682701840.083442431] [nav2_container]: Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libsmoother_server_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [component_container_isolated-3] [INFO] [1682701840.095752612] [nav2_container]: Load Library: /root/ros2_pkgs/install/lib/libamcl_core.so [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'smoother_server' of type 'nav2_smoother::SmootherServer' in container '/nav2_container': Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libsmoother_server_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [component_container_isolated-3] [ERROR] [1682701840.116262721] [nav2_container]: Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libamcl_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'amcl' of type 'nav2_amcl::AmclNode' in container '/nav2_container': Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libamcl_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [component_container_isolated-3] [INFO] [1682701840.144925078] [nav2_container]: Load Library: /root/ros2_pkgs/install/lib/libplanner_server_core.so [component_container_isolated-3] [ERROR] [1682701840.150345071] [nav2_container]: Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libplanner_server_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [component_container_isolated-3] [INFO] [1682701840.162698710] [nav2_container]: Load Library: /root/ros2_pkgs/install/lib/libnav2_lifecycle_manager_core.so [component_container_isolated-3] [ERROR] [1682701840.170959088] [nav2_container]: Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libnav2_lifecycle_manager_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'planner_server' of type 'nav2_planner::PlannerServer' in container '/nav2_container': Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libplanner_server_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'lifecycle_manager_localization' of type 'nav2_lifecycle_manager::LifecycleManager' in container '/nav2_container': Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libnav2_lifecycle_manager_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [component_container_isolated-3] [INFO] [1682701840.201203546] [nav2_container]: Load Library: /root/ros2_pkgs/install/lib/libbehavior_server_core.so [component_container_isolated-3] [ERROR] [1682701840.216321887] [nav2_container]: Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libbehavior_server_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'behavior_server' of type 'behavior_server::BehaviorServer' in container '/nav2_container': Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libbehavior_server_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [component_container_isolated-3] [INFO] [1682701840.229592645] [nav2_container]: Load Library: /root/ros2_pkgs/install/lib/libbt_navigator_core.so [component_container_isolated-3] [ERROR] [1682701840.244705002] [nav2_container]: Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libbt_navigator_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'bt_navigator' of type 'nav2_bt_navigator::BtNavigator' in container '/nav2_container': Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libbt_navigator_core.so: undefined symbol: _ZN12class_loader4impl22AbstractMetaObjectBaseC2ERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES9_S9_) [component_container_isolated-3] [INFO] [1682701840.254544315] [nav2_container]: Load Library: /root/ros2_pkgs/install/lib/libwaypoint_follower_core.so [component_container_isolated-3] [ERROR] [1682701840.259580679] [nav2_container]: Failed to load library: Could not load library (Poco exception = /root/ros2_pkgs/install/lib/libwaypoint_follower_core.so: undefined symbol: _ZN12class_loader4impl2 [... cut ...] RROR] [rviz2-2]: process has died [pid 20351, exit code 127, cmd '/root/ros2_pkgs/install/lib/rviz2/rviz2 -d /workspace/install/share/a1_robot/rviz/nav2_default_view.rviz --ros-args -r __node:=rviz2'].

What I did) I had no choice but to google the issue and haven't find the answer yet.

Is there any idea of how can I solve this problem?
This is more related to the navigation but I think you might have seen this error so just to get some advices to solve this error.
Thank you for the great work!

dusty-nv, 2023-05-01:
@kenloumixx unfortunately I haven't seen this error before and am not sure how to fix it, I'm sorry.  Does this happen with my pre-built ROS Humble container from DockerHub (listed in this table:  https://github.com/dusty-nv/jetson-containers/blob/master/README.md#pre-built-container-images)  

Or, I am curious if this happens for you with Foxy/Galactic, because Humble uses some newer stuff (as it's based on 22.04).  Actually, I realized that my Humble container images are only for JetPack 5 (20.04) because I couldn't get it building for 18.04 on JetPack 4.

kenloumixx, 2023-05-03:
@dusty-nv Hello, I tried with pre-built Galactic container too, which is `dustynv/ros:galactic-pytorch-l4t-r32.7.1 `. Then, the error appears as below 

```
[INFO] [launch]: All log files can be found below /root/.ros/log/2023-05-03-13-32-33-571494-ubuntu-13041
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [robot_state_publisher-1]: process started with pid [13047]
[ERROR] [robot_state_publisher-1]: process has died [pid 13047, exit code 127, cmd '/opt/ros/galactic/install/lib/robot_state_publisher/robot_state_publisher /workspace/install/share/a1_robot/urdf/a1.urdf --ros-args -r __ns:=/ --params-file /tmp/launch_params_zdi4jtjp -r /tf:=tf -r /tf_static:=tf_static'].
[INFO] [rviz2-2]: process started with pid [13048]
[INFO] [map_server-3]: process started with pid [13049]
[INFO] [amcl-4]: process started with pid [13050]
[INFO] [lifecycle_manager-5]: process started with pid [13051]
[INFO] [controller_server-6]: process started with pid [13052]
[INFO] [planner_server-7]: process started with pid [13053]
[INFO] [recoveries_server-8]: process started with pid [13055]
[INFO] [bt_navigator-9]: process started with pid [13062]
[INFO] [waypoint_follower-10]: process started with pid [13074]
[INFO] [lifecycle_manager-11]: process started with pid [13075]
[INFO] [ascot_vision-12]: process started with pid [13078]
[INFO] [ascot_main-13]: process started with pid [13079]
[map_server-3] [INFO] [map_io]: Loading yaml file: /workspace/install/share/a1_robot/maps/a1_navigation.yaml
[map_server-3] [DEBUG] [map_io]: resolution: 0.05
[map_server-3] [DEBUG] [map_io]: origin[0]: -11.975
[map_server-3] [DEBUG] [map_io]: origin[1]: -17.975
[map_server-3] [DEBUG] [map_io]: origin[2]: 0
[map_server-3] [DEBUG] [map_io]: free_thresh: 0.196
[map_server-3] [DEBUG] [map_io]: occupied_thresh: 0.65
[map_server-3] [DEBUG] [map_io]: mode: trinary
[map_server-3] [DEBUG] [map_io]: negate: 0
[map_server-3] [INFO] [map_io]: Loading image_file: /workspace/install/share/a1_robot/maps/a1_navigation.pgm
[robot_state_publisher-1] /opt/ros/galactic/install/lib/robot_state_publisher/robot_state_publisher: symbol lookup error: /opt/ros/galactic/install/lib/robot_state_publisher/robot_state_publisher: undefined symbol: _ZNK12class_loader11ClassLoader14getLibraryPathB5cxx11Ev
[rviz2-2] QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[map_server-3] [INFO] [1683120756.754135558] [map_server]: 
[map_server-3]  map_server lifecycle node launched. 
[map_server-3]  Waiting on external lifecycle transitions to activate
[map_server-3]  See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-3] [INFO] [1683120756.755464190] [map_server]: Creating
[map_server-3] [INFO] [1683120756.838268938] [map_server]: Configuring
[lifecycle_manager-5] [INFO] [1683120756.797588018] [lifecycle_manager_localization]: Creating
[lifecycle_manager-5] [INFO] [1683120756.809058359] [lifecycle_manager_localization]: Creating and initializing lifecycle service clients
[lifecycle_manager-5] [INFO] [1683120756.837047670] [lifecycle_manager_localization]: Starting managed nodes bringup...
[lifecycle_manager-5] [INFO] [1683120756.837212317] [lifecycle_manager_localization]: Configuring map_server
[controller_server-6] /root/ros2_pkgs/install/lib/nav2_controller/controller_server: symbol lookup error: /root/ros2_pkgs/install/lib/nav2_controller/controller_server: undefined symbol: _ZNK12class_loader11ClassLoader14getLibraryPathB5cxx11Ev
[recoveries_server-8] /root/ros2_pkgs/install/lib/nav2_recoveries/recoveries_server: symbol lookup error: /root/ros2_pkgs/install/lib/nav2_recoveries/recoveries_server: undefined symbol: _ZNK12class_loader11ClassLoader14getLibraryPathB5cxx11Ev
[ERROR] [recoveries_server-8]: process has died [pid 13055, exit code 127, cmd '/root/ros2_pkgs/install/lib/nav2_recoveries/recoveries_server --ros-args -r __node:=recoveries_server --params-file /tmp/tmpryhl4x0k -r /tf:=tf -r /tf_static:=tf_static'].
[ERROR] [controller_server-6]: process has died [pid 13052, exit code 127, cmd '/root/ros2_pkgs/install/lib/nav2_controller/controller_server --ros-args --params-file /tmp/tmpxw7ylq96 -r /tf:=tf -r /tf_static:=tf_static'].
[planner_server-7] [INFO] [1683120757.203027812] [planner_server]: 
[planner_server-7]      planner_server lifecycle node launched. 
[planner_server-7]      Waiting on external lifecycle transitions to activate
[planner_server-7]      See https://design.ros2.org/articles/node_lifecycle.html for more information.
[planner_server-7] [INFO] [1683120757.221187043] [planner_server]: Creating
[waypoint_follower-10] [INFO] [1683120757.289173021] [waypoint_follower]: 
[waypoint_follower-10]  waypoint_follower lifecycle node launched. 
[waypoint_follower-10]  Waiting on external lifecycle transitions to activate
[waypoint_follower-10]  See https://design.ros2.org/articles/node_lifecycle.html for more information.
[waypoint_follower-10] [INFO] [1683120757.306386549] [waypoint_follower]: Creating
[lifecycle_manager-11] [INFO] [1683120757.351928538] [lifecycle_manager_navigation]: Creating
[lifecycle_manager-11] [INFO] [1683120757.427095139] [lifecycle_manager_navigation]: Creating and initializing lifecycle service clients
[lifecycle_manager-11] [INFO] [1683120757.534694919] [lifecycle_manager_navigation]: Starting managed nodes bringup...
[lifecycle_manager-11] [INFO] [1683120757.534872206] [lifecycle_manager_navigation]: Configuring controller_server
[... cut ...]
[ascot_vision-12]     import ascot_vision.FastMOT.fastmot as fastmot
[ascot_vision-12]   File "/workspace/build/ascot_vision/ascot_vision/FastMOT/fastmot/__init__.py", line 2, in <module>
[ascot_vision-12]     from .mot import MOT
[ascot_vision-12]   File "/workspace/build/ascot_vision/ascot_vision/FastMOT/fastmot/mot.py", line 5, in <module>
[ascot_vision-12]     import numba as nb
[ascot_vision-12] ModuleNotFoundError: No module named 'numba'
```

I think I can handle other errors, but the 'undefined symbol' still appears and I am struggling with it..

dusty-nv, 2023-05-03:
> /root/ros2_humble/install/lib/robot_state_publisher/robot_state_publisher: undefined symbol: _ZNK12class_loader11ClassLoader14getLibraryPathB5cxx11Ev

I found the same error appears in this thread:  https://github.com/ros/robot_state_publisher/issues/153

kenloumixx, 2023-05-03:
@dusty-nv Yes I also source the right distro using `source /opt/ros/galactic/install/setup.bash` but the error still occurs.

dusty-nv, 2023-05-03:
@kenloumixx sorry, I am not sure how to fix it, or why it would have missing symbols when the package was compiled from source.   Here is how I build/install ROS2 packages on top of the images:

https://github.com/dusty-nv/jetbot_ros/blob/d8e5ee1b17f5c66d038017b86f8920a496197ea9/Dockerfile#L69

closed 2025-02-21 by johnnynunez