dusty-nv/jetson-containers #241: ERROR: yaml_cpp_vendor: No definition of [ament_cmake] for OS version [focal]

XujqGitHub, 2023-05-02:
I used the following command to install ROS2 humble in r35.3.1:

`./scripts/docker_build_ros.sh --distro humble --package desktop`

But I got the following error at `rosdep install`:

```bash
Skip end-of-life distro "ardent"
Skip end-of-life distro "bouncy"
Skip end-of-life distro "crystal"
Skip end-of-life distro "dashing"
Skip end-of-life distro "eloquent"
Add distro "foxy"
Skip end-of-life distro "galactic"
Skip end-of-life distro "groovy"
Add distro "humble"
Skip end-of-life distro "hydro"
Skip end-of-life distro "indigo"
Add distro "iron"
Skip end-of-life distro "jade"
Skip end-of-life distro "kinetic"
Skip end-of-life distro "lunar"
Add distro "melodic"
Add distro "noetic"
Add distro "rolling"
updated cache in /root/.ros/rosdep/sources.cache
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
yaml_cpp_vendor: No definition of [ament_cmake] for OS version [focal]
The command '/bin/bash -c apt-get update &&     cd ${ROS_ROOT} && 	export ROSDISTRO_INDEX_URL=https://mirrors.tuna.tsinghua.edu.cn/rosdistro/index-v4.yaml &&     rosdep update &&     rosdep install -y     	  --ignore-src        --from-paths src 	  --rosdistro ${ROS_DISTRO} 	  --skip-keys "libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv" &&     rm -rf /var/lib/apt/lists/* &&     apt-get clean' returned a non-zero code: 1
```

dusty-nv, 2023-05-02:
@XujqGitHub I will be rebuilding the ROS containers for L4T R35.3.1, but for now you can use the ones for L4T R35.2.1

XujqGitHub, 2023-05-02:
@dusty-nv 
Thanks, I look forward to the ROS containers for L4T R35.3.1.

XujqGitHub, 2023-05-02:
Hi,
@dusty-nv
I have resolved the above-mentioned issue by executing the command `./scripts/docker_build_ros.sh --distro humble --package desktop` multiple times. The reason producing this issue maybe my network quality. 
What's more, I'm newbie for ROS. I also noticed [NVIDIA-ISAAC-ROS](https://github.com/NVIDIA-ISAAC-ROS). I also want to know what are differences between this repository and NVIDIA-ISAAC-ROS? Can you give me some information?

dusty-nv, 2023-05-02:
@XujqGitHub the ROS/ROS2 containers for JetPack 5.1.1 (L4T R35.3.1) have now been pushed to DockerHub, including the base/pytorch/desktop images:

https://hub.docker.com/repository/docker/dustynv/ros/tags?page=1&ordering=last_updated&name=r35.3.1

I have yet to update the table with these in the readme, but will in the coming days.

XujqGitHub, 2023-05-03:
> that were built with CUDA enabled so that they can use GPU acceleration if you so choose

Thanks for your help. In this repo, which type of ROS image supports CUDA? desktop? or pytorch? or all of image? 
What's more, other packages like Gazebo, MoveIt, ros2_control. Do I need to install these packages myself in my container?

XujqGitHub, 2023-05-03:
Do the `ros:*-pytorch-*` images consist of ros-base (excluding ROS GUI) + PyTorch? I can't run Rviz2 by the command `ros2 run rviz2 rviz2`.

closed 2023-05-03 by XujqGitHub