dusty-nv/jetson-containers #242: error mounting "/tmp/.docker.xauth" to rootfs at "/tmp/.docker.xauth"

XujqGitHub, 2023-05-06:
I want to reserve the container, so modify the docker_run.sh by deleting the `--rm` para:
```dockerfile
sudo docker run --runtime nvidia -it --network host \
	$DISPLAY_DEVICE $V4L2_DEVICES \
	$USER_VOLUME $CONTAINER_IMAGE $USER_COMMAND
```
By this way, I can `sudo docker stop` or `docker start` the container successfully.
But when I reboot my jetson AGX devkit and then run the following command to start the exited container, I meet error:
```bash
$ sudo docker ps -a
CONTAINER ID   IMAGE                                    COMMAND       CREATED        STATUS                   PORTS     NAMES
9b48f5758b8e   dustynv/ros:humble-desktop-l4t-r35.3.1   "/bin/bash"   16 hours ago   Exited (0) 8 hours ago             blissful_hellman
$ sudo docker start blissful_hellman
Error response from daemon: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error mounting "/tmp/.docker.xauth" to rootfs at "/tmp/.docker.xauth": mount /tmp/.docker.xauth:/tmp/.docker.xauth (via /proc/self/fd/7), flags: 0x5000: not a directory: unknown: Are you trying to mount a directory onto a file (or vice-versa)? Check if the specified host path exists and is the expected type
Error: failed to start containers: blissful_hellman
```
Can you help me to solve this error? I want to persist the created container, how should I do it?

dusty-nv, 2023-05-06:
@XujqGitHub after you have rebooted, can you run `sudo xhost +si:localuser:root` ?

Or if you aren't using display at all, comment out these lines from `docker/run.sh` script:
https://github.com/dusty-nv/jetson-inference/blob/0c64aa7bdd470a8012036156d1e233e81b548f02/docker/run.sh#L170

Then destroy your container and make a new one without the display stuff mounted.

XujqGitHub, 2023-05-07:
Hi,
Thanks for your reminder and help, inspired by the following piece of code I solved the problem I encountered:
```shell
if [ -n "$DISPLAY" ]; then
	# give docker root user X11 permissions
	sudo xhost +si:localuser:root
	
	# enable SSH X11 forwarding inside container (https://stackoverflow.com/q/48235040)
	XAUTH=/tmp/.docker.xauth
	xauth nlist $DISPLAY | sed -e 's/^..../ffff/' | xauth -f $XAUTH nmerge -
	chmod 777 $XAUTH

	DISPLAY_DEVICE="-e DISPLAY=$DISPLAY -v /tmp/.X11-unix/:/tmp/.X11-unix -v $XAUTH:$XAUTH -e XAUTHORITY=$XAUTH"
fi
```

closed 2023-05-11 by XujqGitHub