dusty-nv/jetson-containers #244: Dockerfile for ros humble

Fibo27, 2023-05-14:
Hi Dusty

The dockerfile for ros humble https://github.com/dusty-nv/jetson-containers/blob/master/Dockerfile.ros.humble uses the base image nvcr.io/nvidia/l4t-base:r32.5.0. 

1. If I have to build from scratch can I use nvcr.io/nvidia/l4t-ml:r35.2.1-py3 - I have jetpack 5.1 and i want to deploy this on a Orin Nano?
2. Does the prebuilt container in dockerhub "dustynv/ros:humble-pytorch-l4t-r35.3.1" include the script that you have in the above dockerfile wherein it access the ros_deep_learning module at the very end? If not, then should I start  with a base image of in the container "dustynv/ros:humble-pytorch-l4t-r35.3.1" and add the script for ros_deep_learning and then build a container from scratch? 
3. My understanding is that the ros_deep_learning script in turn access the various inference networks available in jetson-inference repo - is this understanding correct?

thanks

dusty-nv, 2023-05-15:
> The dockerfile for ros humble https://github.com/dusty-nv/jetson-containers/blob/master/Dockerfile.ros.humble uses the base image nvcr.io/nvidia/l4t-base:r32.5.0.

@Fibo27 the `BASE_IMAGE` arg gets set dynamically by the docker_build_ros.sh script based on what version of JetPack-L4T you are running

> 1. If I have to build from scratch can I use nvcr.io/nvidia/l4t-ml:r35.2.1-py3 - I have jetpack 5.1 and i want to deploy this on a Orin Nano?

Yes, but if you want to use ros_deep_learning it should use jetson-inference container as base image

> 2\. Does the prebuilt container in dockerhub "dustynv/ros:humble-pytorch-l4t-r35.3.1" include the script that you have in the above dockerfile wherein it access the ros_deep_learning module at the very end?

Yes, it does include ros_deep_learning

> 3\. My understanding is that the ros_deep_learning script in turn access the various inference networks available in jetson-inference repo - is this understanding correct?

Yes, this is correct

Fibo27, 2023-05-15:
Ok thanks a lot again!

Fibo27, 2023-05-15:
Hi Dusty - I reopened the issue as there are a few follow ons:

1. Yes, but if you want to use ros_deep_learning it should use jetson-inference container as base image

Regarding this comment of yours, what is the difference between Base, Desktop and Pytorch Containers - my understanding is that Base: uses Ros Base set-up, Desktop: uses Ros Desktop setup. Do both of them include Jetson Inference and consequently Ros Deep Learning? 
What is the Pytorch container? Does it include Jetson Inference and Ros Deep Learning? Is it based on the  Desktop version of Ros?
Thanks

dusty-nv, 2023-05-15:
@Fibo27 only the pytorch images include pytorch/jetson-inference/ros_deep_learning, and the pytorch images use ros-base.  If you want you can change that by modifying the `docker_build_ros.sh` script though and rebuilding it.

Fibo27, 2023-05-15:
Got it - thanks!

closed 2023-05-15 by Fibo27