dusty-nv/jetson-containers #245: Issues with python <3.7 on JetPack 4.6.x

balloch, 2023-05-15:
I am trying to run YOLOX on my Jetson Nano with JetPack 4.6.3 using the the l4t-pytorch:r32.7.1-pth1.10-py3 image. However, it seems like this leads internal conflicts (specifically set off by ONNX and Torchvision). I tried in two ways.

In the first case I tried to simply pull and run the prebuilt image, clone the repo ( https://github.com/Megvii-BaseDetection/YOLOX ) inside the interactive container. However, when I try to simply `pip install -e .` the library, I get the error:
```
(from torchvision->yolox==0.3)
Pillow requires Python '>=3.7' but the running Python is 3.6.9
``` 
The second thing I tried was to build the image from scratch, and even with the suggestions from issue #177 and #202 and manually trying to upgrade my python to 3.8, the docker container still seeks out python3.6 and 3.6-related packages (like libpython3.6-dev) leading to internal build warnings ( like trying to pull numpy 1.24.3 which requires py3.8) and avoids the 3.8 error ultimately leads to the follow error:

(During Step 7/32 : RUN pip3 install --no-cache-dir --verbose onnx) 
```
assert CMAKE, "Could not find cmake executable!"
AssertionError: Could not find cmake executable!
```
I have checked that we have cmake installed. Do you know how I can get around this?

dusty-nv, 2023-05-16:
> (from torchvision->yolox==0.3)

Hi @balloch, I would remove torchvision, torch, and opencv_python from that project's requirements.txt because it may not realize those are already installed and will try to install versions without GPU acceleration.    Manually upgrading to Python 3.8 will leave you without PyTorch wheels that support GPU, because we only build them for the native version of Python that comes with Ubuntu (and on JetPack 4, that's Ubuntu 18.04 and Python 3.6 - you could attempt to rebuild PyTorch for Python 3.8 though)

balloch, 2023-05-16:
I'll try this thanks!

eyast, 2023-05-18:
Thanks for the conversation - I need python 3.7 as well (I use vscode with devcontainers, and I really want to use the vscode debug functionality which requires python 3.7+). If the cuplrit is the OS, which is linked to the version of Jetpack (and assuming that Jetpack 5 has a more recent OS image) - how would one install JP5+ on the Nano DevKit? Thanks!! (PS: this repo helped me get started with my project in little to no time - thanks @dusty-nv and team)

dusty-nv, 2023-05-18:
Hi @eyast, unfortunately Jetson Nano doesn't support JetPack 5, there has been some community efforts on the forums however:  https://forums.developer.nvidia.com/t/ubuntu-22-04-image-beta-test-for-jetson-nano/253301

Other than your typical build issues there's nothing really to prevent you from rebuilding PyTorch/ect for Python 3.7/3.8 and JetPack 4 / Ubuntu 18.04, other than time to build it all depending on what you need.

closed 2025-02-21 by johnnynunez