dusty-nv/jetson-containers #246: Container on Desktop machine

reinzler, 2023-05-16:
Hi, will whis container works on my Desktop PC? Before building container on my Jetson orin nano I want to try it on my desktop PC (Ubuntu 20.04). I want to use container with ROS2 Humble and Pytorch. Also I dont fully understand will the same container works on Jetson Orin Nano (Ubuntu 20.04) and Jetson Nano (Ubuntu 18.04). Thanks in advance.

dusty-nv, 2023-05-16:
> Hi, will whis container works on my Desktop PC?

Hi @reinzler no this container won't work on PC, however their are mainline ROS docker images available for x86 already:   https://hub.docker.com/_/ros/  (you would take their x86 containers and install PyTorch on top of it)

> Also I dont fully understand will the same container works on Jetson Orin Nano (Ubuntu 20.04) and Jetson Nano (Ubuntu 18.04)

There are different container tags built for different versions of JetPack-L4T.   For example, for Orin Nano running JetPack 5.1 / L4T R35.2.1, you would run `dustynv/ros:foxy-pytorch-l4t-r35.2.1`.  Whereas for Jetson Nano running JetPack 4.6 / L4T R32.7.1 you would run `dustynv/ros:foxy-pytorch-l4t-r32.7.1`

You can find lists of the container tags available at https://github.com/dusty-nv/jetson-containers#pre-built-container-images or on dockerhub:  https://hub.docker.com/repository/docker/dustynv/ros/general

reinzler, 2023-05-17:
Thanks for your answer, is it possible to install ROS2 Humble on Jetson Nano?

dusty-nv, 2023-05-17:
@reinzler I haven't been able to get Humble to build on JetPack 4 / Ubuntu 18.04, there are too many package/build issues to overcome

dusty-nv, 2023-05-17:
@reinzler see https://github.com/dusty-nv/jetson-containers/issues/160#issuecomment-1429572145

dusty-nv, 2023-05-20:
@reinzler I built container images for ROS2 Humble on JetPack 4.6 / L4T R32.7 in commit https://github.com/dusty-nv/jetson-containers/commit/9eaecf0012755bbe4734a1f4dd54018cf298c36a and pushed them to DockerHub.

reinzler, 2023-05-20:
Thanks a lot!

closed 2025-02-21 by johnnynunez