dusty-nv/jetson-containers #249: Navigation2 with Humble container

Usimian, 2023-05-31:
I'm having trouble getting Navigation2 on a Humble desktop container to work on a Jetson Xavier.
The Jetson is flashed with 5.1.1 and the humble container is pulled **dustynv/ros:humble-desktop-l4t-r35.3.1**

Nav2 won't install with _sudo apt install ros-humble-navigation2_ (unable to locate pkg)
Tried building Navigation2 from source inside the container but I still get a lot of dependency errors even after running:
rosdep install -y -r -q --from-paths src --ignore-src --rosdistro humble

> ERROR: the following packages/stacks could not have their rosdep keys resolved
> to system dependencies:
> nav2_lifecycle_manager: No definition of [diagnostic_updater] for OS version [focal]
> nav2_behavior_tree: No definition of [behaviortree_cpp_v3] for OS version [focal]
> nav2_bringup: No definition of [slam_toolbox] for OS version [focal]
> nav2_system_tests: No definition of [gazebo_ros_pkgs] for OS version [focal]
> nav2_util: No definition of [bond] for OS version [focal]
> nav2_smac_planner: No definition of [ompl] for OS version [focal]
> nav2_bt_navigator: No definition of [behaviortree_cpp_v3] for OS version [focal]
> nav2_mppi_controller: [xsimd] defined as "not available" for OS version [focal]
> Continuing to install resolvable dependencies...
> executing command [apt-get install -y -qq libgraphicsmagick++1-dev]
> E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
> executing command [apt-get install -y -qq graphicsmagick-libmagick-dev-compat]
> E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
> executing command [apt-get install -y -qq python3-zmq]
> E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
> executing command [apt-get install -y -qq lcov]
> E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
> executing command [apt-get install -y -qq nlohmann-json3-dev]
> E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
> executing command [apt-get install -y -qq libzmq3-dev]
> E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
> executing command [apt-get install -y -qq libomp-dev]
> E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
> executing command [apt-get install -y -qq libceres-dev]
> E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
> ERROR: the following rosdeps failed to install
>   apt: command [apt-get install -y -qq libgraphicsmagick++1-dev] failed
>   apt: command [apt-get install -y -qq graphicsmagick-libmagick-dev-compat] failed
>   apt: command [apt-get install -y -qq python3-zmq] failed
>   apt: command [apt-get install -y -qq lcov] failed
>   apt: command [apt-get install -y -qq nlohmann-json3-dev] failed
>   apt: command [apt-get install -y -qq libzmq3-dev] failed
>   apt: command [apt-get install -y -qq libomp-dev] failed
>   apt: command [apt-get install -y -qq libceres-dev] failed
>   apt: Failed to detect successful installation of [libgraphicsmagick++1-dev]
>   apt: Failed to detect successful installation of [graphicsmagick-libmagick-dev-compat]
>   apt: Failed to detect successful installation of [python3-zmq]
>   apt: Failed to detect successful installation of [lcov]
>   apt: Failed to detect successful installation of [nlohmann-json3-dev]
>   apt: Failed to detect successful installation of [libzmq3-dev]
>   apt: Failed to detect successful installation of [libomp-dev]
>   apt: Failed to detect successful installation of [libceres-dev]

I've tried everything I can think of but I think I'm too deep in the weeds.
I've used the Nav2 stack a lot on Jetson/Galactic natively without issue but Jetson/Humble needs a container.
**_Has anyone gotten Navigation2 running on a Jetson Xavier or Orin Nano using a Humble container?_**

dusty-nv, 2023-06-01:
Hi @Usimian, isaac-ros-core builds nav2 on humble:   

https://github.com/NVIDIA-ISAAC-ROS/isaac_ros_common/blob/b9834877677e759cd19bf878bbc885d02ee54539/docker/Dockerfile.ros2_humble#L146

Usimian, 2023-06-03:
Thank you @dusty-nv for the help, I got it working with one issue.
I can't install ros-humble- packages.  I can build them from source however.
Something like sudo apt install ros-humble-teleop-twist-joy returns a cannot find package msg.
But sudo apt install ros-galactic-teleop-twist-joy does find the package.
I'm not sure how a humble container running on 20.04 finds ros packages.

dusty-nv, 2023-06-04:
> But sudo apt install ros-galactic-teleop-twist-joy does find the package.

Galactic can find the packages, because Galactic is based on 20.04 and OSRF provides the debian packages for Galactic already built for 20.04.    They only provide Humble packages for 22.04 though, which is why you don't see those and need to build them from source.

Usimian, 2023-06-04:
Ahh, of course, thank you.
Sorry for the noob question.  I'm still learning containers.

closed 2023-06-04 by Usimian