dusty-nv/jetson-containers #250: install joint state publisher and xcro packages

Fibo27, 2023-06-03:
Hi @dusty-nv 

Please refer to my earlier issue #244 - i have since been able to create the following:
1) container with humble-pytorch-ros_deep_learning on Ubuntu 20.04
2) I am now trying to install ros-humble-joint-state-publisher-gui, ros-humble-xacro and ros-humble-localization packages as mentioned here https://navigation.ros.org/setup_guides/urdf/setup_urdf.html 

When i try to do that inside the container, i get an error that the packages could not be found. I tried to source Ros2 again but to no avail. would you be able to suggest a work around.

Thanks

dusty-nv, 2023-06-04:
@Fibo27 you need to build Humble packages from source, because OSRF doesn't provide pre-built debian packages for Humble on 20.04 (they only provide them for 22.04)

You should be able to use rosinstall_generator / rosdep to pull the sources & dependencies of the packages you want (assuming that they can in fact be built for 20.04)   This dockerfile has an example of doing that, where it builds/installs additional packages on top of the initial ROS2 build:

https://github.com/dusty-nv/jetbot_ros/blob/d8e5ee1b17f5c66d038017b86f8920a496197ea9/Dockerfile#L72

Fibo27, 2023-06-05:
Understood - thanks

closed 2023-06-05 by Fibo27