dusty-nv/jetson-containers #251: How to source custom ros 2workspace?

mzahana, 2023-06-04:
Hi @dusty-nv 
I am using `dustynv/ros:humble-desktop-l4t-r35.1.0` docker image on Jetson Xavier NX.

I added my custom Dockerfile layer based on `dustynv/ros:humble-desktop-l4t-r35.1.0` . In my custom layer, I need to build a custom `ros2_ws`.  I can execute `colcon build` inside `ros2_ws` successfully.  However, even after I execute `source install/setup.bash`, I am not able to use `ros2 launch pkg_name launch_file.py.launch`.  I get the following error
```bash
Package  'pkg_name' not found: "package 'pkg_name' not found, searching: [ '/ros_deep_learning/install/ros_deep_learning', '/opt/ros/humble/install']"
```

How can I fix this issues such that I can launch nodes from my custom `ros2_ws` ?

Thanks

mzahana, 2023-06-04:
Also one weird thing!
I have multiple ros2 packages inside `ros2_ws/src`. When I execute `colcon build`, only 2 out of the many packages get recognized and built. I am sure that the other packages are proper ros2 packages as I can build them on regular ros installation. This is driving me crazy. What could be the issue here?

dusty-nv, 2023-06-04:
@mzahana for some reason, it looks like I set `AMENT_PREFIX_PATH` in addition to sourcing `pkg/install/setup.bash`:

https://github.com/dusty-nv/jetson-containers/blob/413372747f67dcba487846fb357293bd9c9a747d/scripts/ros_entrypoint.sh#L19

I kind of remember debugging it and ROS not updating that environment var for some reason, but could not figure out why not.

dusty-nv, 2023-06-04:
It would seem that it could be related to https://github.com/colcon/colcon-ros/issues/67

mzahana, 2023-06-04:
I think the colcon tool is not installed properly in the docker image.

dusty-nv, 2023-06-04:
@mzahana colcon gets installed the "normal" way from the `python3-colcon-common-extensions` debian package (which is available)

ament_cmake does get replaced though:  https://github.com/dusty-nv/jetson-containers/blob/413372747f67dcba487846fb357293bd9c9a747d/scripts/ros2_build.sh#L102

mzahana, 2023-06-04:
So, is there a solution that I may try for the problem I mentioned?

dusty-nv, 2023-06-05:
> So, is there a solution that I may try for the problem I mentioned?

@mzahana try including the path to your package in `AMENT_PREFIX_PATH` as mentioned in https://github.com/dusty-nv/jetson-containers/issues/251#issuecomment-1575628217

closed 2025-02-21 by johnnynunez