dusty-nv/jetson-containers #252: ROS nodes discovery issue over multiple machines

mzahana, 2023-06-06:
Hi @dusty-nv 
I am using the ros `humble` with `pytorch` container on Jetson Xavier NX `r35.1.0`. It is working fine except for one network issue described as follows.

* I run the container and then run some nodes. I did not set the `ROS_DOMAIN_ID`
* I have another PC that is  connected to the same WiFi router as the Jetson in the previous step, and it has ros `humble`. I also don't set `ROS_DOMAIN_ID` on the PC. I then execute `ros2 node list` , but I don't see the nodes that are running on the Jetson.

I am not sure what the issue could be here.

Any idea?

Thanks.

mzahana, 2023-06-06:
It might be similar to [this](https://answers.ros.org/question/403647/ros-2-talker-listener-on-2-machines-not-working/) issue. 

Which `rmw` implementation is used in this container @dusty-nv ?

dusty-nv, 2023-06-06:
> Which `rmw` implementation is used in this container @dusty-nv ?

@mzahana it defaults to cyclonedds, but you should be able to override this at runtime:   https://github.com/dusty-nv/jetson-containers/blob/413372747f67dcba487846fb357293bd9c9a747d/Dockerfile.ros2#L43

I'm not sure what is all involved with multicast or multi-host discoverability