dusty-nv/jetson-containers #254: turtlesim package in ros2.humble.desktop.rosinstall refers rolling branch

otets-u, 2023-06-13:
Hello, @dusty-nv -san.
I really appreciate your great work about this repository.

Now, I faced the title error when I run docker build for Humble with PyTorch.

```
./scripts/docker_build_ros.sh --distro humble --package desktop --with-pytorch
```
After while
```
HEAD is now at fbd4968 Rebase from 'upstream'
The command '/bin/bash -c ./ros2_build.sh' returned a non-zero code: 1
```

So I checked the script process step by step and build the previous version of [Dockerfile](https://github.com/dusty-nv/jetson-containers/blob/L4T-R35.1.0/Dockerfile.ros.humble) for debug.

As a result, turtlesim package in ros2.humble.desktop.rosinstall refers rolling branch.

I think that it may have the relation with the error.
How can I fix it ?

Thank you.

```
=== src/turtlesim (git) ===
Could not checkout ref 'release/rolling/turtlesim/1.4.2-1': fatal: invalid reference: release/rolling/turtlesim/1.4.2-1
=== src/uncrustify_vendor (git) ===
Cloning into '.'...
Note: switching to 'release/humble/uncrustify_vendor/2.0.2-2'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at e2bddd9 Rebase from 'upstream' (no changes)
```

dusty-nv, 2023-06-13:
@otets-u my guess is there was a change upstream in the Humble package manifests where the references to that turtlesim package got updated/broken.   The last time I built the Humble packages (~1-2 weeks ago), I did not get that error.

Do you need the turtlesim or desktop packages?    You could try disabling turtlesim by adding it to the skip-keys, or use ros_base instead of desktop.

otets-u, 2023-06-15:
@dusty-nv -san
Thank you for answering.
Finally, I can avoid the error based on your comment.

> Do you need the turtlesim or desktop packages? 

Yes, I want to use the GUI tools like RViz and plotjuggler to check the topic data.
However, I don't use the turtlesim package.

So, I followed your comment and I added the below 2 small changes.

**exclude turtlesim package** 
```
	compressed_depth_image_transport \
	--exclude turtlesim \
> ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall
cat ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall
vcs import src < ros2.${ROS_DISTRO}.${ROS_PKG}.rosinstall
```

**skip-keys**
```
# skip installation of some conflicting packages
SKIP_KEYS="libopencv-dev libopencv-contrib-dev libopencv-imgproc-dev python-opencv python3-opencv turtlesim"
```

And then, I successfully built the software. 
Thank you.

closed 2023-06-15 by otets-u