dusty-nv/jetson-containers #260: Building failing for ROS2 humble desktop image on Jetson Orin AGX

BeLeever, 2023-07-03:
Hi there,

I've been working on the Jetson Orin AGX running Jetpack 5.1.1, and L4T version R35.3.1,

I've successfully pulled your prebuilt image from dockerhub and it works great!,

However when I come to running the following script, with the following parameters in order to build the image locally, it is unable to build successfully,

"./scripts/docker_build_ros.sh --distro humble --package desktop" 

It apprears to be failing at step 21/27, at RUN ./ros2_build.sh, where the final output is 
"The command 'bin/bash -c ./ros2_build.sh' returned a non-zero code: 1"

The script has successfully built the core and base images, just not working for the desktop version.

dusty-nv, 2023-07-03:
@BeLeever there should be other errors up above that - can you post the entire build log here?

BeLeever, 2023-07-03:
Unfortunately, I can't post the entire build logs as it exceeds the character limit, 

i have put a trimmed down version on paste bin,

https://pastebin.com/4AVRb9MK

Hope this is helpful

Towards the end there is a series of these messages 

 Skipping link: none of the wheel's tags match: py2-none-musllinux_1_1_ppc64le, py3-none-musllinux_1_1_ppc64le: https://files.pythonhosted.org/packages/dc/83/4c4bf54b874241ebff53d214a96d5b30314c65b6640c603393c9627cd534/cmake-3.26.4-py2.py3-none-musllinux_1_1_ppc64le.whl#sha256=24110035aff586a04a6a6fcf4609270642e4f503c0620c962dff75b653f81414 (from https://pypi.org/simple/cmake/)
  Skipping link: none of the wheel's tags match: py2-none-musllinux_1_1_s390x, py3-none-musllinux_1_1_s390x: https://files.pythonhosted.org/packages/cc/93/ad4136831157d78bf32baf280a573b5b696c24b69c9ea60ab601c2de17b7/cmake-3.26.4-py2.py3-none-musllinux_1_1_s390x.whl#sha256=3e280e81713408987b7053f5b922c9f94e45668ca6efff1f02846309ca0b5b0f (from https://pypi.org/simple/cmake/)
  Skipping link: none of the wheel's tags match: py2-none-musllinux_1_1_x86_64, py3-none-musllinux_1_1_x86_64: https://files.pythonhosted.org/packages/5e/84/501f9f304b714f1ec09b42899a45ff2d0122bc8dc5c075b5e489c299f046/cmake-3.26.4-py2.py3-none-musllinux_1_1_x86_64.whl#sha256=c3b0e72750c0f6c0373242c1299bc4ffdbebdd5004966ae6df0b2e9845aa6990 (from https://pypi.org/simple/cmake/)
  Skipping link: none of the wheel's tags match: py2-none-win32, py3-none-win32: https://files.pythonhosted.org/packages/79/f7/899657c4d3b4d88cac49978133256ad6a0fd9c90eee11ce11de82b7ba099/cmake-3.26.4-py2.py3-none-win32.whl#sha256=e058e59154a1e490fb9425b420f87e28144292397607638d73e323509f7efae6 (from https://pypi.org/simple/cmake/)
  Skipping link: none of the wheel's tags match: py2-none-win_amd64, py3-none-win_amd64: https://files.pythonhosted.org/packages/66/c8/997eab3e90f030d216a0ccacfe6f4c153f5108784d1b2f41b61ba4deadbb/cmake-3.26.4-py2.py3-none-win_amd64.whl#sha256=b7a6946c345497c14064e0c9585b30f5aaebbefdfc0b245b6bb5a978eb4fc85f (from https://pypi.org/simple/cmake/)
  Skipping link: none of the wheel's tags match: py2-none-win_arm64, py3-none-win_arm64: https://files.pythonhosted.org/packages/49/a7/a103cae38c1cdb3a926d030aa91f93631c92cc5944b97faf7077c9b4d41b/cmake-3.26.4-py2.py3-none-win_arm64.whl#sha256=93a03bad17b9741acaff4a8651f8596496506602fa123e70fe67142f1b21ee2e (from https://pypi.org/simple/cmake/)
  Found link https://files.pythonhosted.org/packages/8a/a7/da765b3f525ac9ab34e1fb700a4b95307ba33bc408461714b61e0938dd88/cmake-3.26.4.tar.gz#sha256=d45b30b9ce7280829888c78650177ab525df2b6785e1a5b3d82b4c147d828c0e (from https://pypi.org/simple/cmake/), version: 3.26.4
Given no hashes to check 102 links for project 'cmake': discarding no candidates
Using version 3.26.4 (newest of versions: 0.1.0, 0.2.0, 0.4.0, 0.5.0, 0.6.0, 0.7.0, 0.7.1, 0.8.0, 0.9.0, 3.6.3, 3.6.3.post1, 3.7.2, 3.8.2, 3.9.6, 3.10.3, 3.11.0, 3.11.4, 3.11.4.post1, 3.12.0, 3.13.0, 3.13.1, 3.13.2, 3.13.2.post1, 3.14.3.post1, 3.14.4.post1, 3.15.3.post1, 3.16.3, 3.16.3.post1, 3.16.5, 3.16.6, 3.16.7, 3.16.8, 3.17.0, 3.17.1, 3.17.2, 3.17.3, 3.18.0, 3.18.2, 3.18.2.post1, 3.18.4, 3.18.4.post1, 3.20.2, 3.20.3, 3.20.4, 3.20.5, 3.21.0, 3.21.1, 3.21.1.post1, 3.21.2, 3.21.3, 3.21.4, 3.22.0, 3.22.1, 3.22.2, 3.22.3, 3.22.4, 3.22.5, 3.22.6, 3.23.3, 3.24.0, 3.24.1, 3.24.1.1, 3.24.2, 3.24.3, 3.25.0, 3.25.2, 3.26.0, 3.26.1, 3.26.3, 3.26.4)
Collecting cmake
  Created temporary directory: /tmp/pip-unpack-1jsul8ly
  Starting new HTTPS connection (1): files.pythonhosted.org:443
  https://files.pythonhosted.org:443 "GET /packages/0d/41/85549e9645097cddc7279886eafeafc7462215f309133ea2eae4941f9c35/cmake-3.26.4-py2.py3-none-manylinux2014_aarch64.manylinux_2_17_aarch64.whl HTTP/1.1" 200 23346463
  Downloading cmake-3.26.4-py2.py3-none-manylinux2014_aarch64.manylinux_2_17_aarch64.whl (23.3 MB)
  Added cmake from https://files.pythonhosted.org/packages/0d/41/85549e9645097cddc7279886eafeafc7462215f309133ea2eae4941f9c35/cmake-3.26.4-py2.py3-none-manylinux2014_aarch64.manylinux_2_17_aarch64.whl#sha256=3175442985558d5415b97f264a6a1bb0af5ecfe10e3f7510257b1ea66bd33848 to build tracker '/tmp/pip-req-tracker-65ekpimn'
  Removed cmake from https://files.pythonhosted.org/packages/0d/41/85549e9645097cddc7279886eafeafc7462215f309133ea2eae4941f9c35/cmake-3.26.4-py2.py3-none-manylinux2014_aarch64.manylinux_2_17_aarch64.whl#sha256=3175442985558d5415b97f264a6a1bb0af5ecfe10e3f7510257b1ea66bd33848 from build tracker '/tmp/pip-req-tracker-65ekpimn'
Installing collected packages: cmake
  Created temporary directory: /tmp/pip-unpacked-wheel-73ayf5zt

  changing mode of /usr/local/bin/cmake to 755
  changing mode of /usr/local/bin/cpack to 755
  changing mode of /usr/local/bin/ctest to 755
Successfully installed cmake-3.26.4
Cleaning up...
Removed build tracker: '/tmp/pip-req-tracker-65ekpimn'
cmake version 3.26.4

CMake suite maintained and supported by Kitware (kitware.com/cmake).
/usr/local/bin/cmake

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
Package 'libpython3.9' is not installed, so not removed
Package 'libpython3.9-dbg' is not installed, so not removed
Package 'libpython3.9-dev' is not installed, so not removed
Package 'libpython3.9-testsuite' is not installed, so not removed
The following packages were automatically installed and are no longer required:
  libgtk2.0-0 libgtk2.0-common
[... cut ...]
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at fbd4968 Rebase from 'upstream'
The command '/bin/bash -c ./ros2_build.sh' returned a non-zero code: 1

dusty-nv, 2023-07-04:
Sorry @BeLeever, neither this or the pastebin has an error in it.  You could try putting the full log on google drive or somewhere, or search it for keywords like `error`.   

If you have problems capturing the entire log, try it like this:

```bash
scripts/docker_build_ros.sh --distro humble --package desktop 2>&1 | tee build_log.txt
```

zymouse, 2023-10-18:
I had the same problem, the ros2_build.sh script was executed in the build container, then I went into the container separately to execute it, the error was pip install flake8-import-order. you can tell it's a network problem, I solved it through a web proxy to this problem

closed 2025-02-21 by johnnynunez