dusty-nv/jetson-containers #261: [l4t-ml] Compatibility issues with Jetpack 4.6.4 (L4T 32.7.4)

LorisET, 2023-07-11:
Hi There!
I'm working on a Jetson Xavier AGX upgraded to Jetpack version 4.6.4 - L4T 32.7.4 (from 4.6.2).
After this update, all docker images based on nvcr.io/nvidia/l4t-ml:r32.7.1-py3 has stop working.
If I run directly a container instance with tag 32.7.1 and trying to import cv2 or torch from python console:

`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/cv2/__init__.py", line 96, in <module>
    bootstrap()
  File "/usr/local/lib/python3.6/dist-packages/cv2/__init__.py", line 86, in bootstrap
    import cv2
ImportError: libcublas.so.10: cannot open shared object file: No such file or directory
`

`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 196, in <module>
    _load_global_deps()
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 149, in _load_global_deps
    ctypes.CDLL(lib_path, mode=ctypes.RTLD_GLOBAL)
  File "/usr/lib/python3.6/ctypes/__init__.py", line 348, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libcurand.so.10: cannot open shared object file: No such file or directory
`

 It's strange because these errors disappears if I use nvcr.io/nvidia/l4t-ml:r34.1.1-py3 but, according to the docs:
 
> note: L4T R32.x containers can run on other versions of R32.x (e.g. R32.7.1 containers can run on R32.7.2)

What tag image should I use?
Thanks in advance

LorisET, 2023-07-12:
Hi @dusty-nv, thank you for the fast reply.
If I start `nvcr.io/nvidia/l4t-base:r32.7.1` container with --runtime nvidia I'm not able to run`python3 -c 'import tensorrt'`.
Command reports `ModuleNotFoundError: No module named 'tensorrt'`
Same thing with `nvcr.io/nvidia/l4t-base:r34.1.1` image.

dusty-nv, 2023-07-12:
@LorisET the r34.1.1 images are not meant to work on r32.x, but the r32.7.1 images should work across r32.7.x installations.  I'm looking into it more with my colleagues who build l4t-base container, but I'm unsure if it's just your system's environment or not.  I'm curious if it would work with a fresh system flashed with SDK Manager as opposed to upgraded...

closed 2025-02-21 by johnnynunez