dusty-nv/jetson-containers #264: Dockrfile for pytorch 2.0.x source builder (container)?

hillct, 2023-07-22:
I'm looking for a build/script/patch/Dockerfile enabling building from source, of pytorch2 with explicit sm_87 & Cuda 11.4 support as appears to have been built and included in nvcr.io/nvidia/l4t-pytorch:r35.2.1-pth2.0-py3 which indicates in part:
```
>>> torch.version.cuda
'11.4'
>>> torch.version.git_version
'0bd8be0232e78dc7b9c66c316ef39977c1c7c0ed'
```
The above commit hash does not exist in the official pytorch repo nor does it exist here https://github.com/dusty-nv/pytorch and presumably includes a patch to enable support of sm_87 where, if built from official sources, pytorch will fail to recognize `sm_87` and default back to `[ sm_72, sm_80, sm_86 ]`

I came across a patch for pytorch 1.x o add sm_87 but not for pytorch2.x. Does such a patch/built script/Dockerfile for source build exist?

I ask this because https://github.com/dusty-nv/jetson-containers/blob/master/Dockerfile.pytorch is just pulling from a whl and whil I'm no agains doing so, I have a requirement to utilize Python 3.9 so I find myself needing to rebuild rebuild from source.

Thanks.

dusty-nv, 2023-07-22:
@hillct you should run `export TORCH_CUDA_ARCH_LIST="7.2;8.7"` before compiling PyTorch - see the `Build from Source` section of this post for more info:  https://forums.developer.nvidia.com/t/pytorch-for-jetson/72048

Also apply this patch to the source:   https://gist.github.com/dusty-nv/ce51796085178e1f38e3c6a1663a93a1#file-pytorch-1-11-jetpack-5-0-patch

hillct, 2023-07-22:
Thanks for the quick response.

> @hillct you should run `export TORCH_CUDA_ARCH_LIST="7.2;8.7"` before compiling PyTorch - see the `Build from Source` section of this post for more info:  https://forums.developer.nvidia.com/t/pytorch-for-jetson/72048

I'd tried both decimal and non-decimal notation there. Neither seemed to cause it to be picked up correctly.

> Also apply this patch to the source:   https://gist.github.com/dusty-nv/ce51796085178e1f38e3c6a1663a93a1#file-pytorch-1-11-jetpack-5-0-patch

That patch seems to be for pyTorch 1.x. is there a corresponding patch for pyTorch 2.x? Or a support for compute 8.7 now built into the core pytorch 2.x codebase?

When I get back in front of a computer I'll post the docker file I'm trying to use as a pie torch builder