dusty-nv/jetson-containers #265: Build error while execute ./scripts/docker_build_ml.sh tensorflow in nVidia Jason Nano device

JackiePei61, 2023-07-26:
From error message, any command or instruction I need to perform?

Thanks for help.

ERROR: Command errored out with exit status 1: /usr/bin/python3 -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-8j3g4s5v/h5py/setup.py'"'"'; __file__='"'"'/tmp/pip-install-8j3g4s5v/h5py/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-yl2dyow0/install-record.txt --single-version-externally-managed --compile --install-headers /usr/local/include/python3.8/h5py Check the logs for full command output.

Error compiling Cython file:
    ------------------------------------------------------------
    ...
    # License:  Standard 3-clause BSD; see "license.txt" for full license terms
    #           and contributor agreement.

    from defs cimport *

    from _objects cimport class ObjectID
                          ^
    ------------------------------------------------------------

    h5py/h5t.pxd:12:22: Expected an identifier

    Error compiling Cython file:
    ------------------------------------------------------------
    ...
        Low-level type-conversion routines.
    """

    from h5 import get_config
    from h5r cimport Reference, RegionReference, hobj_ref_t, hdset_reg_ref_t
    from h5t cimport H5PY_OBJ, typewrap, py_create, TypeID
    ^
    ------------------------------------------------------------

    h5py/_conv.pyx:18:0: 'h5py/h5t/H5PY_OBJ.pxd' not found

    Error compiling Cython file:
    ------------------------------------------------------------
    ...
        Low-level type-conversion routines.
    """

    from h5 import get_config
    from h5r cimport Reference, RegionReference, hobj_ref_t, hdset_reg_ref_t
    from h5t cimport H5PY_OBJ, typewrap, py_create, TypeID
    ^
    ------------------------------------------------------------

    h5py/_conv.pyx:18:0: 'h5py/h5t/typewrap.pxd' not found

    Error compiling Cython file:
    ------------------------------------------------------------
    ...
        Low-level type-conversion routines.
    """

    from h5 import get_config
    from h5r cimport Reference, RegionReference, hobj_ref_t, hdset_reg_ref_t
    from h5t cimport H5PY_OBJ, typewrap, py_create, TypeID
    ^
    ------------------------------------------------------------

    h5py/_conv.pyx:18:0: 'h5py/h5t/py_create.pxd' not found

    Error compiling Cython file:
    ------------------------------------------------------------
    ...
        Low-level type-conversion routines.
    """

    from h5 import get_config
    from h5r cimport Reference, RegionReference, hobj_ref_t, hdset_reg_ref_t
[... 560 lines cut ...]
    File "/usr/lib/python3.8/distutils/dist.py", line 966, in run_commands
      self.run_command(cmd)
    File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/usr/lib/python3/dist-packages/wheel/bdist_wheel.py", line 223, in run
      self.run_command('build')
    File "/usr/lib/python3.8/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/usr/lib/python3.8/distutils/command/build.py", line 135, in run
      self.run_command(cmd_name)
    File "/usr/lib/python3.8/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/usr/lib/python3.8/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/tmp/pip-install-8j3g4s5v/h5py/setup_build.py", line 207, in run
      self.extensions = cythonize(self._make_extensions(config),
    File "/usr/local/lib/python3.8/dist-packages/Cython/Build/Dependencies.py", line 1134, in cythonize
      cythonize_one(*args)
    File "/usr/local/lib/python3.8/dist-packages/Cython/Build/Dependencies.py", line 1301, in cythonize_one
      raise CompileError(None, pyx_file)
  Cython.Compiler.Errors.CompileError: /tmp/pip-install-8j3g4s5v/h5py/h5py/_conv.pyx
  Building wheel for h5py (setup.py): finished with status 'error'
  ERROR: Failed building wheel for h5py

CarkusL, 2023-08-14:
same question

dusty-nv, 2023-08-14:
@CarkusL @JackiePei61 try `pip3 install 'Cython<3'` in the container first

These dockerfiles/ect have been refactored are are passing builds on JetPack 4.6 / JetPack 5.1

https://github.com/dusty-nv/jetson-containers/tree/master/packages/l4t/l4t-tensorflow

closed 2025-02-21 by johnnynunez