dusty-nv/jetson-containers #266: OSError: libcurand.so.10: cannot open shared object file: No such file or directory on build.

watersounds, 2023-07-27:
### I got a error :"OSError: libcurand.so.10: cannot open shared object file: No such file or directory" when i build l4t-pyotch.
Here are the details：
When I execute `scripts/docker_build_ml.sh pytorch`, when building to 
https://github.com/dusty-nv/jetson-containers/blob/74afe69b91ed8256e1cd344a6c52301a36e631bc/Dockerfile.pytorch#L58
, the error is reported.

<details>
<summary> OSError: libcurand.so.10: cannot open shared object file: No such file or directory </summary>

```
 ---> Running in d6dbcce55f3d
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 196, in <module>
    _load_global_deps()
  File "/usr/local/lib/python3.6/dist-packages/torch/__init__.py", line 149, in _load_global_deps
    ctypes.CDLL(lib_path, mode=ctypes.RTLD_GLOBAL)
  File "/usr/lib/python3.6/ctypes/__init__.py", line 348, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libcurand.so.10: cannot open shared object file: No such file or directory
The command '/bin/sh -c python3 -c 'import torch; print(f"PyTorch version: {torch.__version__}"); print(f"CUDA available:  {torch.cuda.is_available()}"); print(f"cuDNN version:   {torch.backends.cudnn.version()}"); print(torch.__config__.show());'' returned a non-zero code: 1
```

</details>

And after I commented this line of code, this error appeared elsewhere like(I remove torchvision&torchaudio)
https://github.com/dusty-nv/jetson-containers/blob/74afe69b91ed8256e1cd344a6c52301a36e631bc/scripts/opencv_install.sh#L79

My Jetson NX install Jetpack 4.6.1 (L4T 32.7.1).
Here are some message about my env.
```
ls -ll /usr/local/cuda/lib64/libcurand*
lrwxrwxrwx 1 root root       15 3月   2  2021 /usr/local/cuda/lib64/libcurand.so -> libcurand.so.10
lrwxrwxrwx 1 root root       23 3月   2  2021 /usr/local/cuda/lib64/libcurand.so.10 -> libcurand.so.10.1.2.300
-rw-r--r-- 1 root root 62698584 3月   2  2021 /usr/local/cuda/lib64/libcurand.so.10.1.2.300
-rw-r--r-- 1 root root 62767380 3月   2  2021 /usr/local/cuda/lib64/libcurand_static.a
```
```
ls -ll /etc/nvidia-container-runtime/host-files-for-container.d/
-rw-r--r-- 1 root root    26 5月  24  2021 cuda.csv
-rw-r--r-- 1 root root  4250 7月  14  2021 cudnn.csv
-rw-r--r-- 1 root root 12240 2月  20  2022 l4t.csv
-rw-r--r-- 1 root root  1590 1月  14  2022 tensorrt.csv
-rw-r--r-- 1 root root   325 8月  11  2020 visionworks.csv
```

### I built another container based on lines 1-58 of `Dockerfile.pytorch` and manually executed the following commands, and this error did not appear.

<details>
<summary> Here are some problems I encountered but solved, you can also check it out if you have time </summary>

- when install numpy: `RuntimeError: Running cythonize failed!`

```
    subprocess.CalledProcessError: Command '['/usr/bin/python3', '-m', 'cython', '-3', '--fast-fail', '-o', '_pcg64.c', '_pcg64.pyx']' returned non-zero exit status 1.
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-pvphsu7t/numpy/setup.py", line 508, in <module>
        setup_package()
      File "/tmp/pip-build-pvphsu7t/numpy/setup.py", line 488, in setup_package
        generate_cython()
      File "/tmp/pip-build-pvphsu7t/numpy/setup.py", line 285, in generate_cython
        raise RuntimeError("Running cythonize failed!")
    RuntimeError: Running cythonize failed!
```
To solve this problem, I manually specified the version of Cpython as 0.29.36.

- when install CMake:`ModuleNotFoundError: No module named 'skbuild' `
```
Running setup.py (path:/tmp/pip-build-qeu9hs36/cmake/setup.py) egg_info for package cmake
    Running command python setup.py egg_info
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-qeu9hs36/cmake/setup.py", line 8, in <module>
        from skbuild import setup
    ModuleNotFoundError: No module named 'skbuild'
```
To fix this, I upgraded pip before installing CMake. `RUN pip3 install --upgrade pip`

</details>

dusty-nv, 2023-07-27:
> OSError: libcurand.so.10: cannot open shared object file: No such file or directory

Hi @watersounds, did you have your default docker runtime set to `nvidia` like here:

https://github.com/dusty-nv/jetson-containers#docker-default-runtime

watersounds, 2023-07-28:
@dusty-nv ,thank you for the reply. I missed that and it really works.
I will close this issue.

closed 2023-07-28 by watersounds