dusty-nv/jetson-containers #268: CMake error on dustynv/ros:melodic-ros-base-l4t-r32.7.1

amir-zviran, 2023-08-03:
Hi,

I've been using this docker image as a base for my docker:
dustynv/ros:melodic-ros-base-l4t-r32.7.1

Following the release of the new image you published a few days ago, my build started to fail while running catkin_make. This is the error I see on the logs:
```
Traceback (most recent call last):
   File "/usr/local/bin/cmake", line 5, in <module>
     from cmake import cmake
ModuleNotFoundError: No module named 'cmake'
Invoking "cmake" failed
```

Any idea why it happens? can you tell what has changed there from the previous image you've published about 17 months ago?

Thanks!

dusty-nv, 2023-08-03:
Hi @amir-zviran, sorry about that - I had been working to unify my ROS/ROS2 builds and didn't capture this in the tests that catkin_make was failing with melodic.   My guess is that the python 2.7-based melodic doesn't like the way I had installed cmake with pip.   Instead, I reverted melodic to install cmake from apt like previously.

Can you try pulling `dustynv/ros:melodic-ros-base-l4t-r32.7.1` again and see if it works for you?

amir-zviran, 2023-08-06:
It works now! Thanks Dustin!

closed 2025-02-21 by johnnynunez