dusty-nv/jetson-containers #269: ZED support

Myzhar, 2023-08-04:
Hi @dusty-nv :)
I suggest you add this optional volume in the [README of the ZED Package](https://github.com/dusty-nv/jetson-containers/tree/master/packages/zed):
`-v ${HOME}/zed_docker_ai/:/usr/local/zed/resources/`
this allows you to not download and optimize the AI models used in the container each it's started.

You could also add this volume to not be obliged to download the camera calibration at each run (mostly if the host is not connected to the internet):
`-v /usr/local/zed/settings/:/usr/local/zed/settings/`

Myzhar, 2023-08-04:
The SDK path is fixed, but you can change the bond folder on the host:

`-v <whatever_you_want>:/usr/local/zed/resources/`
`-v <whatever_you_want>:/usr/local/zed/settings/`

dusty-nv, 2023-08-04:
OK gotcha, thanks Walter.  BTW ya'll have new JetPack out to build ZED SDK for ;)   

https://forums.developer.nvidia.com/t/jetpack-5-1-2-l4t-35-4-1-is-now-live/262030