dusty-nv/jetson-containers #271: ROS2 Foxy + Zed container for Ubuntu 18.04 (L4T 32.6.1)

abejeyapratap, 2023-08-17:
I see that there are various ROS2 foxy containers and Zed containers available. However, it seems that the Zed containers only have the Zed Python API (no Zed tools, ROS2 foxy, ros2-wrapper). 

I'm looking for a docker container that has ROS2 foxy as well as all 3 Zed components (SDK with tools, Python API, ros2-wrapper). Is there any support for this?

dusty-nv, 2023-08-18:
Hi @abejeyapratap, I just staged commit https://github.com/dusty-nv/jetson-containers/commit/cb532a6052f40571f9cdce1fb1c30e032799d93c in the dev branch to have the ZED tools installed.  I have not tried the ZED ros2-wrapper but will add it to my list to give a look.   I think Stereolabs has their own dockerfiles / containers as well?

abejeyapratap, 2023-08-22:
@dusty-nv, great thanks! Yes, I did see they have Docker containers too, but they are for ROS2 Humble and also don't have all 3 components (SDK with tools, Python API, wrapper) together. But since I'm having issues integrating their ros2-wrapper into the current container, I was exploring other options (such as a Docker image that combines all 3 components).

If you do get around to it, let me know - that would definitely be super helpful!

abejeyapratap, 2023-08-23:
Hi @dusty-nv, I just found out that L4T 32.6.1 only supports bionic (Ubuntu 18.04). I believe this means it isn't possible to have both ZED SDK/python API and ros2-wrapper working on our Jetson Nano. Just FYI. I will now be switching to a different  Jetson board that supports L4T 35.4.1.

closed 2023-09-01 by abejeyapratap