dusty-nv/jetson-containers #272: Question about difference of JP46 and JP51 for ROS2 Images

XujqGitHub, 2023-08-18:
Hi,

I noticed that for the same version of ROS2 image, the size of the JP46 and JP51 images are very different. Such as **dustynv/ros:humble-ros-base-l4t-r32.7.1 (2023-08-04, 0.7GB), dustynv/ros:humble-ros-base-l4t-r35.3.1 (2023-08-01, 5.2GB)**

(1) What is the reason for this? Is it because the types of software included are different? 
(2) I also have another question to ask you: If my host machine (agx xavier devkit) runs at r32.7.1, can I run dustynv/ros:humble-ros-base-l4t-r35.3.1 in docker?

dusty-nv, 2023-08-18:
Hi @XujqGitHub, it's because on JetPack 4 the CUDA/cuDNN/TensorRT libraries get mounted into the container, whereas on JetPack 5 the CUDA/cuDNN/TensorRT libraries are installed inside the containers.   JetPack 4 containers cannot be run on JetPack 5 (and vice-versa)

XujqGitHub, 2023-08-18:
Thank you for your reply. You have perfectly answered my question.

closed 2023-08-18 by XujqGitHub