dusty-nv/jetson-containers #274: try to build built container images failed

UserName-wang, 2023-08-24:
Hi there,

my environment is: Jetson AGX xavier running Jetpack 5.1.1, and L4T version R35.3.1,

command: ./build.sh --name=my_container ros:humble-desktop
/usr/lib/python3/dist-packages/requests/__init__.py:89: RequestsDependencyWarning: urllib3 (1.26.16) or chardet (3.0.4) doesn't match a supported version!
  warnings.warn("urllib3 ({}) or chardet ({}) doesn't match a supported "
Namespace(base='', build_flags='', list_packages=False, logs='', multiple=False, name='my_container', package_dirs=[''], packages=['ros:humble-desktop'], push='', show_packages=False, simulate=False, skip_errors=False, skip_packages=[''], skip_tests=[''])
-- L4T_VERSION=35.3.1
-- JETPACK_VERSION=5.1.1
-- CUDA_VERSION=11.4.315
-- LSB_RELEASE=20.04 (focal)
-- Loading /home/agx/study/nvidia/jetson-containers/packages/protobuf/protobuf_cpp/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/tensorflow/config.py
-- Package small-stable-diffusion was disabled by its config
-- Loading /home/agx/study/nvidia/jetson-containers/packages/l4t/l4t-text-generation/config.yml
-- Loading /home/agx/study/nvidia/jetson-containers/packages/l4t/l4t-pytorch/l4t-pytorch.json
-- Loading /home/agx/study/nvidia/jetson-containers/packages/l4t/l4t-tensorflow/l4t-tensorflow.json
-- Loading /home/agx/study/nvidia/jetson-containers/packages/l4t/l4t-diffusion/config.yml
-- Loading /home/agx/study/nvidia/jetson-containers/packages/tritonserver/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/pytorch/torchaudio/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/pytorch/torchvision/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/pytorch/torch2trt/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/pytorch/torch_tensorrt/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/pytorch/config.py
-- Package pytorch:1.10 isn't compatible with L4T r35.3.1 (requires L4T ==32.*)
-- Package pytorch:1.10 was disabled by its config
-- Package pytorch:1.9 isn't compatible with L4T r35.3.1 (requires L4T ==32.*)
-- Package pytorch:1.9 was disabled by its config
-- Loading /home/agx/study/nvidia/jetson-containers/packages/riva-client/config.json
-- Loading /home/agx/study/nvidia/jetson-containers/packages/deepstream/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/rapids/cuml/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/rapids/cudf/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/nemo/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/zed/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/opencv/opencv_builder/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/opencv/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/cuda-python/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/llm/awq/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/llm/text-generation-inference/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/llm/xformers/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/llm/exllama/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/llm/llama_cpp/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/llm/optimum/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/llm/auto_gptq/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/llm/transformers/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/ros/config.py
-- Package ros:melodic-ros-base isn't compatible with L4T r35.3.1 (requires L4T <34)
-- Package ros:melodic-ros-base was disabled by its config
-- Package ros:melodic-ros-core isn't compatible with L4T r35.3.1 (requires L4T <34)
-- Package ros:melodic-ros-core was disabled by its config
-- Package ros:melodic-desktop isn't compatible with L4T r35.3.1 (requires L4T <34)
-- Package ros:melodic-desktop was disabled by its config
-- Loading /home/agx/study/nvidia/jetson-containers/packages/onnxruntime/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/cupy/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/pycuda/config.py
-- Loading /home/agx/study/nvidia/jetson-containers/packages/onnx/config.py
-- Building containers  ['build-essential', 'python', 'cmake', 'numpy', 'opencv', 'ros:humble-desktop']
-- Building container my_container:l4t-r35.3.1-build-essential

sudo docker build --network=host --tag my_container:l4t-r35.3.1-build-essential \
--file /home/agx/study/nvidia/jetson-containers/packages/build-essential/Dockerfile \
--build-arg BASE_IMAGE=nvcr.io/nvidia/l4t-jetpack:r35.3.1 \
/home/agx/study/nvidia/jetson-containers/packages/build-essential \
2>&1 | tee /home/agx/study/nvidia/jetson-containers/logs/20230824_201519/build/my_container_l4t-r35.3.1-build-essential.txt; exit ${PIPESTATUS[0]}

#0 building with "default" instance using docker driver

#1 [internal] load .dockerignore
#1 transferring context: 2B 0.0s done
#1 DONE 0.0s

#2 [internal] load build definition from Dockerfile
#2 transferring dockerfile: 595B done
#2 DONE 0.0s

#3 [auth] nvidia/l4t-jetpack:pull,push token for nvcr.io
#3 DONE 0.0s

#4 [internal] load metadata for nvcr.io/nvidia/l4t-jetpack:r35.3.1
#4 ERROR: failed to authorize: failed to fetch oauth token: unexpected status from GET request to https://nvcr.io/proxy_auth?scope=repository%3Anvidia%2Fl4t-jetpack%3Apull%2Cpush: 401 
------
 > [internal] load metadata for nvcr.io/nvidia/l4t-jetpack:r35.3.1:
------
Dockerfile:7
--------------------
   5 |     #---
   6 |     ARG BASE_IMAGE
   7 | >>> FROM ${BASE_IMAGE}
   8 |     
   9 |     ENV DEBIAN_FRONTEND=noninteractive
--------------------
ERROR: failed to solve: nvcr.io/nvidia/l4t-jetpack:r35.3.1: failed to authorize: failed to fetch oauth token: unexpected status from GET request to https://nvcr.io/proxy_auth?scope=repository%3Anvidia%2Fl4t-jetpack%3Apull%2Cpush: 401 
Traceback (most recent call last):
 File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/agx/study/nvidia/jetson-containers/jetson_containers/build.py", line 93, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.simulate, args.skip_tests, args.push)
  File "/home/agx/study/nvidia/jetson-containers/jetson_containers/container.py", line 119, in build_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)  
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'sudo docker build --network=host --tag my_container:l4t-r35.3.1-build-essential --file /home/agx/study/nvidia/jetson-containers/packages/build-essential/Dockerfile --build-arg BASE_IMAGE=nvcr.io/nvidia/l4t-jetpack:r35.3.1 /home/agx/study/nvidia/jetson-containers/packages/build-essential 2>&1 | tee /home/agx/study/nvidia/jetson-containers/logs/20230824_201519/build/my_container_l4t-r35.3.1-build-essential.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.

Can someone please help me? thank you!

UserName-wang, 2023-08-24:
> @UserName-wang hmm, can you try doing a `sudo docker pull nvcr.io/nvidia/l4t-jetpack:r35.3.1` first?
it woks now!
Thank you a lot for your help and quick reply!

UserName-wang, 2023-08-25:
Dear @dusty-nv，

Now I have a new error about bitsandbytes:

#9 [5/5] RUN pip3 show bitsandbytes && python3 -c 'import bitsandbytes'
#9 1.970 Name: bitsandbytes
#9 1.971 Version: 0.39.1
#9 1.972 Summary: k-bit optimizers and matrix multiplication routines.
#9 1.972 Home-page: https://github.com/TimDettmers/bitsandbytes
#9 1.973 Author: Tim Dettmers
#9 1.974 Author-email: dettmers@cs.washington.edu
#9 1.974 License: MIT
#9 1.975 Location: /usr/local/lib/python3.8/dist-packages
#9 1.976 Requires: 
#9 1.976 Required-by: 
#9 8.551 /usr/local/lib/python3.8/dist-packages/bitsandbytes/cextension.py:34: UserWarning: The installed version of bitsandbytes was compiled without GPU support. 8-bit optimizers, 8-bit multiplication, and GPU quantization are unavailable.
#9 8.551   warn("The installed version of bitsandbytes was compiled without GPU support. "
#9 8.552 
#9 8.552 ===================================BUG REPORT===================================
#9 8.552 Welcome to bitsandbytes. For bug reports, please run
#9 8.552 
#9 8.552 python -m bitsandbytes
#9 8.552 
#9 8.552  and submit this information together with your error trace to: https://github.com/TimDettmers/bitsandbytes/issues
#9 8.552 ================================================================================
#9 8.552 bin /usr/local/lib/python3.8/dist-packages/bitsandbytes/libbitsandbytes_cpu.so
#9 8.552 False
#9 8.552 'NoneType' object has no attribute 'cadam32bit_grad_fp32'
#9 8.552 CUDA SETUP: Required library version not found: libbitsandbytes_cpu.so. Maybe you need to compile it from source?
#9 8.552 CUDA SETUP: Defaulting to libbitsandbytes_cpu.so...
#9 8.552 
#9 8.552 ================================================ERROR=====================================
#9 8.552 CUDA SETUP: CUDA detection failed! Possible reasons:
#9 8.552 1. CUDA driver not installed
#9 8.552 2. CUDA not installed
#9 8.552 3. You have multiple conflicting CUDA libraries
#9 8.552 4. Required library not pre-compiled for this bitsandbytes release!
#9 8.552 CUDA SETUP: If you compiled from source, try again with `make CUDA_VERSION=DETECTED_CUDA_VERSION` for example, `make CUDA_VERSION=113`.
#9 8.552 CUDA SETUP: The CUDA version for the compile might depend on your conda install. Inspect CUDA version via `conda list | grep cuda`.
#9 8.552 ================================================================================
#9 8.552 
#9 8.552 CUDA SETUP: Problem: The main issue seems to be that the main CUDA library was not detected.
#9 8.552 CUDA SETUP: Solution 1): Your paths are probably not up-to-date. You can update them via: sudo ldconfig.
#9 8.552 CUDA SETUP: Solution 2): If you do not have sudo rights, you can do the following:
#9 8.552 CUDA SETUP: Solution 2a): Find the cuda library via: find / -name libcuda.so 2>/dev/null
#9 8.552 CUDA SETUP: Solution 2b): Once the library is found add it to the LD_LIBRARY_PATH: export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:FOUND_PATH_FROM_2a
#9 8.552 CUDA SETUP: Solution 2c): For a permanent solution add the export from 2b into your .bashrc file, located at ~/.bashrc
#9 8.552 CUDA SETUP: Setup Failed!
#9 DONE 9.6s

#10 exporting to image
#10 exporting layers
#10 exporting layers 0.9s done
#10 writing image sha256:d590ef0eef5cf993ac8bf5d21f58a8e2c54ae40df4f53309e9b6eb69c7986638 done
#10 naming to docker.io/library/my_container:r35.3.1-bitsandbytes done
#10 DONE 0.9s
-- Testing container my_container:r35.3.1-bitsandbytes (bitsandbytes/test.py)

sudo docker run -t --rm --runtime=nvidia --network=host \
--volume /home/agx/study/nvidia/jetson-containers/packages/llm/bitsandbytes:/test \
--volume /home/agx/study/nvidia/jetson-containers/data:/data \
--workdir /test \
my_container:r35.3.1-bitsandbytes \
/bin/bash -c 'python3 test.py' \
2>&1 | tee /home/agx/study/nvidia/jetson-containers/logs/20230825_174248/test/my_container_r35.3.1-bitsandbytes_test.py.txt; exit ${PIPESTATUS[0]}

testing bitsandbytes...

===================================BUG REPORT===================================
Welcome to bitsandbytes. For bug reports, please run

python -m bitsandbytes

 and submit this information together with your error trace to: https://github.com/TimDettmers/bitsandbytes/issues
================================================================================
bin /usr/local/lib/python3.8/dist-packages/bitsandbytes/libbitsandbytes_cuda114_nocublaslt.so
False
CUDA SETUP: CUDA runtime path found: /usr/local/cuda/lib64/libcudart.so.11.0
CUDA SETUP: Highest compute capability among GPUs detected: 7.2
CUDA SETUP: Detected CUDA version 114
/usr/local/lib/python3.8/dist-packages/bitsandbytes/cuda_setup/main.py:149: UserWarning: WARNING: Compute capability < 7.5 detected! Only slow 8-bit matmul is supported for your GPU!
  warn(msg)
CUDA SETUP: Required library version not found: libbitsandbytes_cuda114_nocublaslt.so. Maybe you need to compile it from source?
CUDA SETUP: Defaulting to libbitsandbytes_cpu.so...

================================================ERROR=====================================
CUDA SETUP: CUDA detection failed! Possible reasons:
1. CUDA driver not installed
2. CUDA not installed
3. You have multiple conflicting CUDA libraries
4. Required library not pre-compiled for this bitsandbytes release!
CUDA SETUP: If you compiled from source, try again with `make CUDA_VERSION=DETECTED_CUDA_VERSION` for example, `make CUDA_VERSION=113`.
CUDA SETUP: The CUDA version for the compile might depend on your conda install. Inspect CUDA version via `conda list | grep cuda`.
================================================================================

dusty-nv, 2023-08-25:
@UserName-wang can you post the command you ran to start the build, and the entire build log?

UserName-wang, 2023-08-25:
Dear @dusty-nv ,
command I used:  ./build.sh --name=my_container pytorch transformers nemo
#1 DONE 0.0s

#2 [internal] load build definition from Dockerfile
#2 transferring dockerfile: 512B done
#2 DONE 0.0s

#3 [internal] load metadata for docker.io/library/my_container:r35.3.1-cmake
#3 DONE 0.0s

#4 [1/3] FROM docker.io/library/my_container:r35.3.1-cmake
#4 DONE 0.0s

#5 [2/3] RUN pip3 install --no-cache-dir --verbose git+https://github.com/onnx/onnx@main
#5 CACHED

#6 [3/3] RUN pip3 show onnx && python3 -c 'import onnx; print(onnx.__version__)'
#6 CACHED

#7 exporting to image
#7 exporting layers done
#7 writing image sha256:e0150313e6a46d8ebcf12e3df6b664c7fd0a3f4b5f2b4b0c55d39e1c37e058c9 done
#7 naming to docker.io/library/my_container:r35.3.1-onnx done
#7 DONE 0.0s
-- Testing container my_container:r35.3.1-onnx (onnx/test.py)

sudo docker run -t --rm --runtime=nvidia --network=host \
--volume /home/agx/study/nvidia/jetson-containers/packages/onnx:/test \
--volume /home/agx/study/nvidia/jetson-containers/data:/data \
--workdir /test \
my_container:r35.3.1-onnx \
/bin/bash -c 'python3 test.py' \
2>&1 | tee /home/agx/study/nvidia/jetson-containers/logs/20230825_221528/test/my_container_r35.3.1-onnx_test.py.txt; exit ${PIPESTATUS[0]}

testing onnx...
onnx version: 1.15.0
onnx OK

-- Building container my_container:r35.3.1-pytorch

sudo docker build --network=host --tag my_container:r35.3.1-pytorch \
--file /home/agx/study/nvidia/jetson-containers/packages/pytorch/Dockerfile \
--build-arg BASE_IMAGE=my_container:r35.3.1-onnx \
--build-arg PYTORCH_WHL="torch-2.0.0+nv23.05-cp38-cp38-linux_aarch64.whl" \
--build-arg PYTORCH_URL="https://nvidia.box.com/shared/static/i8pukc49h3lhak4kkn67tg9j4goqm0m7.whl" \
/home/agx/study/nvidia/jetson-containers/packages/pytorch \
2>&1 | tee /home/agx/study/nvidia/jetson-containers/logs/20230825_221528/build/my_container_r35.3.1-pytorch.txt; exit ${PIPESTATUS[0]}

#0 building with "default" instance using docker driver

#1 [internal] load build definition from Dockerfile
#1 transferring dockerfile: 1.87kB 0.0s done
#1 DONE 0.0s

#2 [internal] load .dockerignore
#2 transferring context: 2B 0.0s done
#2 DONE 0.0s

#3 [internal] load metadata for docker.io/library/my_container:r35.3.1-onnx
#3 DONE 0.0s

#4 [1/6] FROM docker.io/library/my_container:r35.3.1-onnx
#4 DONE 0.0s

#5 [2/6] RUN apt-get update &&     apt-get install -y --no-install-recommends 		  libopenblas-dev 		  libopenmpi-dev             openmpi-bin             openmpi-common 		  gfortran 		  libomp-dev     && rm -rf /var/lib/apt/lists/*     && apt-get clean
#5 CACHED

#6 [5/6] RUN PYTHON_ROOT=`pip3 show torch | grep Location: | cut -d' ' -f2` &&     TORCH_CMAKE_CONFIG=$PYTHON_ROOT/torch/share/cmake/Torch/TorchConfig.cmake &&     echo "patching _GLIBCXX_USE_CXX11_ABI in ${TORCH_CMAKE_CONFIG}" &&     sed -i 's/  set(TORCH_CXX_FLAGS "-D_GLIBCXX_USE_CXX11_ABI=")/  set(TORCH_CXX_FLAGS "-D_GLIBCXX_USE_CXX11_ABI=0")/g' ${TORCH_CMAKE_CONFIG}
#6 CACHED
[... 382 lines cut ...]
    from bitsandbytes.cextension import COMPILED_WITH_CUDA
  File "/usr/local/lib/python3.8/dist-packages/bitsandbytes/cextension.py", line 20, in <module>
    raise RuntimeError('''
RuntimeError: 
        CUDA Setup failed despite GPU being available. Please run the following command to get more information:

        python -m bitsandbytes

        Inspect the output of the command and see if you can locate CUDA libraries. You might need to add them
        to your LD_LIBRARY_PATH. If you suspect a bug, please take the information from python -m bitsandbytes
        and open an issue at: https://github.com/TimDettmers/bitsandbytes/issues
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/agx/study/nvidia/jetson-containers/jetson_containers/build.py", line 93, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.simulate, args.skip_tests, args.push)
  File "/home/agx/study/nvidia/jetson-containers/jetson_containers/container.py", line 125, in build_container
    test_container(container_name, pkg, simulate)
  File "/home/agx/study/nvidia/jetson-containers/jetson_containers/container.py", line 295, in test_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'sudo docker run -t --rm --runtime=nvidia --network=host --volume /home/agx/study/nvidia/jetson-containers/packages/llm/bitsandbytes:/test --volume /home/agx/study/nvidia/jetson-containers/data:/data --workdir /test my_container:r35.3.1-bitsandbytes /bin/bash -c 'python3 test.py' 2>&1 | tee /home/agx/study/nvidia/jetson-containers/logs/20230825_221528/test/my_container_r35.3.1-bitsandbytes_test.py.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.

dusty-nv, 2023-08-25:
@UserName-wang unfortunately that log doesn't capture the actual build, because bitsandbytes was already built and it's just showing the cached output.  My guess is that it didn't find CUDA or something for some reason.   You could try `./build.sh bitsandbytes` to try building/testing just bitsandbytes (paste the log from that here).   Or you can run build.sh with `--build-flags='--no-cache'`

Also, your docker output looks different than mine - are you using buildkit or something?  This is what my `sudo docker version` shows:

```bash
sudo docker version
Client:
 Version:           20.10.21
 API version:       1.41
 Go version:        go1.18.1
 Git commit:        20.10.21-0ubuntu1~20.04.2
 Built:             Thu Apr 27 05:56:44 2023
 OS/Arch:           linux/arm64
 Context:           default
 Experimental:      true

Server:
 Engine:
  Version:          20.10.21
  API version:      1.41 (minimum version 1.12)
  Go version:       go1.18.1
  Git commit:       20.10.21-0ubuntu1~20.04.2
  Built:            Thu Apr 27 05:37:01 2023
  OS/Arch:          linux/arm64
  Experimental:     false
 containerd:
  Version:          1.6.12-0ubuntu1~20.04.3
  GitCommit:
 nvidia:
  Version:          1.1.4-0ubuntu1~20.04.3
  GitCommit:        629a689
 docker-init:
  Version:          0.19.0
  GitCommit:
```

UserName-wang, 2023-08-26:
@dusty-nv ,
below are my docker information, I guess it's automatically upgraded from your version.
I tried commands:  ./build.sh bitsandbytes and  ./build.sh --build-flags='--no-cache' , almost the same error. but I cannot upload all of the log information. because the log content are too long.

Client: Docker Engine - Community
 Version:           24.0.5
 API version:       1.43
 Go version:        go1.20.6
 Git commit:        ced0996
 Built:             Fri Jul 21 20:35:47 2023
 OS/Arch:           linux/arm64
 Context:           default

Server: Docker Engine - Community
 Engine:
  Version:          24.0.5
  API version:      1.43 (minimum version 1.12)
  Go version:       go1.20.6
  Git commit:       a61e2b4
  Built:            Fri Jul 21 20:35:47 2023
  OS/Arch:          linux/arm64
  Experimental:     false
 containerd:
  Version:          1.6.22
  GitCommit:        8165feabfdfe38c65b599c4993d227328c231fca
 nvidia:
  Version:          1.1.8
  GitCommit:        v1.1.8-0-g82f18fe
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0

#9 [5/5] RUN pip3 show bitsandbytes && python3 -c 'import bitsandbytes'
#9 1.403 Name: bitsandbytes
#9 1.404 Version: 0.39.1
#9 1.404 Summary: k-bit optimizers and matrix multiplication routines.
#9 1.405 Home-page: https://github.com/TimDettmers/bitsandbytes
#9 1.406 Author: Tim Dettmers
#9 1.407 Author-email: dettmers@cs.washington.edu
#9 1.407 License: MIT
#9 1.408 Location: /usr/local/lib/python3.8/dist-packages
#9 1.409 Requires: 
#9 1.409 Required-by: 
#9 6.840 
#9 6.840 ===================================BUG REPORT===================================
#9 6.840 Welcome to bitsandbytes. For bug reports, please run
#9 6.840 
#9 6.840 python -m bitsandbytes
#9 6.840 
#9 6.840  and submit this information together with your error trace to: https://github.com/TimDettmers/bitsandbytes/issues
#9 6.840 ================================================================================
#9 6.840 bin /usr/local/lib/python3.8/dist-packages/bitsandbytes/libbitsandbytes_cpu.so
#9 6.840 False
#9 6.840 'NoneType' object has no attribute 'cadam32bit_grad_fp32'
#9 6.840 CUDA SETUP: Required library version not found: libbitsandbytes_cpu.so. Maybe you need to compile it from source?
#9 6.840 CUDA SETUP: Defaulting to libbitsandbytes_cpu.so...
#9 6.840 
#9 6.840 ================================================ERROR=====================================
#9 6.840 CUDA SETUP: CUDA detection failed! Possible reasons:
#9 6.840 1. CUDA driver not installed
#9 6.840 2. CUDA not installed
#9 6.840 3. You have multiple conflicting CUDA libraries
#9 6.840 4. Required library not pre-compiled for this bitsandbytes release!
#9 6.840 CUDA SETUP: If you compiled from source, try again with `make CUDA_VERSION=DETECTED_CUDA_VERSION` for example, `make CUDA_VERSION=113`.
#9 6.840 CUDA SETUP: The CUDA version for the compile might depend on your conda install. Inspect CUDA version via `conda list | grep cuda`.
#9 6.840 ================================================================================
#9 6.840 
#9 6.840 CUDA SETUP: Problem: The main issue seems to be that the main CUDA library was not detected.
#9 6.840 CUDA SETUP: Solution 1): Your paths are probably not up-to-date. You can update them via: sudo ldconfig.
#9 6.840 CUDA SETUP: Solution 2): If you do not have sudo rights, you can do the following:
[... 71 lines cut ...]
    from bitsandbytes.cextension import COMPILED_WITH_CUDA
  File "/usr/local/lib/python3.8/dist-packages/bitsandbytes/cextension.py", line 20, in <module>
    raise RuntimeError('''
RuntimeError: 
        CUDA Setup failed despite GPU being available. Please run the following command to get more information:

        python -m bitsandbytes

        Inspect the output of the command and see if you can locate CUDA libraries. You might need to add them
        to your LD_LIBRARY_PATH. If you suspect a bug, please take the information from python -m bitsandbytes
        and open an issue at: https://github.com/TimDettmers/bitsandbytes/issues
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/agx/study/nvidia/jetson-containers/jetson_containers/build.py", line 93, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.simulate, args.skip_tests, args.push)
  File "/home/agx/study/nvidia/jetson-containers/jetson_containers/container.py", line 125, in build_container
    test_container(container_name, pkg, simulate)
  File "/home/agx/study/nvidia/jetson-containers/jetson_containers/container.py", line 295, in test_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'sudo docker run -t --rm --runtime=nvidia --network=host --volume /home/agx/study/nvidia/jetson-containers/packages/llm/bitsandbytes:/test --volume /home/agx/study/nvidia/jetson-containers/data:/data --workdir /test bitsandbytes:r35.3.1-bitsandbytes /bin/bash -c 'python3 test.py' 2>&1 | tee /home/agx/study/nvidia/jetson-containers/logs/20230826_141319/test/bitsandbytes_r35.3.1-bitsandbytes_test.py.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.

dusty-nv, 2023-08-26:
@UserName-wang run `./build.sh bitsandbytes --build-flags='--no-cache' | tee logs/bitsandbytes.txt` and then attach the log file here.  I can't tell why it doesn't build with CUDA without the full log.  

Have you set your default docker-runtime to `nvidia` like here?  https://github.com/dusty-nv/jetson-containers/blob/master/docs/setup.md#docker-default-runtime

UserName-wang, 2023-08-27:
Dear @dusty-nv , thank you for your patience!
Here is my /etc/docker/daemon.json:
{
    "runtimes": {
        "nvidia": {
            "path": "nvidia-container-runtime",
            "runtimeArgs": []
        }
    },
    "default-runtime": "nvidia",
    "insecure-registries": ["nas:5555"]
}

and here is the error log file after command: ./build.sh bitsandbytes --build-flags='--no-cache' | tee logs/bitsandbytes.txt

[bitsandbytes.txt](https://github.com/dusty-nv/jetson-containers/files/12447048/bitsandbytes.txt)

closed 2025-02-21 by johnnynunez