dusty-nv/jetson-containers #275: NeMo doesn't support Jetson.

UserName-wang, 2023-08-28:
just let you know that NeMo doesn't support Jetson. https://github.com/NVIDIA/NeMo/issues/7331

but what's the docker image used for? [dustynv/nemo:r35.3.1](https://hub.docker.com/r/dustynv/nemo/tags)

dusty-nv, 2023-08-28:
@UserName-wang you are correct that Nemo doesn't officially support Jetson - however I was able to get it building with some tweaks and am able to use it for some things.   The stuff in Nemo that requires torch.distributed doesn't work (like megatron)

UserName-wang, 2023-08-28:
> @UserName-wang you are correct that Nemo doesn't officially support Jetson - however I was able to get it building with some tweaks and am able to use it for some things. The stuff in Nemo that requires torch.distributed doesn't work (like megatron)

but I even cannot  complete: **import nemo.collections.asr as nemo_asr**
here is the error message:
https://github.com/NVIDIA/NeMo/files/12450309/error_message.txt
I just need TTS, ASR.

closed 2025-02-21 by johnnynunez