dusty-nv/jetson-containers #276: opencv is not working with foxy-base image

JuanCarlos-TiqueRangel, 2023-08-29:
Hello @dusty-nv I am working right now with a Jetson nano using the [foxy-ros-base-l4t-r35.4.1](https://hub.docker.com/layers/dustynv/ros/foxy-ros-base-l4t-r35.4.1/images/sha256-74c33dd75c501e606136df588087ce83ecd87e7d7322ae1392a482c22342cf63?context=explore) after the upgrade I got some issues and I saw that it is recommended to remove old versions with `sudo apt-get purge -y '*opencv'`. After doing this I got the next error 

`cv2.error: OpenCV(4.5.0) /opt/opencv/modules/highgui/src/window_gtk.cpp:624: error: (-2:Unspecified error) Can't initialize GTK backend in function 'cvInitSystem'` 

I tried to reinstall opencv and that wasn't work. This is my command to run the docker container `sudo docker run --runtime nvidia -it --privileged -v /dev/bus/usb:/dev/bus/usb --network host --name windrobo --device-cgroup-rule='c 189:* rmw' -e DISPLAY=$DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix dustynv/ros:foxy-ros-base-l4t-r35.4.1` Do you know what my problem could be?

Thanks.

dusty-nv, 2023-08-30:
@JuanCarlos-TiqueRangel are you on the original Jetson Nano, or Jetson Orin Nano?   The original Jetson Nano uses JetPack 4 and doesn't run JetPack 5 containers (so you would use an `r32.x.x` container on the original Nano, depending on your version of L4T)

What does your `cv2.getBuildInformation()` show?   This is mine, and it appears to be built with GTK enabled:

```
General configuration for OpenCV 4.5.0 =====================================
  Version control:               4.5.0

  Extra modules:
    Location (extra):            /opt/opencv_contrib/modules
    Version control (extra):     4.5.0

  Platform:
    Timestamp:                   2021-10-23T16:52:07Z
    Host:                        Linux 5.10.59-tegra aarch64
    CMake:                       3.16.3
    CMake generator:             Unix Makefiles
    CMake build tool:            /usr/bin/make
    Configuration:               RELEASE

  CPU/HW features:
    Baseline:                    NEON FP16
      required:                  NEON
      disabled:                  VFPV3

  C/C++:
    Built as dynamic libs?:      YES
    C++ standard:                11
    C++ Compiler:                /usr/bin/c++  (ver 9.3.0)
    C++ flags (Release):         -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winit-self -Wsuggest-override -Wno-delete-non-virtual-dtor -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -fvisibility-inlines-hidden -O3 -DNDEBUG  -DNDEBUG
    C++ flags (Debug):           -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winit-self -Wsuggest-override -Wno-delete-non-virtual-dtor -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -fvisibility-inlines-hidden -g  -O0 -DDEBUG -D_DEBUG
    C Compiler:                  /usr/bin/cc
    C flags (Release):           -fsigned-char -W -Wall -Werror=return-type -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Winit-self -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -O3 -DNDEBUG  -DNDEBUG
    C flags (Debug):             -fsigned-char -W -Wall -Werror=return-type -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Winit-self -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -g  -O0 -DDEBUG -D_DEBUG
    Linker flags (Release):      -Wl,--gc-sections -Wl,--as-needed
    Linker flags (Debug):        -Wl,--gc-sections -Wl,--as-needed
    ccache:                      NO
    Precompiled headers:         NO
    Extra dependencies:          m pthread cudart_static dl rt nppc nppial nppicc nppidei nppif nppig nppim nppist nppisu nppitc npps cublas cudnn cufft -L/usr/local/cuda/lib64 -L/usr/lib/aarch64-linux-gnu
    3rdparty dependencies:

  OpenCV modules:
    To be built:                 alphamat aruco bgsegm bioinspired calib3d ccalib core cudaarithm cudabgsegm cudacodec cudafeatures2d cudafilters cudaimgproc cudalegacy cudaobjdetect cudaoptflow cudastereo cudawarping cudev datasets dnn dnn_objdetect dnn_superres dpm face features2d flann freetype fuzzy gapi hfs highgui img_hash imgcodecs imgproc intensity_transform line_descriptor mcc ml objdetect optflow phase_unwrapping photo plot python3 quality rapid reg rgbd saliency shape stereo stitching structured_light superres surface_matching text tracking video videoio videostab xfeatures2d ximgproc xobjdetect xphoto
    Disabled:                    world
    Disabled by dependency:      -
    Unavailable:                 cnn_3dobj cvv hdf java js julia matlab ovis python2 sfm ts viz
    Applications:                apps
    Documentation:               NO
    Non-free algorithms:         YES

  GUI:
    GTK+:                        YES (ver 3.24.20)
      GThread :                  YES (ver 2.64.6)
      GtkGlExt:                  NO
    OpenGL support:              NO
    VTK support:                 NO

  Media I/O:
    ZLib:                        /usr/lib/aarch64-linux-gnu/libz.so (ver 1.2.11)
    JPEG:                        /usr/lib/aarch64-linux-gnu/libjpeg.so (ver 80)
    WEBP:                        build (ver encoder: 0x020f)
    PNG:                         /usr/lib/aarch64-linux-gnu/libpng.so (ver 1.6.37)
    TIFF:                        build (ver 42 - 4.0.10)
    JPEG 2000:                   build (ver 2.3.1)
    OpenEXR:                     build (ver 2.3.0)
    HDR:                         YES
    SUNRASTER:                   YES
    PXM:                         YES
[... cut ...]
    numpy:                       /usr/lib/python3/dist-packages/numpy/core/include (ver 1.17.4)
    install path:                lib/python3.8/dist-packages/cv2/python-3.8

  Python (for build):            /usr/bin/python2.7

  Java:
    ant:                         NO
    JNI:                         NO
    Java wrappers:               NO
    Java tests:                  NO

  Install to:                    /usr/local
-----------------------------------------------------------------
```

Are you able to view any GUI window or GTK window from your container?

JuanCarlos-TiqueRangel, 2023-08-30:
Hi @dusty-nv I am using the original Jetson Nano, so according to your last comment I pulled the container [foxy-ros-base-l4t-r32.7.1](https://hub.docker.com/layers/dustynv/ros/foxy-ros-base-l4t-r32.7.1/images/sha256-5afb547718fdd8362c4a09e49530be3dae1e8111fcd41b3a719d6dcc4fb64f71?context=explore) but unfortunately I got the same error.

This is what cv2.getBuildInformation() shows.

```
General configuration for OpenCV 4.5.0 =====================================
  Version control:               4.5.0

  Extra modules:
    Location (extra):            /opt/opencv_contrib/modules
    Version control (extra):     4.5.0

  Platform:
    Timestamp:                   2021-07-19T21:05:54Z
    Host:                        Linux 4.9.201-tegra aarch64
    CMake:                       3.10.2
    CMake generator:             Unix Makefiles
    CMake build tool:            /usr/bin/make
    Configuration:               RELEASE

  CPU/HW features:
    Baseline:                    NEON FP16
      required:                  NEON
      disabled:                  VFPV3

  C/C++:
    Built as dynamic libs?:      YES
    C++ standard:                11
    C++ Compiler:                /usr/bin/c++  (ver 7.5.0)
    C++ flags (Release):         -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winit-self -Wsuggest-override -Wno-delete-non-virtual-dtor -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -fvisibility-inlines-hidden -O3 -DNDEBUG  -DNDEBUG
    C++ flags (Debug):           -fsigned-char -W -Wall -Werror=return-type -Werror=non-virtual-dtor -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wundef -Winit-self -Wpointer-arith -Wshadow -Wsign-promo -Wuninitialized -Winit-self -Wsuggest-override -Wno-delete-non-virtual-dtor -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -fvisibility-inlines-hidden -g  -O0 -DDEBUG -D_DEBUG
    C Compiler:                  /usr/bin/cc
    C flags (Release):           -fsigned-char -W -Wall -Werror=return-type -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Winit-self -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -O3 -DNDEBUG  -DNDEBUG
    C flags (Debug):             -fsigned-char -W -Wall -Werror=return-type -Werror=address -Werror=sequence-point -Wformat -Werror=format-security -Wmissing-declarations -Wmissing-prototypes -Wstrict-prototypes -Wundef -Winit-self -Wpointer-arith -Wshadow -Wuninitialized -Winit-self -Wno-comment -Wimplicit-fallthrough=3 -Wno-strict-overflow -fdiagnostics-show-option -pthread -fomit-frame-pointer -ffunction-sections -fdata-sections    -fvisibility=hidden -g  -O0 -DDEBUG -D_DEBUG
    Linker flags (Release):      -Wl,--gc-sections -Wl,--as-needed  
    Linker flags (Debug):        -Wl,--gc-sections -Wl,--as-needed  
    ccache:                      NO
    Precompiled headers:         NO
    Extra dependencies:          m pthread cudart_static dl rt nppc nppial nppicc nppicom nppidei nppif nppig nppim nppist nppisu nppitc npps cublas cudnn cufft -L/usr/local/cuda/lib64 -L/usr/lib/aarch64-linux-gnu
    3rdparty dependencies:

  OpenCV modules:
    To be built:                 alphamat aruco bgsegm bioinspired calib3d ccalib core cudaarithm cudabgsegm cudacodec cudafeatures2d cudafilters cudaimgproc cudalegacy cudaobjdetect cudaoptflow cudastereo cudawarping cudev datasets dnn dnn_objdetect dnn_superres dpm face features2d flann freetype fuzzy gapi hfs highgui img_hash imgcodecs imgproc intensity_transform line_descriptor mcc ml objdetect optflow phase_unwrapping photo plot python3 quality rapid reg rgbd saliency shape stereo stitching structured_light superres surface_matching text tracking video videoio videostab xfeatures2d ximgproc xobjdetect xphoto
    Disabled:                    world
    Disabled by dependency:      -
    Unavailable:                 cnn_3dobj cvv hdf java js julia matlab ovis python2 sfm ts viz
    Applications:                apps
    Documentation:               NO
    Non-free algorithms:         YES

  GUI: 
    GTK+:                        YES (ver 3.22.30)
      GThread :                  YES (ver 2.56.4)
      GtkGlExt:                  NO
    OpenGL support:              NO
    VTK support:                 NO

  Media I/O: 
    ZLib:                        /usr/lib/aarch64-linux-gnu/libz.so (ver 1.2.11)
    JPEG:                        /usr/lib/aarch64-linux-gnu/libjpeg.so (ver 80)
    WEBP:                        build (ver encoder: 0x020f)
    PNG:                         /usr/lib/aarch64-linux-gnu/libpng.so (ver 1.6.34)
    TIFF:                        build (ver 42 - 4.0.10)
    JPEG 2000:                   build (ver 2.3.1)
    OpenEXR:                     build (ver 2.3.0)
    HDR:                         YES
    SUNRASTER:                   YES
[... cut ...]
    JNI:                         NO
    Java wrappers:               NO
    Java tests:                  NO

  Install to:                    /usr/local
-----------------------------------------------------------------
```

As well as you I have the GTK enabled but I am still with the error. Also when I tried to display a picture with `imagemagick` I got `display-im6.q16: unable to open X server 'localhost:11.0' @ error/display.c/DisplayImageCommand/432.` but I added the $DISPLAY and the other commands when I ran the container.

dusty-nv, 2023-08-30:
@JuanCarlos-TiqueRangel did you try running `sudo xhost +si:localuser:root` first before you launched the container?

Here are the flags I run/add when enabling display in container:

https://github.com/dusty-nv/jetson-containers/blob/e46bea804ae31a95485e4ff97e10ea7ba0cd8dc7/run.sh#L16

JuanCarlos-TiqueRangel, 2023-08-30:
@dusty-nv after follow your instructions about the `sudo xhost +si:localuser:root` and I added the flags so, the run command is the next:
```
sudo docker run --runtime nvidia -it \
	--privileged \
	-v /dev/bus/usb:/dev/bus/usb \
	--network host --name windrobo \
	--device-cgroup-rule='c 189:* rmw' \
	-e DISPLAY=$DISPLAY \
	-v /tmp/.X11-unix:/tmp/.X11-unix \
	-v $XAUTH:$XAUTH \
	-e XAUTHORITY=$XAUTH \
	--env="QT_X11_NO_MITSHM=1" \
	dustynv/ros:foxy-ros-base-l4t-r32.7.1 \
```	

I was into the container and I updated and upgraded it then I restarted the container and know I have the next error:

```
Error response from daemon: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error mounting "/tmp/.docker.xauth" to rootfs at "/tmp/.docker.xauth": mount /tmp/.docker.xauth:/tmp/.docker.xauth (via /proc/self/fd/7), flags: 0x5000: not a directory: unknown: Are you trying to mount a directory onto a file (or vice-versa)? Check if the specified host path exists and is the expected type
```

Do you know what I did wrong ?

JuanCarlos-TiqueRangel, 2023-09-02:
@dusty-nv an update about this issue. After doing a research about the problem I got that to fix it is neccesary to remove the old Xauthority data. However, I used your run.sh file and now I got this error

'''
Error response from daemon: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error mounting "/tmp/nv_jetson_model" to rootfs at "/tmp/nv_jetson_model": mount /tmp/nv_jetson_model:/tmp/nv_jetson_model (via /proc/self/fd/7), flags: 0x5000: not a directory: unknown: Are you trying to mount a directory onto a file (or vice-versa)? Check if the specified host path exists and is the expected type
'''

Maybe Do you know what is the problem ? Thank you for your time and help in this matter. Sorry to bother you with this case.

dusty-nv, 2023-09-02:
@JuanCarlos-TiqueRangel `cat /proc/device-tree/model > /tmp/nv_jetson_model` should have been run by run.sh here:

https://github.com/dusty-nv/jetson-containers/blob/02fa7dcb3cc97efb5818ab5789aaa1a6a999829b/run.sh#L45C2-L45C52

Can you try running that?

It's done like that because /proc files can't be mounted

JuanCarlos-TiqueRangel, 2023-09-03:
@dusty-nv hi, Thanks for your asnwer. This is my run.sh

```
#!/usr/bin/env bash
# pass-through commands to 'docker run' with some defaults
# https://docs.docker.com/engine/reference/commandline/run/
ROOT="$( cd "$( dirname "${BASH_SOURCE[0]}" )" >/dev/null 2>&1 && pwd )"

# check for V4L2 devices
V4L2_DEVICES=""

for i in {0..9}
do
	if [ -a "/dev/video$i" ]; then
		V4L2_DEVICES="$V4L2_DEVICES --device /dev/video$i "
	fi
done

# check for display
DISPLAY_DEVICE=""

if [ -n "$DISPLAY" ]; then
	# give docker root user X11 permissions
	sudo xhost +si:localuser:root
	
	# enable SSH X11 forwarding inside container (https://stackoverflow.com/q/48235040)
	XAUTH=/tmp/.docker.xauth
	xauth nlist $DISPLAY | sed -e 's/^..../ffff/' | xauth -f $XAUTH nmerge -
	chmod 777 $XAUTH

	DISPLAY_DEVICE="-e DISPLAY=$DISPLAY -v /tmp/.X11-unix/:/tmp/.X11-unix -v $XAUTH:$XAUTH -e XAUTHORITY=$XAUTH"
fi

# check if sudo is needed
if id -nG "$USER" | grep -qw "docker"; then
	SUDO=""
else
	SUDO="sudo"
fi

# run the container
ARCH=$(uname -i)

if [ $ARCH = "aarch64" ]; then

	# this file shows what Jetson board is running
	# /proc or /sys files aren't mountable into docker
	cat /proc/device-tree/model > /tmp/nv_jetson_model

	set -x

	$SUDO docker run --runtime nvidia -it --rm --network host \
		--volume /tmp/argus_socket:/tmp/argus_socket \
		--volume /etc/enctune.conf:/etc/enctune.conf \
		--volume /etc/nv_tegra_release:/etc/nv_tegra_release \
		--volume /tmp/nv_jetson_model:/tmp/nv_jetson_model \
		--volume $ROOT/data:/data \
		--device /dev/snd \
		--device /dev/bus/usb \
		$DATA_VOLUME $DISPLAY_DEVICE $V4L2_DEVICES \
		"$@"

elif [ $ARCH = "x86_64" ]; then

	set -x

	$SUDO docker run --gpus all -it --rm --network=host \
		--shm-size=8g \
		--ulimit memlock=-1 \
		--ulimit stack=67108864 \
		--env NVIDIA_DRIVER_CAPABILITIES=all \
		--volume $ROOT/data:/data \
		$DATA_VOLUME $DISPLAY_DEVICE $V4L2_DEVICES \
		"$@"	
fi
```

As you see, I deleted the `--rm` flag because I want to continue using the container. But when I ran `sudo docker start <container_name>` I got the error 

```
Error response from daemon: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error mounting "/tmp/nv_jetson_model" to rootfs at "/tmp/nv_jetson_model": mount /tmp/nv_jetson_model:/tmp/nv_jetson_model (via /proc/self/fd/7), flags: 0x5000: not a directory: unknown: Are you trying to mount a directory onto a file (or vice-versa)? Check if the specified host path exists and is the expected type
```

So having that in mind, What can I do to avoid this error and still have my container ?

dusty-nv, 2023-09-03:
@JuanCarlos-TiqueRangel remove the `--volume /tmp/nv_jetson_model:/tmp/nv_jetson_model` line from run.sh

JuanCarlos-TiqueRangel, 2023-09-04:
Hi @dusty-nv I followed some of your advices in this issue #212. However, after the reboot I got the next error again:

```
cv2.imshow(window_name, image)
cv2.error: OpenCV(4.5.0) /opt/opencv/modules/highgui/src/window_gtk.cpp:624: error: (-2:Unspecified error) Can't initialize GTK backend in function 'cvInitSystem'
```

this is my `run.sh` file

```

#!/usr/bin/env bash
# pass-through commands to 'docker run' with some defaults
# https://docs.docker.com/engine/reference/commandline/run/
ROOT="$( cd "$( dirname "${BASH_SOURCE[0]}" )" >/dev/null 2>&1 && pwd )"

# check for V4L2 devices
V4L2_DEVICES=""

for i in {0..9}
do
        if [ -a "/dev/video$i" ]; then
                V4L2_DEVICES="$V4L2_DEVICES --device /dev/video$i "
        fi
done

# check for display
DISPLAY_DEVICE=""

if [ -n "$DISPLAY" ]; then
        # give docker root user X11 permissions
        sudo xhost +si:localuser:root

        # enable SSH X11 forwarding inside container (https://stackoverflow.com/q/48235040)
        XAUTH=/home/windrobo/Documents/docker_xauth/.docker.xauth
        xauth nlist $DISPLAY | sed -e 's/^..../ffff/' | xauth -f $XAUTH nmerge -
        chmod 777 $XAUTH

        DISPLAY_DEVICE="-e DISPLAY=$DISPLAY -v /tmp/.X11-unix/:/tmp/.X11-unix -v $XAUTH:$XAUTH -e XAUTHORITY=$XAUTH"
fi

# check if sudo is needed
if id -nG "$USER" | grep -qw "docker"; then
        SUDO=""
else
        SUDO="sudo"
fi

# run the container
ARCH=$(uname -i)

if [ $ARCH = "aarch64" ]; then

        # this file shows what Jetson board is running
        # /proc or /sys files aren't mountable into docker
        cat /proc/device-tree/model > /tmp/nv_jetson_model

        set -x

        $SUDO docker run --runtime nvidia -it --name windrobo --network host \
                --env="QT_X11_NO_MITSHM=1" \
                --volume /tmp/argus_socket:/tmp/argus_socket \
                --volume /etc/enctune.conf:/etc/enctune.conf \
                --volume /etc/nv_tegra_release:/etc/nv_tegra_release \
                --volume $ROOT/data:/data \
                --device /dev/snd \
                --device /dev/bus/usb \
                $DATA_VOLUME $DISPLAY_DEVICE $V4L2_DEVICES \
                "$@"

elif [ $ARCH = "x86_64" ]; then

        set -x

        $SUDO docker run --gpus all -it --network=host \
                --shm-size=8g \
                --ulimit memlock=-1 \
                --ulimit stack=67108864 \
                --env NVIDIA_DRIVER_CAPABILITIES=all \
                --volume $ROOT/data:/data \
                $DATA_VOLUME $DISPLAY_DEVICE $V4L2_DEVICES \
                "$@"    
fi
```

At this point I do not know what I am doing wrong. Do yow know what could be the problem ?

dusty-nv, 2023-09-04:
@JuanCarlos-TiqueRangel at this point, are you able to display other GUI windows within container, like other GTK GUI apps, GEdit, XTerm, imagemagik, ect?

JuanCarlos-TiqueRangel, 2023-09-05:
@dusty-nv No I cannot. After reboot the system If I want to use OpenCV I got this:
```
cv2.imshow(window_name, image)
cv2.error: OpenCV(4.5.0) /opt/opencv/modules/highgui/src/window_gtk.cpp:624: error: (-2:Unspecified error) Can't initialize GTK backend in function 'cvInitSystem'
```

And if I want to use imagemagik, I got the next error after run `display image.jpg`:

```
X11 connection rejected because of wrong authentication.
display-im6.q16: unable to open X server `localhost:10.0' @ error/display.c/DisplayImageCommand/432.
```

Maybe you know what is the problem ? I think is something with the flags but to be honest I do not know exactly what to do in this case.

dusty-nv, 2023-09-05:
Honestly the X11-in-container troubleshooting always baffled me too.   Try it with `nvcr.io/nvidia/l4t-jetpack:r35.4.1` too

JuanCarlos-TiqueRangel, 2023-09-06:
Hi @dusty-nv the same problem. This first time works good but after reboot the system I got the error 

```
cv2.imshow(window_name, image)
cv2.error: OpenCV(4.5.0) /opt/opencv/modules/highgui/src/window_gtk.cpp:624: error: (-2:Unspecified error) Can't initialize GTK backend in function 'cvInitSystem'
```

Maybe do you have some advice about how to fix this ? It is taking me more time that I thought

dusty-nv, 2023-09-06:
> This first time works good but after reboot the system I got the error

Ahh okay, my guess is this is related to you removing the `--rm` and wanting to restart the same container, and potentially the `-v /tmp/.X11-unix/:/tmp/.X11-unix` file changing after reboot since it's in `/tmp`.   I see you already changed the XAUTH file to not be in `/tmp` anymore.

It doesn't seem possible/supported to add a mount dynamically after the container is started:   https://stackoverflow.com/questions/28302178/how-can-i-add-a-volume-to-an-existing-docker-container

TBH, I'm not sure how to fix it without starting new container after reboot, I'm sorry :/

JuanCarlos-TiqueRangel, 2023-09-08:
Hello @dusty-nv After doing research about the containers with docker I realize that creating a container for a long term is bad practice. The solution for this issue is first install the necessary packages in the `Dockerfile` and then mount a volume that has the file that I need in my local and host. Is basically add the next line in the `docker run` command:

```
--volume $ROOT/ros2_ws/src:/root/ros2_ws/src \
```

Thank you for your time and help here.

closed 2023-09-08 by JuanCarlos-TiqueRangel

dusty-nv, 2023-09-08:
Yea, making your own Dockerfile is preferred because then you can re-build in an automated way whenever necessary.   Otherwise you can save changes that you made by hand using `docker commit`

dat-nguyenvn, 2025-01-07:
any better solution for the error instead of create new container again . Error: `Error response from daemon: failed to create shim task: OCI runtime create failed: runc create failed: unable to start container process: error during container init: error mounting "/tmp/nv_jetson_model" to rootfs at "/tmp/nv_jetson_model": mount /tmp/nv_jetson_model:/tmp/nv_jetson_model (via /proc/self/fd/7), flags: 0x5000: not a directory: unknown: Are you trying to mount a directory onto a file (or vice-versa)? Check if the specified host path exists and is the expected type`