dusty-nv/jetson-containers #279: Support for ROS2 Iron Irwini

IamShubhamGupto, 2023-09-06:
### Summary:

I have been trying to build my own docker image with PyTorch 2.0 and ros2 iron but I run into this issue https://stackoverflow.com/questions/71606313/colcon-build-fail-rclpy. Since I may not be the only one looking for a docker image with such requirements, it would be nice implement this and contribute.

# Proposal

Add support for the latest ROS2 version - Iron. While not necessary, communicating with other boards already running ros2 Iron is the goal.

# Supported OS
- Jetpack version 5+
- Jetpack version 4 -> might be already running using older tools so does not make sense to bring support unless in demand

## Priority images
- ros iron 
- ros iron + PyTorch
- ros iron (gui) + PyTorch

dusty-nv, 2023-09-06:
@IamShubhamGupto ros2 iron has already been supported:  https://github.com/dusty-nv/jetson-containers/tree/master/packages/ros

I was able to build it on top of PyTorch with:

```
./build.sh --name=test/ros2-iron-pytorch:r35.2.1 pytorch ros:iron-desktop
```

Interestingly, the reverse did not work because presumably ros installs a different/older version of pybind11:

```
./build.sh --name=test/ros2-iron-pytorch:r35.2.1 ros:iron-desktop pytorch 

 In function ‘void onnx::pybind11_init_onnx_cpp2py_export(pybind11::module&)’:
  /tmp/pip-req-build-qpbtpzra/onnx/cpp2py_export.cc:165:15: error: ‘kw_only’ is not a member of ‘onnx::py’
    165 |           py::kw_only(),
        |               ^~~~~~~
  /tmp/pip-req-build-qpbtpzra/onnx/cpp2py_export.cc:223:15: error: ‘kw_only’ is not a member of ‘onnx::py’
    223 |           py::kw_only(),
        |               ^~~~~~~
  /tmp/pip-req-build-qpbtpzra/onnx/cpp2py_export.cc:306:15: error: ‘kw_only’ is not a member of ‘onnx::py’
    306 |           py::kw_only(),
        |               ^~~~~~~
```

IamShubhamGupto, 2023-09-06:
@dusty-nv 
Perfect! I must have missed this. 
Do you have any suggestions on how I would add realsense support for [dustynv/ros:iron-pytorch-l4t-r35.2.1](https://hub.docker.com/r/dustynv/ros/tags)?

IamShubhamGupto, 2023-09-06:
@dusty-nv 
Great! so im assuming the command is like
./build.sh --name=test/ros2-iron-pytorch:r35.2.1 pytorch ros:iron-desktop realsense

As for the ros node, we were considering to use this
https://github.com/IntelRealSense/realsense-ros

closed 2023-09-06 by IamShubhamGupto

IamShubhamGupto, 2023-09-22:
@dusty-nv Following up on using the above built container. We noticed ros2:iron is not fully compatible with Ubuntu 20.04 (focal) 

Is it possible to create a Jetson image by explicitly specifying the ubuntu version? (ubuntu 22 jammy)

dusty-nv, 2023-09-22:
@IamShubhamGupto it then wouldn't have CUDA in it, otherwise you could just run the official OSRF containers

JetPack 6 with Ubuntu 22.04 will be out for Orin in the November timeframe.

I was able to build iron:desktop without issue on 20.04-based JetPack 5, and normally desktop is what would give me incompatibility issues, but I could see how it's possible with other 3rd-party complex packages.  Regardless you'll need to build most packages from source.  Using rosinstall_generator is the easy way for packages in the ROS index.

IamShubhamGupto, 2023-09-23:
@dusty-nv Thank you for the update

I was hoping to get an immediate fix. For example, the https://github.com/IntelRealSense/realsense-ros node requires a dependency called `xacro`and `launch_pytest` which I was unable to add manually (not found in focal). Would rosinstall_generator solve this issue?

Diverging from the topic, is it possible to build and maintain my own containers for the Jetson platform? For example the Jetson nano does not seem to be receiving any jetpack updates but it would be nice to run Ubuntu 22 (with iron) on it as well

dusty-nv, 2023-09-23:
@IamShubhamGupto here's an example where i built realsense-ros package on top of humble in 20.04 container, and yea I added `xacro` to rosinstall_generator:

https://github.com/dusty-nv/create3_nano/blob/main/Dockerfile

IamShubhamGupto, 2023-10-05:
@dusty-nv Do you know if we can upgrade Xavier NX to JP6? if not, can we run JP6 docker containers on a Xavier NX running JP5