dusty-nv/jetson-containers #281: Failed to import pyrealsense2

IamShubhamGupto, 2023-09-06:
Based on our discussion in #279 
I am trying to build my own docker image with ros2 iron pytorch2 and realsense. 

The command
```
./build.sh --name nyuultraviolet/ros2-iron-pytorch2-realsense:r35.3.1 \
pytorch:2.0 ros:iron-desktop realsense torch2trt torch_tensorrt torchvision numba numpy onnx cupy
```

The error 
```
Removing intermediate container 107458e0d1af
 ---> 74d72dc429c6
Step 6/6 : RUN python3 -c 'import pyrealsense2'
 ---> Running in 3a6b598bdcce
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ModuleNotFoundError: No module named 'pyrealsense2'
The command '/bin/bash -c python3 -c 'import pyrealsense2'' returned a non-zero code: 1
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/ultraviolet/Downloads/jetson-containers/jetson_containers/build.py", line 95, in <module>
    build_container([args.name](http://args.name/), args.packages, args.base, args.build_flags, args.simulate, args.skip_tests, args.test_only, args.push)
  File "/home/ultraviolet/Downloads/jetson-containers/jetson_containers/container.py", line 128, in build_container
    status = subprocess.run(cmd.replace(NEWLINE, ' '), executable='/bin/bash', shell=True, check=True)  
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'docker build --network=host --tag nyuultraviolet/ros2-iron-pytorch2-realsense:r35.3.1-r35.3.1-realsense --file /home/ultraviolet/Downloads/jetson-containers/packages/realsense/Dockerfile --build-arg BASE_IMAGE=nyuultraviolet/ros2-iron-pytorch2-realsense:r35.3.1-r35.3.1-ros_iron-desktop /home/ultraviolet/Downloads/jetson-containers/packages/realsense 2>&1 | tee /home/ultraviolet/Downloads/jetson-containers/logs/20230906_151420/build/nyuultraviolet_ros2-iron-pytorch2-realsense_r35.3.1-r35.3.1-realsense.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.
```

Device
Jetson Orin

IamShubhamGupto, 2023-09-07:
UPDATE

Using the sources ive built realsense before on docker, I updated the docker file to the following

```
#---
# name: realsense
# group: sensors
# depends: [python, cmake]
# test: test.py
# notes: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md
#---
ARG BASE_IMAGE
FROM ${BASE_IMAGE}

RUN apt-get update && \
    apt-get install -y --no-install-recommends \
		  libssl-dev \
		  libusb-1.0-0-dev \
            libgtk-3-dev \
            libglfw3-dev \
		  libgl1-mesa-dev \
		  libglu1-mesa-dev \
		  qtcreator && \
    if [ $(lsb_release -cs) = "bionic" ]; then \
        apt-get install -y --no-install-recommends python-dev; \
    else \
        apt-get install -y --no-install-recommends python2-dev; \
    fi \
    && rm -rf /var/lib/apt/lists/* \
    && apt-get clean

# https://github.com/IntelRealSense/librealsense/issues/11931
ARG LIBREALSENSE_VERSION=development

RUN git clone --branch ${LIBREALSENSE_VERSION} --depth=1 https://github.com/IntelRealSense/librealsense && \
    cd librealsense && \
    mkdir build && \
    cd build && \
    cmake \
        -DBUILD_EXAMPLES=true \
	   -DFORCE_RSUSB_BACKEND=true \
	   -DBUILD_WITH_CUDA=true \
	   -DCMAKE_BUILD_TYPE=release \
	   -DBUILD_PYTHON_BINDINGS=bool:true \
	   -DPYTHON_EXECUTABLE=/usr/bin/python3 \
	   -DPYTHON_INSTALL_DIR=$(python3 -c 'import sys; print(f"/usr/lib/python{sys.version_info.major}.{sys.version_info.minor}/")') \
	   ../ && \
    make -j$(($(nproc)-1)) && \
    make install && \
    cd ../ && \
    cp ./config/99-realsense-libusb.rules /etc/udev/rules.d/ && \
    mkdir /usr/lib/python3.8/pyrealsense2 && \
    ln -s ./build/wrappers/python/pybackend2.cpython-38-$(uname -m)-linux-gnu.so /usr/lib/python3.8/pyrealsense2/ && \
    ln -s ./build/wrappers/python/pyrealsense2.cpython-38-$(uname -m)-linux-gnu.so /usr/lib/python3.8/pyrealsense2/ && \
    rm -rf librealsense

RUN echo 'PYTHONPATH="/usr/lib/python3.8/pyrealsense2:$PYTHONPATH"' >> ~/.bashrc && \
	echo 'export PYTHONPATH' >> ~/.bashrc 
RUN source ~/.bashrc 
RUN /lib/systemd/systemd-udevd --daemon && udevadm control --reload-rules && \
	udevadm trigger
RUN python3 -c 'import pyrealsense2'
```

The above Dockerfile solved the import issue but I now seem to be getting stuck at something very basic
```
Step 8/9 : RUN /lib/systemd/systemd-udevd --daemon && udevadm control --reload-rules && 	udevadm trigger
 ---> Running in 5db2d5b4a8ee
Starting version 245.4-4ubuntu3.22
Removing intermediate container 5db2d5b4a8ee
 ---> 62b11ef2c1c2
Step 9/9 : RUN python3 -c 'import pyrealsense2'
 ---> Running in 8c27eee15bc5
Removing intermediate container 8c27eee15bc5
 ---> b07177a66ab7
Successfully built b07177a66ab7
Successfully tagged nyuultraviolet/ros2-iron-pytorch2-realsense:r35.3.1-r35.3.1-realsense
-- Testing container nyuultraviolet/ros2-iron-pytorch2-realsense:r35.3.1-r35.3.1-realsense (realsense/test.py)

docker run -t --rm --runtime=nvidia --network=host \
--volume /home/ultraviolet/Downloads/jetson-containers/packages/realsense:/test \
--volume /home/ultraviolet/Downloads/jetson-containers/data:/data \
--workdir /test \
nyuultraviolet/ros2-iron-pytorch2-realsense:r35.3.1-r35.3.1-realsense \
/bin/bash -c 'python3 test.py' \
2>&1 | tee /home/ultraviolet/Downloads/jetson-containers/logs/20230906_200511/test/nyuultraviolet_ros2-iron-pytorch2-realsense_r35.3.1-r35.3.1-realsense_test.py.txt; exit ${PIPESTATUS[0]}

sourcing   /opt/ros/iron/install/setup.bash
ROS_DISTRO iron
ROS_ROOT   /opt/ros/iron
testing pyrealsense2...
Traceback (most recent call last):
  File "test.py", line 5, in <module>
    print('pyrealsense2 version:', rs.__version__)
AttributeError: module 'pyrealsense2' has no attribute '__version__'
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/ultraviolet/Downloads/jetson-containers/jetson_containers/build.py", line 95, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.simulate, args.skip_tests, args.test_only, args.push)
  File "/home/ultraviolet/Downloads/jetson-containers/jetson_containers/container.py", line 135, in build_container
    test_container(container_name, pkg, simulate)
  File "/home/ultraviolet/Downloads/jetson-containers/jetson_containers/container.py", line 307, in test_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'docker run -t --rm --runtime=nvidia --network=host --volume /home/ultraviolet/Downloads/jetson-containers/packages/realsense:/test --volume /home/ultraviolet/Downloads/jetson-containers/data:/data --workdir /test nyuultraviolet/ros2-iron-pytorch2-realsense:r35.3.1-r35.3.1-realsense /bin/bash -c 'python3 test.py' 2>&1 | tee /home/ultraviolet/Downloads/jetson-containers/logs/20230906_200511/test/nyuultraviolet_ros2-iron-pytorch2-realsense_r35.3.1-r35.3.1-realsense_test.py.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 1.
```

Thanks

IamShubhamGupto, 2023-09-08:
Hey @dusty-nv 

Please do check out this issue and let me know if you need any more information

IamShubhamGupto, 2023-09-20:
could not figure out how to fit it
as for `pyrealsense2` module, did an additional `pip install pyrealsense2` which adds pipeline() and other functions. The module is usable and `real sense-viewer` also works

Mattzi, 2023-09-26:
Still having the same issue and just a simple pip install pyrealsense2 in the docker file does not help. Anybody has an idea?

mzahana, 2024-01-03:
I am having the same issue. My command is:
```bash
./build.sh --name=noetic_realsense pytorch realsense ros:noetic-desktop
```

Any possible fix? @dusty-nv  @IamShubhamGupto 

Thanks.

mzahana, 2024-01-03:
I found a related issue [here}(https://github.com/IntelRealSense/librealsense/issues/5790).
It seems that `pyrealsense2` has no attribute `__version__`.
After commenting this line
https://github.com/dusty-nv/jetson-containers/blob/cb6c847f88df221e705397a1ee98424c2e893243/packages/realsense/test.py#L5

AND
Adding 
```docker
RUN pip install pyrealsense2
```
before this line 
https://github.com/dusty-nv/jetson-containers/blob/cb6c847f88df221e705397a1ee98424c2e893243/packages/realsense/Dockerfile#L51

The docker building process went fine.

IamShubhamGupto, 2024-02-01:
> I found a related issue [here}(https://github.com/[IntelRealSense/librealsense/issues/5790](https://github.com/IntelRealSense/librealsense/issues/5790)). It seems that `pyrealsense2` has no attribute `__version__`. After commenting this line
> 
> https://github.com/dusty-nv/jetson-containers/blob/cb6c847f88df221e705397a1ee98424c2e893243/packages/realsense/test.py#L5
> 
> AND Adding
> 
> ```
> RUN pip install pyrealsense2
> ```
> 
> before this line
> 
> https://github.com/dusty-nv/jetson-containers/blob/cb6c847f88df221e705397a1ee98424c2e893243/packages/realsense/Dockerfile#L51
> 
> The docker building process went fine.

 I ran into this issue again while building an image for JP6 DP. I think it is unresolved and possibly deserves a PR @dusty-nv

my command:
```
./build.sh --name=test/ros2-iron-pytorch:r36 pytorch:2.1 tensorrt onnx onnxruntime  ros:iron-desktop realsense
```

IamShubhamGupto, 2024-02-09:
@dusty-nv heres a Dockerfile I am using to build realsense. It builds the library and the ros2 wrapper (humble). I think both should be supplied in the same docker file unless we are trying to save space

```
ARG BASE_IMAGE=nyuultraviolet/cv-dockerfile:jp60-pytorch21-humble-tensorrt

FROM ${BASE_IMAGE}

RUN apt-get install -y --no-install-recommends \
    python3 \
    python3-pip \
    python3-dev && apt-get update && apt-get clean all

RUN apt-get update && \
    apt-get install -y --no-install-recommends \
		  libssl-dev \
		  libusb-1.0-0-dev \
            libgtk-3-dev \
            libglfw3-dev \
		  libgl1-mesa-dev \
		  libglu1-mesa-dev \
		  udev \
		  qtcreator && \
    rm -rf /var/lib/apt/lists/* \
    && apt-get clean

# https://github.com/IntelRealSense/librealsense/issues/11931
ARG LIBREALSENSE_VERSION=development

RUN git clone --branch ${LIBREALSENSE_VERSION} --depth=1 https://github.com/IntelRealSense/librealsense && \
    cd librealsense && \
    mkdir build && \
    cd build && \
    cmake \
        -DBUILD_EXAMPLES=true \
	   -DFORCE_RSUSB_BACKEND=true \
	   -DBUILD_WITH_CUDA=true \
	   -DCMAKE_BUILD_TYPE=release \
	   -DBUILD_PYTHON_BINDINGS=bool:true \
	   -DPYTHON_EXECUTABLE=/usr/bin/python3 \
	   ../ && \
    make -j$(($(nproc)-1)) && \
    make install && \
    cd ../ && \
    cp -r . /usr/local/ && \
    cp ./config/99-realsense-libusb.rules /etc/udev/rules.d/ && \
    cp ./config/99-realsense-d4xx-mipi-dfu.rules /etc/udev/rules.d/ && \
    rm -rf librealsense

ENV PYTHONPATH=${PYTHONPATH}:/usr/local/lib

ENV DEBIAN_FRONTEND=noninteractive

RUN apt-get update \
 && apt-get install -y --no-install-recommends \	
    libusb-1.0-0 \
    udev \
    apt-transport-https \
    ca-certificates \
    curl \
    software-properties-common \
 && rm -rf /var/lib/apt/lists/*

# The following steps are based on: https://github.com/IntelRealSense/realsense-ros/tree/ros2-development
ENV WS_DIR="/ros2_ws"
WORKDIR ${WS_DIR}
RUN apt-get update -y \
 && mkdir src \
 && cd src \
 && git clone https://github.com/IntelRealSense/realsense-ros.git -b ros2-development \
 && cd .. \
 && apt-get install -y python3-rosdep \
 && source /opt/ros/${ROS_DISTRO}/install/setup.bash \
 && rm /etc/ros/rosdep/sources.list.d/20-default.list \
 && rosdep init \
 && rosdep update \
 && rosdep install -i --from-path src --rosdistro ${ROS_DISTRO} --skip-keys=librealsense2 -y \
 && source /opt/ros/${ROS_DISTRO}/setup.bash \
 && colcon build
 
RUN source ./install/local_setup.sh
RUN cd

ENV DEBIAN_FRONTEND=dialog

CMD ["/bin/bash"]

```

closed 2024-02-09 by IamShubhamGupto