dusty-nv/jetson-containers #284: how to use "apt install ros-humble-*" on ros humble docker

snmsung716, 2023-09-13:
Hi
I am using jetson agx orin 64gb and a docker image which has ros2 humble desktop. 

Then I tried to type apt install ros-humble-gazabo-dev but got an error unable to use ros-humble-gazebo-dev.

So I typed apt list and found ros-humble doesn't exist on the list. It has only two ros-versions, galitic and rolling inside the list.

In this case, should I do "build from source" to get some ros-humble-repos?

Thank you for reading this issue.

dusty-nv, 2023-09-13:
Hi @snmsung716, ROS Humble only has apt/deb packages available for Ubuntu 22.04 (which will be coming in JetPack 6), but on JetPack 5 / 20.04 you would build from source.   

One thing you can do, is mount a directory for your ROS2 workspace, and build the additional packages you want from source, while running ros:humble-desktop container.  

You could also write your own Dockerfile which used ros:humble-desktop as base container, and rosinstall_generator to pull the sources of the packages you want:  https://github.com/dusty-nv/jetson-containers/blob/47aa733d9a1c21b08e9333a627718f98d733539c/packages/ros/ros2_build.sh#L79

Or you can just add the packages you want right there and re-build ros:humble-desktop.  The downside of this, is if there are errors related to a package you added, you'll have to go through the full build again.

omerts, 2023-12-19:
@dusty-nv Even with Jetpack 6.0, `apt install` if failing:
```
#0 732.6 You might want to run 'apt --fix-broken install' to correct these.
#0 732.6 The following packages have unmet dependencies:
#0 732.8  libopencv-viz-dev : Depends: libopencv-core-dev (= 4.5.4+dfsg-9ubuntu4) but it is not going to be installed
#0 732.8  ros-humble-ament-cmake-cppcheck : Depends: ros-humble-ament-cppcheck but it is not going to be installed
#0 732.8  ros-humble-cv-bridge : Depends: libopencv-dev but it is not going to be installed
#0 732.8 E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
#0 733.1 Reading package lists...
#0 736.7 Building dependency tree...
#0 737.3 Reading state information...
#0 737.3 You might want to run 'apt --fix-broken install' to correct these.
#0 737.3 The following packages have unmet dependencies:
#0 737.5  libopencv-viz-dev : Depends: libopencv-core-dev (= 4.5.4+dfsg-9ubuntu4) but it is not going to be installed
#0 737.6  ros-humble-cv-bridge : Depends: libopencv-dev but it is not going to be installed
#0 737.6  ros-humble-pcl-ros : Depends: libpcl-dev but it is not going to be installed
#0 737.6                       Depends: ros-humble-pcl-conversions but it is not going to be installed
#0 737.6 E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
#0 737.8 Reading package lists...
#0 741.5 Building dependency tree...
#0 742.1 Reading state information...
#0 742.2 You might want to run 'apt --fix-broken install' to correct these.
#0 742.2 The following packages have unmet dependencies:
#0 742.4  libopencv-viz-dev : Depends: libopencv-core-dev (= 4.5.4+dfsg-9ubuntu4) but it is not going to be installed
#0 742.4  ros-humble-camera-info-manager : Depends: ros-humble-camera-calibration-parsers but it is not going to be installed
#0 742.4  ros-humble-cv-bridge : Depends: libopencv-dev but it is not going to be installed
#0 742.4 E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
#0 742.7 Reading package lists...
#0 746.2 Building dependency tree...
#0 746.7 Reading state information...
#0 746.8 You might want to run 'apt --fix-broken install' to correct these.
#0 746.8 The following packages have unmet dependencies:
#0 746.9  libopencv-viz-dev : Depends: libopencv-core-dev (= 4.5.4+dfsg-9ubuntu4) but it is not going to be installed
#0 747.0  ros-humble-cv-bridge : Depends: libopencv-dev but it is not going to be installed
#0 747.0  ros-humble-image-geometry : Depends: libopencv-dev but it is not going to be installed
#0 747.0 E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
#0 774.5 ERROR: the following rosdeps failed to install
#0 774.5   apt: command [apt-get install -y ros-humble-nav2-bringup] failed
#0 774.5   apt: command [apt-get install -y ros-humble-nav2-util] failed
#0 774.5   apt: command [apt-get install -y ros-humble-xacro] failed
#0 774.5   apt: command [apt-get install -y ros-humble-imu-tools] failed
#0 774.5   apt: command [apt-get install -y ros-humble-tf-transformations] failed
#0 774.5   apt: command [apt-get install -y ros-humble-tf2-ros] failed
#0 774.5   apt: command [apt-get install -y ros-humble-ament-copyright] failed
#0 774.5   apt: command [apt-get install -y ros-humble-ament-flake8] failed
#0 774.5   apt: command [apt-get install -y ros-humble-ament-pep257] failed
#0 774.5   apt: command [apt-get install -y ros-humble-ament-cmake] failed
#0 774.5   apt: command [apt-get install -y ros-humble-rclcpp] failed
#0 774.5   apt: command [apt-get install -y ros-humble-rclcpp-components] failed
#0 774.5   apt: command [apt-get install -y ros-humble-rcutils] failed
#0 774.5   apt: command [apt-get install -y ros-humble-cv-bridge] failed
#0 774.5   apt: command [apt-get install -y ros-humble-ament-cmake-pytest] failed
#0 774.5   apt: command [apt-get install -y ros-humble-ament-lint-auto] failed
#0 774.5   apt: command [apt-get install -y ros-humble-ament-lint-common] failed
#0 774.5   apt: command [apt-get install -y ros-humble-launch] failed
#0 774.5   apt: command [apt-get install -y ros-humble-launch-testing] failed
#0 774.5   apt: command [apt-get install -y ros-humble-launch-testing-ros] failed
#0 774.5   apt: command [apt-get install -y ros-humble-launch-testing-ament-cmake] failed
#0 774.5   apt: command [apt-get install -y ros-humble-rmw-implementation-cmake] failed
#0 774.5   apt: command [apt-get install -y ros-humble-rclcpp-action] failed
#0 774.5   apt: command [apt-get install -y ros-humble-rclcpp-lifecycle] failed
#0 774.5   apt: command [apt-get install -y ros-humble-visualization-msgs] failed
#0 774.5   apt: command [apt-get install -y ros-humble-nav2-msgs] failed
#0 774.5   apt: command [apt-get install -y ros-humble-nav-msgs] failed
#0 774.5   apt: command [apt-get install -y ros-humble-geometry-msgs] failed
#0 774.5   apt: command [apt-get install -y ros-humble-builtin-interfaces] failed
#0 774.5   apt: command [apt-get install -y ros-humble-nav2-costmap-2d] failed
#0 774.5   apt: command [apt-get install -y ros-humble-nav2-core] failed
#0 774.5   apt: command [apt-get install -y ros-humble-pluginlib] failed
#0 774.5   apt: command [apt-get install -y ros-humble-std-srvs] failed
[... 16 lines cut ...]
#0 774.5   apt: Failed to detect successful installation of [ros-humble-ament-pep257]
#0 774.5   apt: Failed to detect successful installation of [ros-humble-cv-bridge]
[... cut ...]
#0 774.5   apt: Failed to detect successful installation of [ros-humble-apriltag]
#0 774.5   apt: Failed to detect successful installation of [ros-humble-ament-cmake-clang-format]
#0 774.5   apt: Failed to detect successful installation of [ros-humble-ament-cmake-cppcheck]
#0 774.5   apt: Failed to detect successful installation of [ros-humble-pcl-ros]
#0 774.5   apt: Failed to detect successful installation of [ros-humble-camera-info-manager]
#0 774.5   apt: Failed to detect successful installation of [ros-humble-image-geometry]
```

omerts, 2023-12-19:
@dusty-nv Thank you for the quick reply :). I will try and update.

snknitheesh, 2025-01-28:
@omerts  has this been resolved ? Having a similar issue while installing rosdeps

closed 2025-02-21 by johnnynunez