dusty-nv/jetson-containers #285: Repository does not have a release file

Mekala02, 2023-09-14:
Hi @dusty-nv 

When attempting to build a Docker image with the 'build-essential' package using the following command:

```sudo ./build.sh --base=dustynv/ros:foxy-pytorch-l4t-r35.1.0 --name=test build-essential```

I encounter an error during the build process.
```
Step 4/5 : RUN apt-get update &&     apt-get install -y --no-install-recommends                   build-essential                 software-properties-common               apt-transport-https             ca-certificates                 lsb-release             pkg-config      gnupg            git             wget            curl            nano            zip             unzip     && rm -rf /var/lib/apt/lists/*     && apt-get clean
 ---> Running in 802873751d89
Get:1 http://ports.ubuntu.com/ubuntu-ports focal InRelease [265 kB]
Get:2 https://repo.download.nvidia.com/jetson/common r35.1 InRelease [2,555 B]
Get:3 http://packages.ros.org/ros2/ubuntu focal InRelease [4,685 B]
Get:4 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease [114 kB]
Ign:5 https://apt.kitware.com/ubuntu focal InRelease
Get:6 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease [108 kB]
Get:7 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease [114 kB]
Get:8 https://repo.download.nvidia.com/jetson/common r35.1/main arm64 Packages [19.1 kB]
Err:9 https://apt.kitware.com/ubuntu focal Release
  Certificate verification failed: The certificate is NOT trusted. The received OCSP status response is invalid.  Could not handshake: Error in the certificate verification. [IP: 66.194.253.25 443]
Get:10 http://packages.ros.org/ros2/ubuntu focal/main arm64 Packages [1,171 kB]
Get:11 http://ports.ubuntu.com/ubuntu-ports focal/multiverse arm64 Packages [139 kB]
Get:12 http://ports.ubuntu.com/ubuntu-ports focal/universe arm64 Packages [11.1 MB]
Get:13 http://ports.ubuntu.com/ubuntu-ports focal/main arm64 Packages [1,234 kB]
Get:14 http://ports.ubuntu.com/ubuntu-ports focal/restricted arm64 Packages [1,317 B]
Get:15 http://ports.ubuntu.com/ubuntu-ports focal-updates/main arm64 Packages [2,631 kB]
Get:16 http://ports.ubuntu.com/ubuntu-ports focal-updates/restricted arm64 Packages [46.2 kB]
Get:17 http://ports.ubuntu.com/ubuntu-ports focal-updates/multiverse arm64 Packages [9,393 B]
Get:18 http://ports.ubuntu.com/ubuntu-ports focal-updates/universe arm64 Packages [1,294 kB]
Get:19 http://ports.ubuntu.com/ubuntu-ports focal-backports/main arm64 Packages [54.8 kB]
Get:20 http://ports.ubuntu.com/ubuntu-ports focal-backports/universe arm64 Packages [27.8 kB]
Get:21 http://ports.ubuntu.com/ubuntu-ports focal-security/universe arm64 Packages [992 kB]
Get:22 http://ports.ubuntu.com/ubuntu-ports focal-security/main arm64 Packages [2,236 kB]
Get:23 http://ports.ubuntu.com/ubuntu-ports focal-security/multiverse arm64 Packages [3,262 B]
Get:24 http://ports.ubuntu.com/ubuntu-ports focal-security/restricted arm64 Packages [45.9 kB]
Reading package lists...
E: The repository 'https://apt.kitware.com/ubuntu focal Release' does not have a Release file.
The command '/bin/bash -c apt-get update &&     apt-get install -y --no-install-recommends                build-essential                 software-properties-common               apt-transport-https             ca-certificates                 lsb-release             pkg-config               gnupg                   git             wget            curl            nano            zip             unzip     && rm -rf /var/lib/apt/lists/*     && apt-get clean' returned a non-zero code: 100
Traceback (most recent call last):
  File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "/home/mekson/mekalacar/docker/jetson-containers/jetson_containers/build.py", line 93, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.simulate, args.skip_tests, args.push)
  File "/home/mekson/mekalacar/docker/jetson-containers/jetson_containers/container.py", line 119, in build_container
    status = subprocess.run(cmd.replace(_NEWLINE_, ' '), executable='/bin/bash', shell=True, check=True)
  File "/usr/lib/python3.8/subprocess.py", line 516, in run
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'sudo docker build --network=host --no-cache --tag test:r35.1.0-build-essential --file /home/mekson/mekalacar/docker/jetson-containers/packages/build-essential/Dockerfile --build-arg BASE_IMAGE=dustynv/ros:foxy-pytorch-l4t-r35.1.0 /home/mekson/mekalacar/docker/jetson-containers/packages/build-essential 2>&1 | tee /home/mekson/mekalacar/docker/jetson-containers/logs/20230914_123024/build/test_r35.1.0-build-essential.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 100.
```

dusty-nv, 2023-09-14:
@Mekala02 I think that image was made from before the kitware/apt issue was fixed - can you try these base images instead?

* [dustynv/ros:foxy-pytorch-l4t-r35.2.1](https://hub.docker.com/r/dustynv/ros/tags)
* [dustynv/ros:foxy-pytorch-l4t-r35.3.1](https://hub.docker.com/r/dustynv/ros/tags)

Mekala02, 2023-09-14:
Thanks, using r35.2.1 solved the issue.

closed 2023-09-14 by Mekala02