dusty-nv/jetson-containers #286: Prebuilt container for Ros Iron and jetson inference

Fibo27, 2023-09-14:
Hi @dusty-nv 
I am visiting your repo after a while - the revamped version looks well organized - kudos for that. 

Do you have a prebuilt container that i can download from the hub that has Ros iron and Jetson Inference included - I want to use it on a Orin Nano

Thanks

dusty-nv, 2023-09-14:
@Fibo27 sorry I don't - haven't integrated that build into new jetson-containers yet.  If you switch back to the `legacy` branch though, you could try building it with the `scripts/docker_build_ros.sh --distro=iron --with-pytorch`

Fibo27, 2023-09-14:
ok - i would presume then I have to add "iron" in line 47 of [https://github.com/dusty-nv/jetson-containers/blob/bc8d0264ef25aa0d1d25a54e4658f491d2fa130f/scripts/docker_build_ros.sh ](url) - is that correct?

How about if I were to build a container using your latest script and then commit the changes using a new image name on dockerhub?

dusty-nv, 2023-09-14:
The legacy branch already has iron in it:
https://github.com/dusty-nv/jetson-containers/blob/legacy/scripts/docker_build_ros.sh
https://github.com/dusty-nv/jetson-containers/blob/legacy/scripts/ros_distro.sh

jiefengsun, 2023-09-17:
@dusty-nv, thanks for the well-organized updates!
I would like to ask a few related questions:
1. Is is still true: `The package options are: ros_base, ros_core, and desktop - note that the ROS2 Desktop packages only build on JetPack 5.x.`  in the `readme.md`. Because I see: dustynv/ros:humble-pytorch-l4t-r32.7.1 appeared 10 days ago.. I am wondering what it is based on,  (the base, core, or desktop version)?
3. Is there a modularized way to build `jetson-inference model` as a package that we can select? 
4. I saw that it has `jetson-inference` package built in [humble-pytorch-l4t-r32.7.1](https://hub.docker.com/layers/dustynv/ros/humble-pytorch-l4t-r32.7.1/images/sha256-90f184c0930e4aeb6ae8f62326746278fe40cf80659a72aa903c7233e666992c?context=explore). Is this true for all these with `pytorch` tag?

I would like to have a container like: `humble-desktop-pytorch-jetsoninference-l4t-r32.7.1`. 
With the current code, I can easily build 'humble-desktop-pytorch-l4t-r32.7.1`, but it seems not straightforward to add the `jetson-inference` library.

I am not sure what is the most convenient way to do this right now since the structure of building containers changed a lot.

dusty-nv, 2023-09-17:
1. It's based on ros-base
2. Not yet, but I plan on adding it.  In the jetson-inference `dev` branch I have begun migrating it
3. Yes, the pytorch tags actually have jetson-inference in them (in addition to pytorch)

For now, I would just revert to the jetson-containers `legacy` branch and you can continue building `ros:*-pytorch` containers the existing way.  Hope that helps!

jiefengsun, 2023-09-18:
@dusty-nv 
Thanks!  [humble-pytorch-l4t-r32.7.1](https://hub.docker.com/layers/dustynv/ros/humble-pytorch-l4t-r32.7.1/images/sha256-90f184c0930e4aeb6ae8f62326746278fe40cf80659a72aa903c7233e666992c?context=explore) works for me. I am able to pull it without building it locally. 

I  am looking forward to `dev` updates on building `jetson inference` as a package in containers. It will be a really useful function!

.

closed 2023-09-21 by Fibo27