dusty-nv/jetson-containers #287: libopencv-dev not installed

tiesus, 2023-09-20:
We are using the following Docker base image to build our ros application: `dustynv/ros:noetic-ros-base-l4t-r34.1.1`
Building our application upon this base we are encountering the following issue: 

` ros-noetic-cv-bridge : Depends: libopencv-dev but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
Reading package lists...
Building dependency tree...
Reading state information...
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 libpcl-dev : Depends: libflann-dev but it is not going to be installed
              Depends: libvtk7-dev but it is not going to be installed
              Depends: libqhull-dev but it is not going to be installed
              Depends: libvtk7-qt-dev but it is not going to be installed
              Depends: libopenni-dev but it is not going to be installed`

Any hints how this can be solved? May of our packages need libopencv-dev. @MeHelge

dusty-nv, 2023-09-20:
@tiesus OpenCV is actually installed, with CUDA enabled, it was just built/installed from source via deb packages as opposed to apt.  So the name gets different.

ROS Noetic is also built from source in container, so you probably just want to add your package to the rosinstall_generator command here:

https://github.com/dusty-nv/jetson-containers/blob/5dece56e74b6cc14ced9557cc103affcdde728c1/packages/ros/Dockerfile.ros.noetic#L61

rather than mix-and-match ROS apt packages with the source installation

tiesus, 2023-09-21:
Thank's for the quick response. Try out later, but pretty sure this solves my issue.

closed 2023-09-21 by tiesus

nakai-omer, 2024-01-04:
@dusty-nv We are also having this issue just for different packages. Any package that depends on opencv throws the above error, I know we could just purge opencv and install it using apt, but we would like to get the cuda enabled opencv.

dusty-nv, 2024-01-05:
Hi @nakai-omer, here is one such example of where I reinstall opencv_cuda after it was purged due to the ubuntu version in apt:

https://github.com/dusty-nv/jetson-containers/blob/cb6c847f88df221e705397a1ee98424c2e893243/packages/diffusion/stable-diffusion-webui/Dockerfile#L24

I keep that `install_opencv.sh` script under /opt for these kind of scenarios so that it can be re-installed as needed.

nakai-omer, 2024-01-06:
Ah nice, so first purge, install all  apt deps, and then reinstall, ok will try it out thanks a lot!

Arief-AK, 2024-06-12:
> @tiesus OpenCV is actually installed, with CUDA enabled, it was just built/installed from source via deb packages as opposed to apt. So the name gets different.
>
> ROS Noetic is also built from source [...]

@dusty-nv Hello, beginner here, just need your confirmation on this.

So is the solution something like this? Where `ros-noetic-cv-bridge` is placed in the command?
```shell
rosinstall_generator ${ROS_PKG} vision_msgs image_transport ros-noetic-cv-bridge --rosdistro ${ROS_DISTRO} --deps --tar > ${ROS_DISTRO}-${ROS_PKG}.rosinstall && \
```