dusty-nv/jetson-containers #299: Not download models on llamaspeak

johnnynunez, 2023-10-05:
Hi @dusty-nv, I think that your path model is not indexed.

johnnynunez, 2023-10-05:
.bin models are deprectated, use instead codellama-7b-.Q4_K_M.gguf

dusty-nv, 2023-10-06:
Thanks @johnnynunez, yes I need to update the docs for llamaspeak.  Actually I am in the process of changing the LLM backend from oogabooga to MLC/TVM to get ~double the performance, and also integrate multimodality and Llava models.  Will update soon hopefully!

johnnynunez, 2023-10-06:
> Thanks @johnnynunez, yes I need to update the docs for llamaspeak. Actually I am in the process of changing the LLM backend from oogabooga to MLC/TVM to get ~double the performance, and also integrate multimodality and Llava models. Will update soon hopefully!

Yes, It’s working nice with llama7b
[image]

closed 2023-10-06 by johnnynunez