dusty-nv/jetson-containers #302: support for network enabled realsense

gyillikci, 2023-10-09:
Hello,

I am currently using a industrial version of d455 which is based on Ethernet based connectivity not USB based. I would the use the container but there is no support as far as I see. 

Any help would be appreciated

dusty-nv, 2023-10-09:
@gyillikci I don't have a d455 and aren't sure what is required to enable ethernet in librealsense, but presumably you can enable it in the realsense dockerfile here:

https://github.com/dusty-nv/jetson-containers/blob/master/packages/realsense/Dockerfile

gyillikci, 2023-10-09:
thanks Dustin,
i am compiling wit the flag of -DBUILD_NETWORK_DEVICE=ON and omitting -DFORCE_RSUSB_BACKEND=true just I did successive compiling on my Orin NX. I also omit the copiying udev rules.  would it be relevant to modify run.sh in order to add network parameter? 
[image]

my dockerfile is as follows:
`#---
# name: realsense
# group: sensors
# depends: [python, cmake]
# test: test.py
# notes: https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md
#---
ARG BASE_IMAGE
FROM ${BASE_IMAGE}

RUN apt-get update && \
    apt-get install -y --no-install-recommends \
		  libssl-dev \
		  libusb-1.0-0-dev \
            libgtk-3-dev \
            libglfw3-dev \
		  libgl1-mesa-dev \
		  libglu1-mesa-dev \
		  qtcreator && \
    if [ $(lsb_release -cs) = "bionic" ]; then \
        apt-get install -y --no-install-recommends python-dev; \
    else \
        apt-get install -y --no-install-recommends python2-dev; \
    fi \
    && rm -rf /var/lib/apt/lists/* \
    && apt-get clean

# https://github.com/IntelRealSense/librealsense/issues/11931
ARG LIBREALSENSE_VERSION=development

RUN git clone --branch v2.50.0 --depth=1 https://github.com/IntelRealSense/librealsense && \
    cd librealsense && \
    mkdir build && \
    cd build && \
    cmake \
        -DBUILD_EXAMPLES=true \
        -DBUILD_NETWORK_DEVICE=ON \
	   -DBUILD_WITH_CUDA=true \
	   -DCMAKE_BUILD_TYPE=release \
	   -DBUILD_PYTHON_BINDINGS=bool:true \
	   -DPYTHON_EXECUTABLE=/usr/bin/python3 \
	   -DPYTHON_INSTALL_DIR=$(python3 -c 'import sys; print(f"/usr/lib/python{sys.version_info.major}.{sys.version_info.minor}/dist-packages")') \
	   ../ && \
    make -j$(($(nproc)-1)) && \
    make install 
    #cd ../ && \
    #cp ./config/99-realsense-libusb.rules /etc/udev/rules.d/ && \
    #rm -rf librealsense
  
RUN python3 -c 'import pyrealsense2'

#RUN udevadm control --reload-rules && udevadm trigger
`

dusty-nv, 2023-10-09:
> would it be relevant to modify run.sh in order to add network parameter?

run.sh automatically adds `--network host`, but you can change that behavior here if you want:

https://github.com/dusty-nv/jetson-containers/blob/f46caf6843bf248258c91cf0c39bfdd3217d35fa/run.sh#L49

closed 2025-02-21 by johnnynunez