dusty-nv/jetson-containers #305: When Building Pytorch:1.12 I am getting GPG issues from build essentials base image

dustycodes, 2023-10-18:
Just running:
`./build.sh pytorch:1.12`
It worked last friday but now it seems like there are some repos that are down. No issues updating on my orin just in the containers. I'll try again later but I thought I would report an issue.

Here is my log:
```
docker build --network=host --tag pytorch:1.12-r35.2.1-build-essential \
--file /home/paladin/jetson-containers/packages/build-essential/Dockerfile \
--build-arg BASE_IMAGE=nvcr.io/nvidia/l4t-jetpack:r35.2.1 \
/home/paladin/jetson-containers/packages/build-essential \
2>&1 | tee /home/paladin/jetson-containers/logs/20231018_123903/build/pytorch_1.12-r35.2.1-build-essential.txt; exit ${PIPESTATUS[0]}

DEPRECATED: The legacy builder is deprecated and will be removed in a future release.
            Install the buildx component to build images with BuildKit:
            https://docs.docker.com/go/buildx/

Sending build context to Docker daemon  12.29kB
Step 1/5 : ARG BASE_IMAGE
Step 2/5 : FROM ${BASE_IMAGE}
 ---> c36b76efa9e5
Step 3/5 : ENV DEBIAN_FRONTEND=noninteractive
 ---> Using cache
 ---> caaff82891e3
Step 4/5 : RUN apt-get update &&        apt-get install --fix-missing &&        apt-get install -y --no-install-recommends      build-essential         software-properties-common      apt-transport-https     ca-certificates         lsb-release     pkg-config      gnupg   git     wget    curl nano     zip     unzip   && rm -rf /var/lib/apt/lists/*  && apt-get clean
 ---> Running in ec09bf1c6dd2
Get:1 https://repo.download.nvidia.com/jetson/common r35.2 InRelease [2555 B]
Err:1 https://repo.download.nvidia.com/jetson/common r35.2 InRelease
  Couldn't create temporary file /tmp/apt.conf.WraApg for passing config to apt-key
Hit:2 http://ports.ubuntu.com/ubuntu-ports focal InRelease
Err:2 http://ports.ubuntu.com/ubuntu-ports focal InRelease
  Couldn't create temporary file /tmp/apt.conf.rCbpqj for passing config to apt-key
Get:3 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease [114 kB]
Err:3 http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease
  Couldn't create temporary file /tmp/apt.conf.kxibZm for passing config to apt-key
Get:4 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease [108 kB]
Err:4 http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease
  Couldn't create temporary file /tmp/apt.conf.gpQXBs for passing config to apt-key
Get:5 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease [114 kB]
Err:5 http://ports.ubuntu.com/ubuntu-ports focal-security InRelease
  Couldn't create temporary file /tmp/apt.conf.VS2WNw for passing config to apt-key
Fetched 339 kB in 1s (250 kB/s)
Reading package lists...
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: https://repo.download.nvidia.com/jetson/common r35.2 InRelease: Couldn't create temporary file /tmp/apt.conf.WraApg for passing config to apt-key
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://ports.ubuntu.com/ubuntu-ports focal InRelease: Couldn't create temporary file /tmp/apt.conf.rCbpqj for passing config to apt-key
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://ports.ubuntu.com/ubuntu-ports focal-updates InRelease: Couldn't create temporary file /tmp/apt.conf.kxibZm for passing config to apt-key
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://ports.ubuntu.com/ubuntu-ports focal-backports InRelease: Couldn't create temporary file /tmp/apt.conf.gpQXBs for passing config to apt-key
W: An error occurred during the signature verification. The repository is not updated and the previous index files will be used. GPG error: http://ports.ubuntu.com/ubuntu-ports focal-security InRelease: Couldn't create temporary file /tmp/apt.conf.VS2WNw for passing config to apt-key
W: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/dists/focal/InRelease  Couldn't create temporary file /tmp/apt.conf.rCbpqj for passing config to apt-key
W: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/dists/focal-updates/InRelease  Couldn't create temporary file /tmp/apt.conf.kxibZm for passing config to apt-key
W: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/dists/focal-backports/InRelease  Couldn't create temporary file /tmp/apt.conf.gpQXBs for passing config to apt-key
W: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/dists/focal-security/InRelease  Couldn't create temporary file /tmp/apt.conf.VS2WNw for passing config to apt-key
W: Failed to fetch https://repo.download.nvidia.com/jetson/common/dists/r35.2/InRelease  Couldn't create temporary file /tmp/apt.conf.WraApg for passing config to apt-key
W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists...
Building dependency tree...
Reading state information...
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Reading package lists...
Building dependency tree...
Reading state information...
build-essential is already the newest version (12.8ubuntu1.1).
build-essential set to manually installed.
ca-certificates is already the newest version (20211016ubuntu0.20.04.1).
gnupg is already the newest version (2.2.19-3ubuntu2.2).
gnupg set to manually installed.
wget is already the newest version (1.20.3-1ubuntu2).
The following additional packages will be installed:
  distro-info-data gir1.2-glib-2.0 gir1.2-packagekitglib-1.0 git-man
  libappstream4 libcurl4 liberror-perl libgirepository-1.0-1 libglib2.0-bin
  libglib2.0-data liblmdb0 libpackagekit-glib2-18 libpolkit-agent-1-0
[... 79 lines cut ...]
E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/d/distro-info-data/distro-info-data_0.43ubuntu1.11_all.deb  404  Not Found [IP: 185.125.190.39 80]
E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/g/glib2.0/libglib2.0-data_2.64.6-1~ubuntu20.04.4_all.deb  404  Not Found [IP: 185.125.190.39 80]
E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/p/python-apt/python-apt-common_2.0.1_all.deb  404  Not Found [IP: 185.125.190.39 80]
E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/p/python-apt/python3-apt_2.0.1_arm64.deb  404  Not Found [IP: 185.125.190.39 80]
E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/c/curl/libcurl4_7.68.0-1ubuntu2.15_arm64.deb  404  Not Found [IP: 185.125.190.39 80]
E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/c/curl/curl_7.68.0-1ubuntu2.15_arm64.deb  404  Not Found [IP: 185.125.190.39 80]
E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/g/git/git-man_2.25.1-1ubuntu3.8_all.deb  404  Not Found [IP: 185.125.190.39 80]
E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/g/git/git_2.25.1-1ubuntu3.8_arm64.deb  404  Not Found [IP: 185.125.190.39 80]
E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/g/glib2.0/libglib2.0-bin_2.64.6-1~ubuntu20.04.4_arm64.deb  404  Not Found [IP: 185.125.190.39 80]
E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/s/software-properties/python3-software-properties_0.99.9.10_all.deb  404  Not Found [IP: 185.125.190.39 80]
E: Failed to fetch http://ports.ubuntu.com/ubuntu-ports/pool/main/s/software-properties/software-properties-common_0.99.9.10_all.deb  404  Not Found [IP: 185.125.190.39 80]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
The command '/bin/sh -c apt-get update &&       apt-get install --fix-missing &&        apt-get install -y --no-install-recommends      build-essential         software-properties-common      apt-transport-https     ca-certificates         lsb-release     pkg-config      gnupg   git     wget curl     nano    zip     unzip   && rm -rf /var/lib/apt/lists/*  && apt-get clean' returned a non-zero code: 100
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/paladin/jetson-containers/jetson_containers/build.py", line 96, in <module>
    build_container(args.name, args.packages, args.base, args.build_flags, args.simulate, args.skip_tests, args.test_only, args.push, args.no_github_api)
[... cut ...]
    raise CalledProcessError(retcode, process.args,
subprocess.CalledProcessError: Command 'docker build --network=host --tag pytorch:1.12-r35.2.1-build-essential --file /home/paladin/jetson-containers/packages/build-essential/Dockerfile --build-arg BASE_IMAGE=nvcr.io/nvidia/l4t-jetpack:r35.2.1 /home/paladin/jetson-containers/packages/build-essential 2>&1 | tee /home/paladin/jetson-containers/logs/20231018_123903/build/pytorch_1.12-r35.2.1-build-essential.txt; exit ${PIPESTATUS[0]}' returned non-zero exit status 100.
```

dusty-nv, 2023-10-18:
Hi @dustycodes, I just did a rebuild of this container with the command below, and it succeeded without issue:

```bash
./build.sh --name=build/ --build-flags='--no-cache' pytorch:1.12
```

It looks like there were some `404  Not Found [IP: 185.125.190.39 80]` errors in your log - maybe it was a temporary networking issue somewhere?

dustycodes, 2023-10-19:
That's what I was hoping. Thanks for verifying. I'll check again

closed 2023-10-19 by dustycodes