dusty-nv/jetson-containers #309: ROS2 humble, not able to install common ros binaries?

PaddyCube, 2023-10-22:
Hi there, 

I was able to run ROSS humble container 32.7.1. I need to install additional packages, which I commonly install as binary by ```apt install ros-humble-....```

but I always get 

```unable to localte package...```

So you build ROS from source in these containers? Are they based on Ubuntu20 instead of 22.04?

I spend a lot of time with my Jetson Nano and it seems NVidia don't want to let me use it for something real.

Can't we just simply install a common Ubuntu22.04 server image to SD card? I don't need ML or OpenCV

mischa-ai, 2023-12-07:
I have the same problem for **Ros2 Iron** image (dustynv/ros:iron-ros-base-l4t-r32.7.1), trying to use the `ros-iron-ros2-control` packages. After spending hours trying to figure out why it doesn't work (again) , it seems that even the containers are running Ubuntu 18 !

```
# lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.6 LTS
Release:	18.04
Codename:	bionic
```

So basically everything is built from source inside the containers and if you want to use something you have to figure out how to build it from source. Unfortunately that is not so easy ...

Another option is to use **Ros2 Eloquent** images , that aims for Ubuntu 18 and `apt install` should work, but is outdated since 2020 (see [here](https://github.com/ros-infrastructure/rep/blob/master/rep-2000.rst#eloquent-elusor-november-2019---november-2020))

So yeah, go and get an RPi if you don't need a GPU for machine learning on edge. Safe yourself a lot of time and frustration - I wish someone told me this months ago, because in all the 'getting started' videos it looks so easy, but I spent more time on setting up the Nano than actually doing anything on it.

dusty-nv, 2023-12-07:
@mischa-ai @PaddyCube if you don't need GPU acceleration, you can just use the upstream OSRF aarch64 containers using whatever base OS and the ROS packages from apt:

* https://github.com/osrf/docker_images
* https://hub.docker.com/r/arm64v8/ros/tags

if you want GPU acceleration, then you can compile ROS packages from source using rosinstall_generator / rosdep like here:

* https://github.com/dusty-nv/jetson-containers/blob/c5efc12d51e42bf56e2c4408933ce47e4d1c2483/packages/ros/ros2_build.sh#L79
* https://github.com/dusty-nv/jetbot_ros/blob/d8e5ee1b17f5c66d038017b86f8920a496197ea9/Dockerfile#L72

mischa-ai, 2023-12-07:
@dusty-nv thank you the link, I will try it out.

Yes that is exactly my problem, I want to run AI on the Nano.

I am using OSRF images on my notebook where everything works fine - so my question would be: why is it  not possible to use ubuntu 22.04 as base image for the nvidia images? I guess driver issues?

closed 2025-02-21 by johnnynunez