dusty-nv/jetson-containers #323: Build ROS Iron on Jetson Nano fails with opencv image build ImportError

michael-schaefer-eu, 2023-11-06:
I have a Jetson Nano on which I try to build the ROS Iron which fails.

Jetpack 4.6.4
L4T 32.7.4
Cuda 10.2.300
OpenCV 4.1.1

When I run `./build.sh ros:iron-ros-base` I get the following error when the `opencv` image is built:

```
#8 79.12 Traceback (most recent call last):
#8 79.12   File "<string>", line 1, in <module>
#8 79.12   File "/usr/local/lib/python3.6/dist-packages/cv2/__init__.py", line 96, in <module>
#8 79.12     bootstrap()
#8 79.12   File "/usr/local/lib/python3.6/dist-packages/cv2/__init__.py", line 86, in bootstrap
#8 79.12     import cv2
#8 79.12 ImportError: libcublas.so.10: cannot open shared object file: No such file or directory
#8 ERROR: process "/bin/sh -c cd /opt && ./opencv_install.sh" did not complete successfully: exit code: 1
```

From the logs:

logs/20231106_122338/build/opencv_r32.7.4-opencv.sh

```
#!/usr/bin/env bash

docker build --network=host --tag opencv:r32.7.4-opencv \
--file /home/jetbot/jetbot-usb/jetson-containers/packages/opencv/Dockerfile \
--build-arg BASE_IMAGE=opencv:r32.7.4-numpy \
--build-arg OPENCV_URL="https://nvidia.box.com/shared/static/5v89u6g5rb62fpz4lh0rz531ajo2t5ef.gz" \
--build-arg OPENCV_DEB="OpenCV-4.5.0-aarch64.tar.gz" \
/home/jetbot/jetbot-usb/jetson-containers/packages/opencv \
2>&1 | tee /home/jetbot/jetbot-usb/jetson-containers/logs/20231106_122338/build/opencv_r32.7.4-opencv.txt; exit ${PIPESTATUS[0]}
```

logs/20231106_122338/build/opencv_r32.7.4-opencv.txt

```
#1 [internal] load .dockerignore
#1 transferring context: 2B done
#1 DONE 0.1s

#2 [internal] load build definition from Dockerfile
#2 transferring dockerfile: 677B 0.0s done
#2 DONE 0.1s

#3 [internal] load metadata for docker.io/library/opencv:r32.7.4-numpy
#3 DONE 0.0s

#4 [internal] load build context
#4 transferring context: 39B done
#4 DONE 0.1s

#5 [1/4] FROM docker.io/library/opencv:r32.7.4-numpy
#5 DONE 0.4s

#6 [2/4] COPY opencv_install.sh /opt/opencv_install.sh
#6 DONE 0.1s

#7 [3/4] RUN cd /opt &&     sed -i "2i OPENCV_URL=https://nvidia.box.com/shared/static/5v89u6g5rb62fpz4lh0rz531ajo2t5ef.gz" opencv_install.sh &&     sed -i "3i OPENCV_DEB=OpenCV-4.5.0-aarch64.tar.gz" opencv_install.sh &&     cat opencv_install.sh
#7 0.249 #!/usr/bin/env bash
#7 0.249 OPENCV_URL=https://nvidia.box.com/shared/static/5v89u6g5rb62fpz4lh0rz531ajo2t5ef.gz
#7 0.249 OPENCV_DEB=OpenCV-4.5.0-aarch64.tar.gz
#7 0.249 OPENCV_URL=${1:-"$OPENCV_URL"}
#7 0.249 OPENCV_DEB=${2:-"$OPENCV_DEB"}
#7 0.249 
#7 0.249 echo "OPENCV_URL=$OPENCV_URL"
#7 0.249 echo "OPENCV_DEB=$OPENCV_DEB"
#7 0.249 
[... 252 lines cut ...]
------
 > [4/4] RUN cd /opt && ./opencv_install.sh:
#8 78.54 testing cv2 module under python...
#8 78.54 + echo 'testing cv2 module under python...'
#8 78.54 + python3 -c 'import cv2; print('\''OpenCV version:'\'', str(cv2.__version__)); print(cv2.getBuildInformation())'
#8 79.12 Traceback (most recent call last):
#8 79.12   File "<string>", line 1, in <module>
#8 79.12   File "/usr/local/lib/python3.6/dist-packages/cv2/__init__.py", line 96, in <module>
#8 79.12     bootstrap()
#8 79.12   File "/usr/local/lib/python3.6/dist-packages/cv2/__init__.py", line 86, in bootstrap
#8 79.12     import cv2
#8 79.12 ImportError: libcublas.so.10: cannot open shared object file: No such file or directory
------
Dockerfile:25
--------------------
  23 |         cat opencv_install.sh
  24 |     
  25 | >>> RUN cd /opt && ./opencv_install.sh
  26 |     
--------------------
ERROR: failed to solve: process "/bin/sh -c cd /opt && ./opencv_install.sh" did not complete successfully: exit code: 1

```

Is it not possible to build ROS Iron on Jetson Nano or am I doing something wrong?

dusty-nv, 2023-11-07:
Hi @michael-schaefer-eu, have you set your default-docker runtime to nvidia like here?

https://github.com/dusty-nv/jetson-containers/blob/master/docs/setup.md#docker-default-runtime

Do you have CUDA Toolkit installed on your device, outside container?   You should find that libcublas.so.10 file under `/usr/local/cuda/lib64/libcublas` ?   On JetPack 4, CUDA/cuDNN/TensorRT are mounted from the host device.

Are you able to run pre-built l4t-pytorch container and `import torch` ?   If not, the nvidia container runtime isn't working properly.

michael-schaefer-eu, 2023-11-07:
Hi @dusty-nv , thank you very much for your help.

Yes, docker runtime is set to nvidia:

```
sudo docker info | grep 'Default Runtime'
Default Runtime: nvidia
```

Cuda Toolkit is installed on the host:

```
ls -hal /usr/local/cuda/lib64/ | grep libcublas
lrwxrwxrwx 1 root root   17 Mar  1  2021 libcublasLt.so -> libcublasLt.so.10
lrwxrwxrwx 1 root root   25 Mar  1  2021 libcublasLt.so.10 -> libcublasLt.so.10.2.3.300
-rw-r--r-- 1 root root  33M Mar  1  2021 libcublasLt.so.10.2.3.300
-rw-r--r-- 1 root root  35M Mar  1  2021 libcublasLt_static.a
lrwxrwxrwx 1 root root   15 Mar  1  2021 libcublas.so -> libcublas.so.10
lrwxrwxrwx 1 root root   23 Mar  1  2021 libcublas.so.10 -> libcublas.so.10.2.3.300
-rw-r--r-- 1 root root  78M Mar  1  2021 libcublas.so.10.2.3.300
-rw-r--r-- 1 root root  93M Mar  1  2021 libcublas_static.a
```

Also OpenCV (if that matters):

```
python3 -c "import cv2; print('OpenCV version:', str(cv2.__version__));"
OpenCV version: 4.1.1
```

I will try out the torch container as soon as possible, I guess it's that one: [dustynv/pytorch:1.10-r32.7.1](https://hub.docker.com/layers/dustynv/pytorch/1.10-r32.7.1/images/sha256-a8aa00d70e4b4c1854e941d8d3782590b11c9cf50b825c76b355dc749bc4f5fd)

If the nvidia container runtime isn't working properly, can I fix that or do I have to reinstall the complete OS again?

michael-schaefer-eu, 2023-11-07:
Also `/usr/local/cuda/include/` contains no cudnn headers, is it possible that it has not been installed when I upgraded the system from L4T `32.5` to `32.7` ?

Is there a _script_ available to check all the required libs on the host? I couldn't find out anything.

dusty-nv, 2023-11-07:
If the container mounts aren't working on JetPack 4, typically I just reflash the system because they are tricky to debug/fix

michael-schaefer-eu, 2023-11-13:
thanks @dusty-nv , 

I **re-flashed** my jetson nano with the latest _Jetpack 4.6.4_ with _L4T 32.7.4_ (via the Docker SDK Manager, that alone took me a while to get it running ... )

The build process still failed with the same error as above, but this time at least the `run` worked for me:

`./run.sh dustynv/l4t-pytorch:r32.7.1`

and

`./run.sh dustynv/ros:iron-ros-base-l4t-r32.7.1`

and both can load CV2 inside the container :

```
root@jbot:/# python3 -c "import cv2; print('OpenCV version:', str(cv2.__version__));"
OpenCV version: 4.5.0
```

closed 2023-11-13 by michael-schaefer-eu