dusty-nv/jetson-containers #335: How to use CSI camera with gstreamer in l4t-ml container?

ottokruse, 2023-11-20:
I'm running into `WARNING: erroneous pipeline: no element "nvarguscamerasrc"`

This is my gstreamer pipeline, that works fine on the Jetson when not in a Jetson container:

```
nvarguscamerasrc sensor-id=0 exposuretimerange="1250000 1250000" maxperf=true ! video/x-raw(memory:NVMM), width=(int)1920, height=(int)1080, framerate=(fraction)60/1 ! nvvidconv flip-method=0 ! video/x-raw, width=(int)1920, height=(int)1080, format=(string)BGRx ! videoconvert ! video/x-raw,format=(string)BGR ! appsink
```

ottokruse, 2023-11-20:
Tracked this down to `ultralytics`.

If I import `ultralytics` before I start the video, I get the error. However, if I do it after I start the video, the gstreamer pipeline works, and frames start coming in. So apparently this library has some side effect that is hampering me.

Working code:

```python
import cv2
import signal
import time
from pathlib import Path
#from ultralytics import YOLO # DO NOT YET IMPORT HERE

here = Path(__file__).parent

video = cv2.VideoCapture(
    'nvarguscamerasrc sensor-id=0 exposuretimerange="1000000 1250000" ! video/x-raw(memory:NVMM), width=(int)1920, height=(int)1080, framerate=(fraction)60/1 ! nvvidconv flip-method=0 ! video/x-raw, width=(int)1920, height=(int)1080, format=(string)BGRx ! videoconvert ! video/x-raw,format=(string)BGR ! appsink',
    cv2.CAP_GSTREAMER,
)
if not video.isOpened():
    raise Exception("Failed to open camera")

from ultralytics import YOLO # MOVED HERE
model = YOLO(here / "best.pt").to("cuda")

print("Capturing video ...")

run = [True]

def stop(*_):
    run[0] = False

signal.signal(signal.SIGINT, stop)

try:
    while run[0]:
        ret, frame = video.read()

        start = time.time()
        results = model.predict(frame)
        print(f"Inference took {(time.time() - start) * 1000:.2f} ms.")

        if not ret:
            raise Exception("Unexpected end of video stream")

        frame_nr = int(video.get(cv2.CAP_PROP_POS_FRAMES))
        print(f"Frame {frame_nr} ({frame.nbytes})")

finally:
    video.release()

```

closed 2023-11-20 by ottokruse

dusty-nv, 2023-11-20:
Thanks @ottokruse, I have seen weirdness with `cv2` module before and the order in which it is imported.  This doesn't seem specific to aarch64/Jetson, but rather the OpenCV module itself.