dusty-nv/jetson-containers #336: Navigation2 with Humble container on Jetson Nano

elgarbe, 2023-11-21:
Is it possible to build navigation2 stack on humble container on a jetson nano?

dusty-nv, 2023-11-21:
Hi @elgarbe, I just tried following the nav2 build instructions from https://navigation.ros.org/development_guides/build_docs/index.html#released-distribution-binaries using `dustynv/ros:humble-desktop-l4t-r32.7.1` , but was met with errors during rosdep like:

```bash
$ rosdep install -y --from-paths ./src --ignore-src

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
nav2_bt_navigator: No definition of [behaviortree_cpp_v3] for OS version [bionic]
nav2_smac_planner: No definition of [ompl] for OS version [bionic]
nav2_util: No definition of [bond] for OS version [bionic]
nav2_behavior_tree: No definition of [behaviortree_cpp_v3] for OS version [bionic]
nav2_bringup: No definition of [slam_toolbox] for OS version [bionic]
nav2_lifecycle_manager: No definition of [diagnostic_updater] for OS version [bionic]
nav2_system_tests: No definition of [gazebo_ros_pkgs] for OS version [bionic]
nav2_mppi_controller: [xsimd] defined as "not available" for OS version [bionic]
```

So instead, I try it like this using `rosinstall_generator` (and excluding [xsimd](https://github.com/xtensor-stack/xsimd) package, which could not be found)

```
# create a workspace in a directory that you have mounted into the container
cd /mount
mkdir -p ros2_ws/src
cd ros2_ws

# pull the nav2 sources
source /opt/ros/humble/install/setup.bash
export ROS_PACKAGE_PATH=${AMENT_PREFIX_PATH}
rosinstall_generator --deps --exclude RPP --rosdistro humble navigation2 > ros2.humble.nav2.rosinstall

# install dependencies
apt-get update
rosdep install -y --ignore-src --from-paths src --rosdistro humble --skip-keys "xsimd"

# build nav2
colcon build --symlink-install --base-paths src --event-handlers console_direct+
```

It's building now, but will see if it hits an error around xsimd, and if that needs to be installed manually.

elgarbe, 2023-11-21:
> Hi @elgarbe, I just tried following the nav2 build instructions from https://navigation.ros.org/development_guides/build_docs/index.html#released-distribution-binaries using `dustynv/ros:humble-desktop-l4t-r32.7.1` , but was met with errors during rosdep like:
> 
> ```shell
> $ rosdep install -y --from-paths ./src --ignore-src
> 
> ERROR: the following packages/stacks could not have their rosdep keys resolved
> to system dependencies:
> nav2_bt_navigator: No definition of [behaviortree_cpp_v3] for OS version [bionic]
> nav2_smac_planner: No definition of [ompl] for OS version [bionic]
> nav2_util: No definition of [bond] for OS version [bionic]
> nav2_behavior_tree: No definition of [behaviortree_cpp_v3] for OS version [bionic]
> nav2_bringup: No definition of [slam_toolbox] for OS version [bionic]
> nav2_lifecycle_manager: No definition of [diagnostic_updater] for OS version [bionic]
> nav2_system_tests: No definition of [gazebo_ros_pkgs] for OS version [bionic]
> nav2_mppi_controller: [xsimd] defined as "not available" for OS version [bionic]
> ```
> 
> So instead, I try it like this using `rosinstall_generator` (and excluding [xsimd](https://github.com/xtensor-stack/xsimd) package, which could not be found)
> 
> ```
> # create a workspace in a directory that you have mounted into the container
> cd /mount
> mkdir -p ros2_ws/src
> cd ros2_ws
> 
> # pull the nav2 sources
> source /opt/ros/humble/install/setup.bash
> export ROS_PACKAGE_PATH=${AMENT_PREFIX_PATH}
> rosinstall_generator --deps --exclude RPP --rosdistro humble navigation2 > ros2.humble.nav2.rosinstall
> 
> # install dependencies
> apt-get update
> rosdep install -y --ignore-src --from-paths src --rosdistro humble --skip-keys "xsimd"
> 
> # build nav2
> colcon build --symlink-install --base-paths src --event-handlers console_direct+
> ```
> 
> It's building now, but will see if it hits an error around xsimd, and if that needs to be installed manually.

Is there some missing commands on your post? I've tried the same and `rosinstall_generator --deps --exclude RPP --rosdistro humble navigation2 > ros2.humble.nav2.rosinstall` generate the rosinstall file with all the repos, but there isn't command to pull them...

dusty-nv, 2023-11-21:
Ah sorry, yes - `vcs import src/ < ros2.humble.nav2.rosinstall`

It failed to build though:

```
In file included from /opt/ros/humble/install/include/rclcpp/rclcpp/callback_group.hpp:24:0,
                 from /opt/ros/humble/install/include/rclcpp/rclcpp/any_executable.hpp:20,
                 from /opt/ros/humble/install/include/rclcpp/rclcpp/memory_strategy.hpp:25,
                 from /opt/ros/humble/install/include/rclcpp/rclcpp/memory_strategies.hpp:18,
                 from /opt/ros/humble/install/include/rclcpp/rclcpp/executor_options.hpp:20,
                 from /opt/ros/humble/install/include/rclcpp/rclcpp/executor.hpp:37,
                 from /opt/ros/humble/install/include/rclcpp/rclcpp/executors/multi_threaded_executor.hpp:25,
                 from /opt/ros/humble/install/include/rclcpp/rclcpp/executors.hpp:21,
                 from /opt/ros/humble/install/include/rclcpp/rclcpp/rclcpp.hpp:155,
                 from /mount/test/ros2_ws/src/navigation2/nav2_util/include/nav2_util/service_client.hpp:20,
                 from /mount/test/ros2_ws/src/navigation2/nav2_util/include/nav2_util/lifecycle_service_client.hpp:24,
                 from /mount/test/ros2_ws/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:15:
/opt/ros/humble/install/include/rclcpp/rclcpp/client.hpp: In instantiation of ‘std::optional<std::variant<std::promise<typename ServiceT::Response::SharedPtr>, std::tuple<std::function<void(std::shared_future<typename ServiceT::Response::SharedPtr>)>, std::shared_future<typename ServiceT::Response::SharedPtr>, std::promise<typename ServiceT::Response::SharedPtr> >, std::tuple<std::function<void(std::shared_future<std::pair<typename ServiceT::Request::SharedPtr, typename ServiceT::Response::SharedPtr> >)>, typename ServiceT::Request::SharedPtr, std::shared_future<std::pair<typename ServiceT::Request::SharedPtr, typename ServiceT::Response::SharedPtr> >, std::promise<std::pair<typename ServiceT::Request::SharedPtr, typename ServiceT::Response::SharedPtr> > > > > rclcpp::Client<ServiceT>::get_and_erase_pending_request(int64_t) [with ServiceT = lifecycle_msgs::srv::GetState; typename ServiceT::Response::SharedPtr = std::shared_ptr<lifecycle_msgs::srv::GetState_Response_<std::allocator<void> > >; typename ServiceT::Request::SharedPtr = std::shared_ptr<lifecycle_msgs::srv::GetState_Request_<std::allocator<void> > >; int64_t = long int]’:
/opt/ros/humble/install/include/rclcpp/rclcpp/client.hpp:551:38:   required from ‘void rclcpp::Client<ServiceT>::handle_response(std::shared_ptr<rmw_request_id_s>, std::shared_ptr<void>) [with ServiceT = lifecycle_msgs::srv::GetState]’
/mount/test/ros2_ws/src/navigation2/nav2_util/src/lifecycle_service_client.cpp:102:1:   required from here
/opt/ros/humble/install/include/rclcpp/rclcpp/client.hpp:821:12: error: could not convert ‘value’ from ‘std::variant<std::promise<std::shared_ptr<lifecycle_msgs::srv::GetState_Response_<std::allocator<void> > > >, std::tuple<std::function<void(std::shared_future<std::shared_ptr<lifecycle_msgs::srv::GetState_Response_<std::allocator<void> > > >)>, std::shared_future<std::shared_ptr<lifecycle_msgs::srv::GetState_Response_<std::allocator<void> > > >, std::promise<std::shared_ptr<lifecycle_msgs::srv::GetState_Response_<std::allocator<void> > > > >, std::tuple<std::function<void(std::shared_future<std::pair<std::shared_ptr<lifecycle_msgs::srv::GetState_Request_<std::allocator<void> > >, std::shared_ptr<lifecycle_msgs::srv::GetState_Response_<std::allocator<void> > > > >)>, std::shared_ptr<lifecycle_msgs::srv::GetState_Request_<std::allocator<void> > >, std::shared_future<std::pair<std::shared_ptr<lifecycle_msgs::srv::GetState_Request_<std::allocator<void> > >, std::shared_ptr<lifecycle_msgs::srv::GetState_Response_<std::allocator<void> > > > >, std::promise<std::pair<std::shared_ptr<lifecycle_msgs::srv::GetState_Request_<std::allocator<void> > >, std::shared_ptr<lifecycle_msgs::srv::GetState_Response_<std::allocator<void> > > > > > >’ to ‘std::optional<std::variant<std::promise<std::shared_ptr<lifecycle_msgs::srv::GetState_Response_<std::allocator<void> > > >, std::tuple<std::function<void(std::shared_future<std::shared_ptr<lifecycle_msgs::srv::GetState_Response_<std::allocator<void> > > >)>, std::shared_future<std::shared_ptr<lifecycle_msgs::srv::GetState_Response_<std::allocator<void> > > >, std::promise<std::shared_ptr<lifecycle_msgs::srv::GetState_Response_<std::allocator<void> > > > >, std::tuple<std::function<void(std::shared_future<std::pair<std::shared_ptr<lifecycle_msgs::srv::GetState_Request_<std::allocator<void> > >, std::shared_ptr<lifecycle_msgs::srv::GetState_Response_<std::allocator<void> > > > >)>, std::shared_ptr<lifecycle_msgs::srv::GetState_Request_<std::allocator<void> > >, std::shared_future<std::pair<std::shared_ptr<lifecycle_msgs::srv::GetState_Request_<std::allocator<void> > >, std::shared_ptr<lifecycle_msgs::srv::GetState_Response_<std::allocator<void> > > > >, std::promise<std::pair<std::shared_ptr<lifecycle_msgs::srv::GetState_Request_<std::allocator<void> > >, std::shared_ptr<lifecycle_msgs::srv::GetState_Response_<std::allocator<void> > > > > > > >’
     return value;
```

Will try some other things, but it occurred to me that I don't think nav2 uses GPU?   so yea, you could use it in a 20.04-based container and just install it the normal way (or use the upstream ROS containers)

elgarbe, 2023-11-21:
I made some progress.
The first one is [use this](https://github.com/ros2/rclcpp/issues/1963#issuecomment-1175880242). Edit `/opt/ros/rolling/src/ros2/rclcpp/rclcpp/include/rclcpp/client.hpp` and replace `return value;` with `return std::move(value);`
Then I have to install several package, I think that they has to be installed with `rosdep install` but I'm not sure why didn't happend.

```
apt-mark hold '*opencv*' && \
        apt-get update && apt-get install -y \
        build-essential \
        git \
        python3-colcon-common-extensions \
        python3-pip \
        python3-rosdep \
        python3-rosinstall-generator \
        python3-vcstool \
        xorg-dev \
        build-essential \
        libboost-dev \
        libboost-filesystem-dev \
        libboost-python-dev \
        libboost-system-dev \ 
        libboost-thread-dev \ 
        libboost-program-options-dev \ 
        libboost-test-dev \
        uuid-dev \
        qtbase5-dev \
        libgraphicsmagick++1-dev \
        wget 
```
I'm building a Dockerfile while running command on the docker container and solving problems.

What could it be the best way to make the change needed on client.hpp file?

BTW I have 35/58 package build right now.

elgarbe, 2023-11-22:
I'm trying to make a Dokerfile with nav2 build process, but I'm facing a problem:
```
FROM dustynv/ros:humble-ros-base-l4t-r32.7.1

RUN mkdir -p /root/nav2_ws/src
WORKDIR /root/nav2_ws

ENV ROS_PACKAGE_PATH=/opt/ros/humble/install

# ROS fundamentals
RUN apt-mark hold '*opencv*' \
    && apt-get update \
    && apt autoremove -y \
    && rosinstall_generator --deps --exclude RPP --rosdistro humble navigation2 > ros2.humble.nav2.rosinstall \
    && vcs import src/ < ros2.humble.nav2.rosinstall \
    # &&  sudo rosdep fix-permissions \
    &&  rosdep update

RUN rosdep install -y --ignore-src --from-paths src --rosdistro humble --skip-keys "xsimd" --skip-keys "ignition-cmake2" --skip-keys "ignition-math6"
```
The las RUN command returns several errors:
`ERROR: the following packages/stacks could not have their rosdep keys resolved`
in the image build process.

But, if I comment out that RUN command, build the image and run it, then inside the container a can run that command without errors.
What am I missing?

dusty-nv, 2023-11-22:
@elgarbe hmm, not entirely sure - if you put `rosdep install -y --ignore-src --from-paths src --rosdistro humble --skip-keys "xsimd" --skip-keys "ignition-cmake2" --skip-keys "ignition-math6"` in the RUN command above it (commented as "ROS fundamentals"), does it fail then?  If so, I think it would point to an issue with files that were removed in that RUN command, but persist after it (due to how docker filesystem layers work)

Or is perhaps some networking issue difference between `docker build` and `docker run` modes?

elgarbe, 2023-11-22:
> @elgarbe hmm, not entirely sure - if you put `rosdep install -y --ignore-src --from-paths src --rosdistro humble --skip-keys "xsimd" --skip-keys "ignition-cmake2" --skip-keys "ignition-math6"` in [...]

This:

```
# ROS fundamentals
RUN apt-mark hold '*opencv*' \
    && apt-get update \
    && apt autoremove -y \
    && rosinstall_generator --deps --exclude RPP --rosdistro humble navigation2 > ros2.humble.nav2.rosinstall \
    && vcs import src/ < ros2.humble.nav2.rosinstall \
    # &&  sudo rosdep fix-permissions \
    &&  rosdep update \
    &&  rosdep install -y --ignore-src --from-paths src --rosdistro humble --skip-keys "xsimd" --skip-keys "ignition-cmake2" --skip-keys "ignition-math6"
```
also fail.
I'm not sure why I have to run the image in bridge mode. Running in host mode doesn't work, I mean I don't have internet connection in host mode.

dusty-nv, 2023-11-22:
> I'm not sure why I have to run the image in bridge mode. Running in host mode doesn't work, I mean I don't have internet connection in host mode.

That's strange hmm, I haven't seen that before and always use `--network=host`.   I wonder if you can set defaults in `/etc/docker/daemon.json` so that your preferred network gets used during `docker build` operations:

https://docs.docker.com/network/drivers/bridge/#configure-the-default-bridge-network
https://docs.docker.com/engine/reference/commandline/dockerd/#daemon-configuration-file

Is your Jetson connected to the internet over wifi, ethernet, or are you on a micro-USB connection through a PC?

elgarbe, 2023-11-22:
I'm connected by ethernet cable to the same router as my PC.
Extrange that `vcs import src/ < ros2.humble.nav2.rosinstall` requier internet connection and works as expected, but `rosdep install -y --ignore-src --from-paths src --rosdistro humble --skip-keys "xsimd" --skip-keys "ignition-cmake2" --skip-keys "ignition-math6"` doesn't.

dusty-nv, 2023-11-22:
I wonder if rosdep uses different ports or something which are blocked or having some firewall issue, but yea strange.  Sorry I don't know how to fix that.

elgarbe, 2023-11-22:
I forget to source the ROS distro.... 
```

FROM dustynv/ros:humble-ros-base-l4t-r32.7.1

RUN mkdir -p /root/nav2_ws/src
WORKDIR /root/nav2_ws

ENV ROS_PACKAGE_PATH=/opt/ros/humble/install

# ROS fundamentals
RUN apt-mark hold '*opencv*' \
    && apt-get update \
    && apt autoremove -y \
    && rosinstall_generator --deps --exclude RPP --rosdistro humble navigation2 > ros2.humble.nav2.rosinstall \
    && vcs import src/ < ros2.humble.nav2.rosinstall \
    && . /opt/ros/$ROS_DISTRO/install/setup.sh \
    && rosdep fix-permissions \
    && rosdep update \
    && rosdep install -y --ignore-src --from-paths src --rosdistro humble --skip-keys "xsimd" --skip-keys "ignition-cmake2" --skip-keys "ignition-math6"
```

elgarbe, 2023-11-23:
https://github.com/ros-planning/navigation2/issues/3984

elgarbe, 2024-01-02:
@dusty-nv starting from a custom ubuntu 20.04 image (https://github.com/Qengineering/Jetson-Nano-Ubuntu-20-image) is there a way to run ros2 humble on docker and install nav2 package?

dusty-nv, 2024-01-02:
@elgarbe no idea about CUDA support on Nano 4GB with 20.04, it's unsupported.  If you don't need CUDA for those nodes, you can just use the official ROS images from https://hub.docker.com/_/ros/ for running nav2 in those.

elgarbe, 2024-01-02:
Oh!! yes I can use my nodes without CUDA.
I have limited experience with docker, thank for your patience answering my questions.

closed 2025-02-21 by johnnynunez