dusty-nv/jetson-containers #338: Issue: Python Version Compatibility with Jetson Orin AGX 64G and Text Generation WebUI

MrBrabus75, 2023-11-24:
### Environment Details
Hardware: Jetson Orin AGX 64G
PyTorch Version: [Python 3.8, aarch64 compatible](https://developer.download.nvidia.cn/compute/redist/jp/v511/pytorch/torch-2.0.0+nv23.05-cp38-cp38-linux_aarch64.whl)
Container Used: text-generation-webui from your containers.

### Issue Description
After installing the prerequisites, I executed the following command to run the Docker container for text-generation-webui:
`user@ubuntu:~/Desktop/jetson-containers$ ./run.sh $(./autotag dustynv/text-generation-webui) /bin/bash -c "cd /opt/text-generation-webui && python3 server.py --model-dir=/data/models/text-generation-webui --model=Intel_neural-chat-7b-v3-1 --loader=transformers --api --bf16 --verbose --listen --extension openai --auto-devices"
`
However, I encountered this error:
`
2023-11-24 13:05:12 ERROR:Failed to load the extension "openai".
Traceback (most recent call last):
  File "/opt/text-generation-webui/modules/extensions.py", line 36, in load_extensions
    exec(f"import extensions.{name}.script")
  File "<string>", line 1, in <module>
  File "/opt/text-generation-webui/extensions/openai/script.py", line 27, in <module>
    from .typing import (
  File "/opt/text-generation-webui/extensions/openai/typing.py", line 8, in <module>
    class GenerationOptions(BaseModel):
  File "/opt/text-generation-webui/extensions/openai/typing.py", line 9, in GenerationOptions
    preset: str | None = Field(default=None, description="The name of a file under text-generation-webui/presets (without the .yaml extension). The sampling parameters that get overwritten by this option are the keys in the default_preset() function in modules/presets.py.")
TypeError: unsupported operand type(s) for |: 'type' and 'NoneType'
`

### Possible Cause
Upon investigation, it seems the issue stems from the [text-generation-webui](https://github.com/oobabooga/text-generation-webui) Docker container using Python 3.10+ syntax, while only Python 3.8 is available in the Docker. This discrepancy leads to the above error. An attempt to rewrite the typing part in Python 3.8 resolved this error, but embedding still did not function correctly.

### Request for Assistance
Could you please verify and address this compatibility issue? Any help in resolving this would be greatly appreciated.

Thank you in advance for your assistance.

Best Regards,

MrBrabus75, 2023-12-05:
Up please

johnnynunez, 2023-12-05:
> Up please

jetpack 6 is coming with python 3.10(ubuntu 22.04). Release today

closed 2023-12-11 by MrBrabus75