dusty-nv/jetson-containers #339: Can't get LLama Speak to run with USB Audio

lesreaper, 2023-11-29:
nVidia Jetson Orin AGX, Respeaker 4.0 USB Mic Array
[LlamaSpeak Tutorial](https://github.com/dusty-nv/jetson-containers/tree/master/packages/llm/llamaspeak)

Installed Riva and Python Client no problem. Tested and works with USB audio.

Left Riva container running in the background. Set up SSH Key. Loaded the model into the [text-generation-webui](https://github.com/dusty-nv/jetson-containers/blob/master/packages/llm/text-generation-webui), and it's on `port 7860`.

However, when I go to run `llamaspeak`, it dies every time. Command I use is:
```
./run.sh --env SSL_CERT=/data/cert.pem --env SSL_KEY=/data/key.pem $(./autotag llamaspeak)
```
The output is:
```
./run.sh --workdir=/opt/llamaspeak \
>   --env SSL_CERT=/data/cert.pem \
>   --env SSL_KEY=/data/key.pem \
>   $(./autotag llamaspeak) \
>   python3 chat.py --verbose --debug
Namespace(disable=[''], output='/tmp/autotag', packages=['llamaspeak'], prefer=['local', 'registry', 'build'], quiet=False, user='dustynv', verbose=False)
-- L4T_VERSION=35.4.1  JETPACK_VERSION=5.1.2  CUDA_VERSION=11.4.315
-- Finding compatible container image for ['llamaspeak']
dustynv/llamaspeak:r35.4.1
localuser:root being added to access control list
+ docker run --runtime nvidia -it --rm --network host --volume /tmp/argus_socket:/tmp/argus_socket --volume /etc/enctune.conf:/etc/enctune.conf --volume /etc/nv_tegra_release:/etc/nv_tegra_release --volume /tmp/nv_jetson_model:/tmp/nv_jetson_model --volume /home/nvidia/Documents/jetson-containers/data:/data --device /dev/snd --device /dev/bus/usb -e DISPLAY=:1 -v /tmp/.X11-unix/:/tmp/.X11-unix -v /tmp/.docker.xauth:/tmp/.docker.xauth -e XAUTHORITY=/tmp/.docker.xauth --workdir=/opt/llamaspeak --env SSL_CERT=/data/cert.pem --env SSL_KEY=/data/key.pem dustynv/llamaspeak:r35.4.1 python3 chat.py --verbose --debug
Namespace(audio_channels=1, audio_chunk=1600, automatic_punctuation=True, boosted_lm_score=4.0, boosted_lm_words=None, debug=True, input_device=None, language_code='en-US', list_devices=False, llm_api_port=5000, llm_server='0.0.0.0', llm_streaming_port=5005, log_level=2, max_new_tokens=256, metadata=None, no_punctuation=False, no_verbatim_transcripts=False, output_device=None, profanity_filter=False, sample_rate_hz=48000, server='localhost:50051', speaker_diarization=False, ssl_cert='/data/cert.pem', ssl_key='/data/key.pem', use_ssl=False, verbatim_transcripts=True, verbose=True, voice='English-US.Female-1', web_port=8050, web_server='0.0.0.0')
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/urllib3/connection.py", line 159, in _new_conn
    conn = connection.create_connection(
  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 84, in create_connection
    raise err
  File "/usr/lib/python3/dist-packages/urllib3/util/connection.py", line 74, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused
```
I think it comes down to `llm_streaming_port` value being wrong, but I have no idea. What am I doing wrong?

dusty-nv, 2023-11-29:
@lesreaper unfortunately what I believe has happened is that oobabooga has removed/changed their API.  llamaspeak:v2 from the local_llm container.  I will look into rolling back the oobabooga commit SHA for the llamaspeak:v1 build.

dusty-nv, 2023-12-05:
OK @lesreaper, sorry for the delay - there we some issues I had to workaround.  Can you try running the `dustynv/text-generation-webui:1.7-r35.4.1` container instead?

```bash
./run.sh --workdir /opt/text-generation-webui dustynv/text-generation-webui:1.7-r35.4.1 \
   python3 server.py --listen --verbose --api \
	--model-dir=/data/models/text-generation-webui
```

This version still has the compatible oobabooga API with llamaspeak.

lesreaper, 2023-12-05:
Thank you! I tried building that locally, and it was a disaster trying to get everything in sync on that project on the Orin.

I ran the `text-generation-webui`, and it now runs the API endpoint at 5000, at least I think it does. It models, and I can speak on port 7860 to that model, but I can only load a 7b or 13b model. I tried a 30b model and it ran out of memory. That's a separate issue, but at least it's saying it's working now. 

The llama-speak does load up!

I'm install unable to get the USB to be recognized however. The USB microphone is a ReSpeaker USB Mic Array, and it's active and works fine in the desktop setup. However, when I select the device in Firefox, and then try to say smoething, nothing gets picked up. Also, I can't type anything into the box. This is `berbose` logging from the llama-speak startup:
```
Namespace(audio_channels=1, audio_chunk=1600, automatic_punctuation=True, boosted_lm_score=4.0, boosted_lm_words=None, debug=False, input_device=None, language_code='en-US', list_devices=False, llm_api_port=5000, llm_server='0.0.0.0', llm_streaming_port=5005, log_level=1, max_new_tokens=256, metadata=None, no_punctuation=False, no_verbatim_transcripts=False, output_device=None, profanity_filter=False, sample_rate_hz=48000, server='localhost:50051', speaker_diarization=False, ssl_cert='/data/cert.pem', ssl_key='/data/key.pem', use_ssl=False, verbatim_transcripts=True, verbose=True, voice='English-US.Female-1', web_port=8050, web_server='0.0.0.0')
...
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
Expression 'alsa_snd_pcm_hw_params_set_rate_near( pcm, hwParams, &setRate, NULL )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 3201
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2642:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
-- running ASR service (en-US)
-- running TTS service (en-US, English-US.Female-1)
-- running AudioMixer thread
-- starting webserver @ 0.0.0.0:8050
-- running LLM service (teknium_OpenHermes-2.5-Mistral-7B)
 * Serving Flask app 'webserver'
 * Debug mode: on
WARNING: This is a development server. Do not use it in a production deployment. Use a production WSGI server instead.
 * Running on all addresses (0.0.0.0)
 * Running on https://127.0.0.1:8050
 * Running on https://10.0.1.183:8050
```

lesreaper, 2023-12-08:
Is this the problem? Not sure what it means?
```
Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
```
I have a demo next Thursday at an Ivy and I'd love to showcase what these AGX's can do. Thanks!

dusty-nv, 2023-12-08:
Sorry for the delay @lesreaper, and that it can be a bit tricky to get the audio all working with the web setup - are you able to use llamaspeak if you run the browser on a PC client, as opposed to the Jetson?

Normally if USB audio device is directly attached to jetson, I would use `--input-device=N` and `--output-device=N` (where N is the device index that `--list-devices` shows)

Also, if you haven't already, I would test that Riva works with your audio device:   https://github.com/dusty-nv/jetson-containers/tree/master/packages/audio/riva-client#list-audio-devices

lesreaper, 2023-12-12:
Sorry for taking so long to get back on this one.

I ran this and here was the response:
```
./run.sh \
--workdir /opt/riva/python-clients \
$(./autotag riva-client:python) \
   python3 scripts/list_audio_devices.py
   
   Output:
 0: ReSpeaker 4 Mic Array (UAC1.0): USB Audio (hw:0,0) (inputs=6   outputs=0   sampleRate=16000)
 1: NVIDIA Jetson AGX Orin HDA: HDMI 0 (hw:1,3)        (inputs=0   outputs=8   sampleRate=44100)
 2: NVIDIA Jetson AGX Orin HDA: HDMI 1 (hw:1,7)        (inputs=0   outputs=8   sampleRate=44100)
 3: NVIDIA Jetson AGX Orin HDA: HDMI 2 (hw:1,8)        (inputs=0   outputs=8   sampleRate=44100)
 4: NVIDIA Jetson AGX Orin HDA: HDMI 3 (hw:1,9)        (inputs=0   outputs=8   sampleRate=44100)
```

Ran the `riva` samples in the `python-client` library and had no problem with them.

It won't let me run the `--input-device=N` on the `llamaspeak`. Tells me it's an unknown flag.

I'm wondering since there are 6 inputs if I need to modify anything in the containers, or set something else up. Any ideas?

dusty-nv, 2023-12-12:
> It won't let me run the `--input-device=N` on the `llamaspeak`. Tells me it's an unknown flag.

Hmm, here it is in the code for chat.py:

https://github.com/dusty-nv/jetson-containers/blob/93739890ec4f575c113e0c8cfad76a6e4e1a51d3/packages/llm/llamaspeak/chat.py#L36

Did you try running it like this?

```bash
./run.sh --workdir=/opt/llamaspeak \
  --env SSL_CERT=/data/cert.pem \
  --env SSL_KEY=/data/key.pem \
  $(./autotag llamaspeak) \
  python3 chat.py --verbose --input-device=0
```

> I'm wondering since there are 6 inputs if I need to modify anything in the containers, or set something else up. Any ideas?

It looks like it's possible that you may need to also set `--sample-rate-hz=16000`

cj401, 2023-12-28:
Hello @dusty-nv, I am trying to run llamaspeak on Jetson. 
I followed the tutorial to set RIVA up and test transcribe_mic.py with riva docker container running in the background. It works.

```
./run.sh --workdir /opt/text-generation-webui dustynv/text-generation-webui:1.7-r35.4.1 \
   python3 server.py --listen --verbose --api \
	--model-dir=/data/models/text-generation-webui
```
works as well. 

The docker container can be started as shown in the pic:
[image]
However, when trying to speak to the mic, there was no response from the the following interface

[image]
I reckon it is due to the ASR transcribe did not work and wonder how I can get it work. :) 

Plus, I watched your video  [here](https://www.youtube.com/watch?v=wzLHAgDxMjQ), which is awesome. I wonder how you got agent option on the top right as well? 

Many thanks.

cj401, 2024-01-02:
@dusty-nv  I wonder if you will have time to have a look at this? Many thanks.

dusty-nv, 2024-01-02:
> ./run.sh \
> >     -e HUGGINGFACE_TOKEN=hf_xxxxxx \
> >     -e SSL_KEY=/data/key.pem \
> >     -e SSL_CERT=/data/cert.pem \
> >     $(./autotag local_llm) \
> >     python3 -m local_llm.agents.web_chat \
> >     --model liuhaotian/llava-v1.5-13b \
> >     --api=mlc --verbose

Hi @cj401, sorry for the delay - the command above will use browser mic, not USB mic.  For USB mic connected to Jetson, you would specify `--audio-input-device` option.

> Not sure why the user is 'dustynv'. How can I change this to $USER?

This is referring to the dockerhub user to look for containers under, not the linux $USER.   dustynv is my dockerhub user where all the container images are, so it is correct.

> `23:12:05 | DEBUG | send_message() - no websocket clients connected, dropping json message`

Can you view the browser developer console log by pressing `Ctrl+Shift+I`?   And see if it is connecting to the server.

cj401, 2024-01-08:
Hi @dusty-nv  thank you for your reply and I know you must be very busy with various things.
 with --audio-input-device, the command line audio transcribe works.

However, I have not managed to get the multi-modal model work as shown [here](https://github.com/dusty-nv/jetson-containers/tree/master/packages/llm/local_llm#voice-chat) and [this video](https://www.youtube.com/watch?v=wzLHAgDxMjQ). 

Though I tried to various ways to configure the speaker as shown in the below pic:
[image]
I wonder what your command line is for getting the above video.  Many thanks.

Adassiruseri, 2024-10-08:
[log_doc.txt](https://github.com/user-attachments/files/17292690/log_doc.txt)
Hi @dusty-nv , I  am trying to run llamaspeak on Jetson  AGX Orin DevKit 64GB. The web chat page is accessible and also checked on terminal that webchat system is ready. However my voice is not being taken as an input even though the device is detected in the Audio settings window of the chat page. tts is working fine (generated output via text chat box is received on mic).  I suppose issue is with asr (tried replacing 'riva' with 'whisper' in the below commands but behavior is the same).

   `git clone https://github.com/dusty-nv/jetson-containers
bash jetson-containers/install.sh
jetson-containers run --env HUGGINGFACE_TOKEN=hf_xxxxxx \
  $(autotag nano_llm) \
  python3 -m nano_llm.agents.web_chat --api=mlc \
    --model meta-llama/Meta-Llama-3-8B-Instruct \
    --asr=riva --tts=piper` 
    
Do i need to start the riva server before running the above commands. Is there any dependency?
I have attached the log of the terminal ( USB mic parameters are also logged in it). Please let me know about how to perform asr through my USB mic.

closed 2025-02-21 by johnnynunez